linuxcnc-ethercat/linuxcnc-ethercat #2: Identify stuck pull requests and merge as many as possible.

scottlaird, 2022-12-31:
Traditionally, pull requests on sittner/linuxcnc-ethernet were fairly slow.  There should be at least a few queued up that would be worth merging.  Identify and resolve.

ThomasEg, 2023-01-02:
@satiowadahc On [another thread](https://github.com/sittner/linuxcnc-ethercat/pull/86) you wrote that you dont make PR's for every driver you write... but now is your chance to get some of that sweet, sweet github-contributer credit ;-) 

If you know of any drivers that could/should be included, this would be an good place to list them...or maybe create PRs?

Bouni, 2023-04-21:
+1 for this!

I can totally understand that @sittner don't has the time to handle all this and having a central place where more people can help getting the project improve is a nice thing!

Once I have a setup I'll help where I can!

closed 2024-01-02 by scottlaird