linuxcnc-ethercat/linuxcnc-ethercat #3: Set up basic standards for submission, reviews, CI/CD, etc

scottlaird, 2022-12-31:
I don't have a ton of experience running Github projects, but at an absolute minimum we want to make sure that every pending pull request still builds.  Tests would be nice, although hardware projects in C aren't easy to test.  Find a reasonable set of configs and steal?

scottlaird, 2023-01-02:
I'd like to *at least* verify that things build before merging.  Unfortunately, this will require having LinuxCNC libraries installed at a minimum.

ThomasEg, 2023-01-02:
Agreed. It would be very nice to have a guide on how to write new drivers and how to get them "approved"/merged. As an ethercat newbie this has a very steep learning curve(like everything else around LinuxCNC and ethercat)...

Bouni, 2023-05-17:
We also should think about a GH action that runs a linter on every PR to keep code style consistent.

I've never set one up for C code but clang-format seems to be the tool of choice and is available as a GH action

https://github.com/jidicula/clang-format-action

scottlaird, 2023-05-18:
I agree, but it'd be nice to get the whole codebase passing, which I doubt it does, before we enforce it for new PRs.  Doing that will ~break pending changes, which isn't *fatal*, but should be done carefully.  This might be a good time for it, though...

scottlaird, 2023-05-18:
FWIW, `clang-format` touches 101 files, with 6742 insertions and 4743 deletions right now, using the default format.  Spot-checking, it's doing some (IMO) dumb 80-ish column wrapping and breaks the alignment of `#defines` that looked better initially.

I bumped the column limit to 120 (probably excessive) and enabled `AlignConsecutiveMacros: AcrossComments`, then compared the various default styles to see how big of a diff they provide (via diffstat).

* `--style=LLVM` gives +4343/-4424 lines of diff
* `--style=GNU` gives +10650/-8786
* `--style=Google` gives +3580, -3457
* `--style=Chromium` gives +5149/-4970
* `--style=Microsoft` gives +14551/-12931
* `--style=Mozilla` gives +6370/-4193
* `--style=WebKit` gives +13467/-13208

Looking at these numbers, `--style=Google` seems to produce the smallest diff.

Adding two extra options fixes a few weird formatting inconsistencies:

```
DerivePointerAlignment: false
PointerAlignment: Right
```

HOWEVER, there are a bunch of places in the code that `clang-format`  mangles, because the existing code includes additional spaces to impart logical structure and automatic formatting gets rid of it.  Example:

```
   // initialize sync info
   lcec_syncs_init(&hal_data->syncs);
-    lcec_syncs_add_sync(&hal_data->syncs, EC_DIR_OUTPUT, EC_WD_DEFAULT);
-    lcec_syncs_add_sync(&hal_data->syncs, EC_DIR_INPUT, EC_WD_DEFAULT);
-    lcec_syncs_add_sync(&hal_data->syncs, EC_DIR_OUTPUT, EC_WD_DEFAULT);
-      lcec_syncs_add_pdo_info(&hal_data->syncs, 0x0018);
-        lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0086, 0x01,  16); // control-word
-        lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0018, 0x01,  32); // velo-command
-    lcec_syncs_add_sync(&hal_data->syncs, EC_DIR_INPUT, EC_WD_DEFAULT);
-      lcec_syncs_add_pdo_info(&hal_data->syncs, 0x0010);
-        lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0087, 0x01, 16); // status word
-        lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0033, 0x01, 32); // position feedback
-        lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0054, 0x01, 16); // torque feedback
-        if (hal_data->chan.fb2_enabled) {
-          lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0035, 0x01, 32); // position feedback 2
-        }
-        if (hal_data->chan.diag_enabled) {
-          lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0186, 0x01, 32); // diagnostic number
-        }
+  lcec_syncs_add_sync(&hal_data->syncs, EC_DIR_OUTPUT, EC_WD_DEFAULT);
+  lcec_syncs_add_sync(&hal_data->syncs, EC_DIR_INPUT, EC_WD_DEFAULT);
+  lcec_syncs_add_sync(&hal_data->syncs, EC_DIR_OUTPUT, EC_WD_DEFAULT);
+  lcec_syncs_add_pdo_info(&hal_data->syncs, 0x0018);
+  lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0086, 0x01, 16);  // control-word
+  lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0018, 0x01, 32);  // velo-command
+  lcec_syncs_add_sync(&hal_data->syncs, EC_DIR_INPUT, EC_WD_DEFAULT);
+  lcec_syncs_add_pdo_info(&hal_data->syncs, 0x0010);
+  lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0087, 0x01, 16);  // status word
+  lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0033, 0x01,
+                           32);  // position feedback
+  lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0054, 0x01,
+                           16);  // torque feedback
+  if (hal_data->chan.fb2_enabled) {
+    lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0035, 0x01,
+                             32);  // position feedback 2
+  }
+  if (hal_data->chan.diag_enabled) {
+    lcec_syncs_add_pdo_entry(&hal_data->syncs, 0x0186, 0x01,
+                             32);  // diagnostic number
+  }
```

I could manually disable formatting for these blocks, or find an alternative structure that formats correctly (`do { ... } while 0`?), but that doesn't look like an improvement overall.

I'll look into this more ~tomorrow.

Bouni, 2023-05-22:
I totally agree that this style checks should keep changes at minimum but also think we should find a way to enforce a consistent code style. 
I see you almost got to the point where clang-format is kind of happy with the current code base with just a few places that needs special attention.

In your example you refer to this section in the code: https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/master/src/lcec_ax5200.c#L82-L114

In my opinion using additional indentation is not a very good choice in my opinion, additional newlines with comments would do a much better job and clang-format would be satisfied with that.

Bouni, 2023-05-22:
would you min setting up a new branch with a `.clang-format` in it so that we can tweak the parameters from there on?

scottlaird, 2023-12-17:
FWIW, I just merged a basic CI action that manages to install LinuxCNC and Ethercat Master, and then verify that LinuxCNC-Ethercat at least manages to build.  I'm making a bunch of CNC changes this week, and should finally be able to make progress here.

scottlaird, 2023-12-17:
Okay, I'm fairly happy with the build workflow now.  I'm sure it could be better, but it functions and it's relatively quick, considering the number of moving pieces.  That leaves formatting and lint issues.  I'm not at all convinced that linting C device driver code is going to accomplish anything useful, at least on existing code.

We should probably set a standard for new code formatting now, but not consider applying it to existing code until *after* we're not worried about merging anymore.  I suspect that the clang-format from above is probably as good as we're going to get; if anyone really wants to poke at it then feel free, otherwise I'll look at adding it to documentation and maybe adding a script.

Assuming that we actually have documentation, of course :-).

scottlaird, 2023-12-17:
Okay, I've updated the CONTRIBUTING doc and added a .clang--format file; feel free to propose changes from there.

scottlaird, 2024-01-02:
We still need better docs on how to write modules and which code to use as examples.  Some is *clearly* better than others.

scottlaird, 2024-01-11:
Closing, as we at least have "basic standards" now.  Documentation can always be better, but that's easy enough to iterate on and there's a different bug for that.

closed 2024-01-11 by scottlaird