linuxcnc-ethercat/linuxcnc-ethercat #7: Better documentation

scottlaird, 2022-12-31:
Setting up LinuxCNC with EtherCAT is difficult.  We need better documentation, ideally including a basic sample config for a simple 3-axis CNC.

Bouni, 2023-04-21:
In the long run I think it might be a good idea to have ready to use packages that can be installed through apt without the need to build them from source.
I'm not sure if that's even possible because of several config options available in the build process but we might figure something out that works at least for a basic 3-axis CNC as you said!

scottlaird, 2024-01-08:
Things that need to be documented soonish:

- [ ] Installing from packages (since this is getting pretty standard now)
- [ ] The meaning of various fields in `ethercat.xml`.  See https://forum.linuxcnc.org/ethercat/44113-ethercat-xml-file-configuration for `appTimePeriod` and `refClockSyncCycles`.
- [ ] How to set up generic devices in `ethercat.xml`.
- [ ] Parameters for other, non-generic devices in `ethercat.xml`.
- [ ] A reference config for a generic 3-axis CNC would be nice as well.