linuxcnc-ethercat/linuxcnc-ethercat #8: Identify known EtherCAT devices, their support status, and pointers to people with sample hardware

scottlaird, 2022-12-31:
The actual support matrix is kind of messy right now.  It doesn't help that Beckhoff produces a huge number of very similar devices; ELxxxx and EPxxxx devices are frequently completely compatible, but use different IDs so drivers won't work with both out of the box.

We should create a Wiki page that contains all of the allegedly-working hardware as well as additional available modules that *may* work and ones that have no support at all.

Bouni, 2023-04-21:
I have a good amount of various EtherCat hardware available to help test or even implement new stuff.

### Beckhoff 
- CX20XX CPU modules
- C6015 IPCs 
- ELXXXX I/O terminals
- EPXXXX I/O terminals
- EKXXXX EtherCat bus terminals
- AX8000 Servodrives

### Omron
- G5 Servodrives
- NX Couplers and I/O terminals

### Sanyo Denki 
- RS2 Servodrives

Maybe even more stuff that I don't remember at the moment 😅

Bouni, 2023-04-28:
A Matrix like this would be great to document the state of the project:

| Device | Manufacturer | Implemented | Tested |
|--------|--------------|-------------|--------|
| EP2316 | Beckhoff | @scottlaird | @bouni |

Bouni, 2023-05-23:
After thinking about it this could be automated by a GH action which grabs predefined ESI files from various manufacturers,compares them to the PID's listed in the .h files and generates a list out of the results.

Ideally we could add something like `// implemented by scottlaird, tested by bouni` after each PID to compile even more info into the list automatically.

Bouni, 2023-05-26:
I did a test python script that could possibly be used to generate a `SUPPORT_MATRIX.md` file as a GH action cron job and if it differs from the one in the repo commit it automatically.

I download ESI files for Beckhoff and OMRON, parse them, compare them to the PIDs in the header files and generate a Markdown list out of it.

This list looks like this at the moment: https://gist.github.com/Bouni/b93fe1aa8b9099fad4f6ca154dbdb7e0

Let me know what you think about this.

UweSauter, 2023-11-19:
I think it should also somehow be indicated if different variations of the devices need different configuration. E.g. I have both EL2024 and EL2024-0010 (24V vs. 12V variants). If I try to identify the EL2024-0010 correctly in the XML (`<slave idx="7"  type="EL2024-0010" />`  then `lcec_conf` shows an error:
```console
$ halrun
halcmd: loadrt threads name1=master period1=1000000
Note: Using POSIX realtime
halcmd: loadusr -W lcec_conf ec.xml
lcec_conf: ERROR: Invalid slave type EL2024-0010
lcec_conf: ERROR: Parse error at line 10: parsing aborted
<stdin>:2: waitpid failed lcec_conf lcec_conf
<stdin>:2: lcec_conf exited without becoming ready
```

I know that there are EtherCAT devices where variations (`-XXXX`) provide different functionality and thus need different configuration. E.g. [EL2521-xxxx](https://infosys.beckhoff.com/english.php?content=../content/1033/el252x/1702170635.html)

scottlaird, 2023-12-18:
I've made a bit of initial progress on this; see the notes from PR #27 and https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/master/documentation/DEVICES.md

scottlaird, 2023-12-18:
Huh.  Well *that* would have been easier.  From the ESI ZIP file:

```
<Type ProductCode="#x03ea3052" RevisionNo="#x00000000">EL1002</Type>
<Name LcId="1033"><![CDATA[ EL1002 2Ch. Dig. Input 24V, 3ms ]]></Name>
<Name LcId="1031"><![CDATA[ EL1002 2K. Dig. Eingang 24V, 3ms ]]></Name>
<URL LcId="1033"><![CDATA[ http://www.beckhoff.com/EL1002 ]]></URL>
<URL LcId="1031"><![CDATA[ http://www.beckhoff.de/EL1002 ]]></URL>
```

I spent way too much time manually finding URLs and descriptive names.  With this, I can probably generate the rest of the documentation fairly quickly.

scottlaird, 2023-12-18:
Also, I should add a link to the source file (.c, presumably) from each device.

scottlaird, 2023-12-19:
Okay, I'm now generating the DEVICE.md file from ESI data.  This broke most of the non-Beckhoff entries, but there aren't very many and they should be fairly easy to fix in a couple more PRs.

closed 2023-12-20 by scottlaird

scottlaird, 2024-02-09:
> I think it should also somehow be indicated if different variations of the devices need different configuration. E.g. I have both EL2024 and EL2024-0010 (24V vs. 12V variants). If I try to identify the EL2024-0010 correctly in the XML (`<slave idx="7" type="EL2024-0010" />` then `lcec_conf` shows an error:
> 
> ```
> $ halrun
> halcmd: loadrt threads name1=master period1=1000000
> Note: Using POSIX realtime
> halcmd: loadusr -W lcec_conf ec.xml
> lcec_conf: ERROR: Invalid slave type EL2024-0010
> lcec_conf: ERROR: Parse error at line 10: parsing aborted
> <stdin>:2: waitpid failed lcec_conf lcec_conf
> <stdin>:2: lcec_conf exited without becoming ready
> ```
> 
> I know that there are EtherCAT devices where variations (`-XXXX`) provide different functionality and thus need different configuration. E.g. [EL2521-xxxx](https://infosys.beckhoff.com/english.php?content=../content/1033/el252x/1702170635.html)

This is an interesting problem in general; *most* Beckhoff devices with the same major model number but different submodel numbers (EL2521-0000 vs EL2521-0024, for instance) are fairly compatible.  The EL2521 is an interesting case, as *most* (but not all) of the flavors have pretty much identical PDO definitions.

According to [esi-data](https://linuxcnc-ethercat.github.io/esi-data/devices/EL2521), the EL2521-0000, -0024, -0025, and -1001 are identical.  The [EL2521-0124](https://linuxcnc-ethercat.github.io/esi-data/devices/EL2521-0124), however, is completely different.

So far, I've been approaching this by using the base name whenever possible, and leaving `-0000` out.  We have a few cases where we have different drivers for different sub-models; the EL7041, EL7041-0052, and EL7041-1000 come to mind, although I don't remember what the exact difference is between them off the top of my head.  I know that the setup SDOs are different, but that's *mostly* fine.

We should probably have this written out as a policy somewhere.  I have nothing against adding submodel-specific code, or even entire drivers dedicated to a single sub-module, but I'd like to be as consistent as possible, so we don't end up with `EL7041_1000` and `EL7041-1000` again, or with giant hordes of `EPxxxx-xxxx` modules being added to the definitions even though they're functionally identical with each other.   I'm calling out EP modules specifically, because Beckhoff *mostly* only refers to them by their full model+submodel name on their website, unlike the EL modules which usually bury the submodel deeper in the documentation.

PR #39 (2023-12-20): Update Beckhoff device descriptions for consistency
Update the device type field for most Beckhoff devices for consistency.  At this point, I'm out of known issues with DEVICES.md.

Fixes #8.