linuxcnc-ethercat/linuxcnc-ethercat #42: Regularize e[lp][12]xxx drivers

scottlaird, 2023-12-20:
The digital I/O drives (`lcec_el1xxx.[ch]` and `lcec_el2xxx.[ch]`) include support for quite a few devices, but have a few shortcomings:

- People have added PIDs one at a time as they need them, vs bulk-adding all of the devices from the ESI file that should work.
- Devices with a mix of input and output end up needing custom code (`lcec_el1859.c` and `lcec_ep23*.c`).
- The same basic code gets duplicated into `lcec_el7041` and other devices that have digital I/O as a secondary function.

I'd like to fix these.  First, add a bit of library code for defining PDOs for digital input and for digital output.  Then make `lcec_el1xxx` and `lcec_el2xxx` use that.  Finally, merge the two, along with at least `lcec_el1859` and `lcec_ep23xx.c`, driving the setup with a table derived from `scripts/esi.yml`.  Obviously we don't want the C code reading YML, so we'd probably want something to extract data from the ESI file and produce a table that the C code could use.

scottlaird, 2023-12-20:
Partial list of digital I/O devices (as opposed to just-input or just-output):

- EL1259 (8+8, with timestamp)
- EL1852 (8+8)
- EL1859 (8+8)
- EL2911 (4+1, with TwinSAFE)
- EP1957 (8+4, with TwinSAFE)
- EP2338 (8, each port I or O)
- EP2339 (16, each port I or O)
- EP2349 (16, each port I or O)
- EPP1859 (8+8)
- EPP2334 (4, each port I or O)
- EPP2338 (8, each port I or O)
- EPP2339 (16, each port I or O)
- EPP2349 (16, each port I or O)
- EQ2339 (16, each port I or O)
- ER2338 (8, each port I or O)
- ER2339 (16, each port I or O)
- ER2349 (16, each port I or O)

The 'timestamp' and 'TwinSAFE' models may or may not fit into the same driver.  I suspect that timestamps *might*, but TwinSAFE probably won't.  The rest of these should be pretty similar in software.

Also, a number of non-el[12]xxx devices have either digital I or O (or both) as a secondary function, including:

- EP6224 (0, 4, or 8 I, depending on the sub-model)
- EP7041 (2 I on at least some models)
- EP7047 (2 I)
- EP7211 (2 I)
- EP7402 (8, each port I or O)
- EP8309 (8, each port I or O, plus 2 analog in and 1 analog out)
- EL8601 (4 I, 4 more I/counter/encoder multifunction, 2 O w/ PWM)

Hmm.  PWM is another thing that some modules have that *might* fit into a standard library.

My expectation is that we'll end up with a set of I/O PDO modules, and we could end up saying something like this (pseudocode):

```
el2124_pdos = [
    DigitalOut(0, 0x7000),
    DigitalOut(1, 0x7010),
    DigitalOut(2, 0x7020),
    DigitalOut(3, 0x7030),
]

el1852_pdos = [
  DigitalOut(0, 0x7000),
  DigitalOut(1, 0x7010),
  ...
  DigitalIn(0, 0x6000),
  DigitalIn(1, 0x6010),
  ...
]
```

Presumably we'd actually build a struct with VID, PID, maybe revision #, and this data.  Then we'd use the VID/PID/RevNo at scan time and create a set of PDOs for it then.

scottlaird, 2024-01-18:
Okay, I've cleaned up the el1xxx driver a bit, and pulled all of the actual HW access out into `lcec_class_din`.  I'll add `lcec_class_dout` and `el2xxx` next, then look for random din/dout pins in other drivers that could be cleaned up.

Then on to the in/out devices.

scottlaird, 2024-01-18:
One more thing--as things stand, the el1xxx driver is almost dead simple.  It's basically just boilerplate plus a *tiny* bit of logic for mapping pin->PDO address.  The el2xxx driver will be ~identical.

Given that, there's no point in complicating them for random I/O devices with different mappings.  We can add new drivers easier than we can add alternate control flow in el1xxx and el2xxx.

scottlaird, 2024-01-18:
Hmm.  Although we could go a bit further and change `lcec_din_allocate_pins` to return a struct instead of just a `**lcec_class_din_pin_t`, and we could include the number of pins allocated in the struct.

Then we could provide a `lcec_din_read_all()` that would loop over all of the pins.  That's slightly easier than the current model, and it gets rid of the "scary" `**` that will probably make people with less C experience twitchy.

scottlaird, 2024-01-20:
Looking at things as they stand, I think the current EL1859 driver is actually capable of handling anything in el1xxx, el2xxx, or el1252; all we'd have to do is add the PID definitions at the top.

Given that, I'll probably rename `lcec_el1859` to something more generic (`lcec_digitalio`?) and migrate all of the remaining el[12] devices into it.  But not today.

thomam04, 2024-01-20:
I would probably stay with Beckhoff's naming convention for their ethercat-terminals - but that's just me. 

el1xxx
el2xxx
el3xxx

It's great that we combine, add and simplify adding new driver but at some point I would like to add more modParam to the driver to match all of the options the module can support. I am also wondering if a single driver then gets too big and difficult to modify, at least for people like me with little to no C language skills. The el1xxx fairly simple but I am thinking of others.

scottlaird, 2024-01-20:
The problem is the weird cases--the EL185x, the EP23xx, the EK18xx -- they're all a mix of 1xxx and 2xxx, and don't fit into either.  Right now, they're all in `lcec_el1859.c`, which isn't exactly where you'd expect to find the driver for the EPP2349, for example, but there it is.

At a minimum, I want to rename the EL1859 driver, as it's absolutely mis-named.  Merging the el1xxx and el2xxx into it would be trivial -- it literally supports everything that they can do, they'd just need per-device definitions added to `types[]` at the top.  But, the el1xxx and el2xxx drivers are utterly simple today, and there's a point in not messing with that simplicity, even if it's just to use them as sample code for others.

But that still leaves us with a name for the (formerly) EL1859 driver.  "lcec_digitalcombo", maybe?

closed 2024-01-22 by scottlaird