linuxcnc-ethercat/linuxcnc-ethercat #115: Support 64-bit IEEE floats in lcec_generic

scottlaird, 2024-01-03:
As per https://forum.linuxcnc.org/ethercat/51187-new-linuxcnc-ethercat-git-tree#289534, there's a request for 64-bit IEEE float support for generic devices.  LinuxCNC uses `double` internally, and our Ethercat library can send and receive 64-bit floats, so we just need to extend the syntax a bit.

scottlaird, 2024-01-03:
As an aside, looking through all of the devices (including revisions) in `scripts/esi.yml`, I see 650 64-bit PDOs.  They're all either LINT or ULINT.  So Beckhoff, at least, doesn't make any hardware that uses 64-bit floats.  Or I'm parsing something wrong.

MetalMusings, 2024-01-11:
Hi, was about to test this today, but the generic driver wasn't included in lcec_conf (I think) Unknown slave type generic was the message.

scottlaird, 2024-01-11:
Err... that's not good.  Just a sec.

scottlaird, 2024-01-11:
Once #143 finishes merging, can you check again?

MetalMusings, 2024-01-11:
nooo... more problem with lcec_generic
First time with unmodified ethercat-conf.xml (I know the float-double line is wrong but I didn't come that far)
```
lcec_conf: ERROR: Invalid slave attribute vid
lcec_conf: ERROR: Parse error at line 3: parsing aborted
./core_sim_lathe.hal:8: waitpid failed lcec_conf lcec_conf
./core_sim_lathe.hal:8: lcec_conf exited without becoming ready
```
I removed the vid and pid attributes from the slave line in ethercat-conf.xml and instead get
```
lcec_conf: ERROR: Invalid slave attribute configPdos
lcec_conf: ERROR: Parse error at line 3: parsing aborted
./core_sim_lathe.hal:8: waitpid failed lcec_conf lcec_conf
./core_sim_lathe.hal:8: lcec_conf exited without becoming ready
```
I also remove the configPdos attribute and get
```
lcec_conf: ERROR: syncManager is only allowed on generic slaves
lcec_conf: ERROR: Parse error at line 5: parsing aborted
./core_sim_lathe.hal:8: waitpid failed lcec_conf lcec_conf
./core_sim_lathe.hal:8: lcec_conf exited without becoming ready
```
And then I give up. My ethercat-conf.xml is as follows and the test is to change the float-double setting
```
<masters>
  <master idx="0" appTimePeriod="1000000" refClockSyncCycles="-1"> <!-- was "5" -->
    <slave idx="0" type="generic" vid="00aaa" pid="000bbbccc" configPdos="true">
      <dcConf assignActivate="300" sync0Cycle="*1" sync0Shift="0"/> 
      <syncManager idx="2" dir="out">
        <pdo idx="1600">
          <pdoEntry idx="7000" subIdx="00" bitLen="32" halPin="enc-pos-scale" halType="s32"/>
        </pdo>
        <pdo idx="1601">
          <pdoEntry idx="7001" subIdx="00" bitLen="32" halPin="enc-index-latch-enable" halType="u32"/>
        </pdo>
        <pdo idx="1602">
          <pdoEntry idx="7002" subIdx="01" bitLen="64" halPin="commanded-position" halType="float-double"/>
          <pdoEntry idx="7002" subIdx="02" bitLen="64" halPin="commanded-velocity" halType="float-double"/>
        </pdo>
      </syncManager>
      <syncManager idx="3" dir="in">
        <pdo idx="1a00">
          <pdoEntry idx="6000" subIdx="00" bitLen="64" halPin="enc-position" halType="float-double"/>
        </pdo>
        <pdo idx="1a01">
          <pdoEntry idx="6001" subIdx="00" bitLen="64" halPin="enc-frequency" halType="float-double"/>
        </pdo> 
        <pdo idx="1a02">
          <pdoEntry idx="6002" subIdx="00" bitLen="32" halPin="DiffT" halType="u32"/>
        </pdo>
        <pdo idx="1a03">
          <pdoEntry idx="6003" subIdx="00" bitLen="32" halPin="index-byte" halType="u32"/>
        </pdo>
        <pdo idx="1a04">
          <pdoEntry idx="6004" subIdx="00" bitLen="32" halPin="index-status" halType="u32"/>
        </pdo>
        <pdo idx="1a05">
          <pdoEntry idx="6005" subIdx="01" bitLen="64" halPin="actual-position" halType="float-double"/>
        </pdo>
      </syncManager> 
    </slave>
  </master>
</masters>
```

scottlaird, 2024-01-12:
Okay, I *think* I have it now.  There were a couple more bugs, and I didn't really have a test for generics at all.

To fix the second part of this, I converted a "spare" set of EL1018 / EL2008 digital in/out modules to use the generic module, and I'm verifying that I can send and receive with them now as part of my normal test suite.  These tests are now passing.  So it's certainly possible that there are still bugs lurking here, but it's not completely broken anymore.

MetalMusings, 2024-01-12:
At least the new double works, so that's great. Both Tx and Rx works just fine. It is tested with byte-aligned doubles, it might still be the case that when a record of for example subIndex 1: 1 bit, subIndex 2: lreal, that the lreal isn't read/written right, that I didn't test.
I can test that too, will take a day or two.

MetalMusings, 2024-01-12:
I tested to add a mix of bit, real and lreal. It looks like there is padding automatically inserted after a bit. It may or may not be a general rule, I can in any case not generate a misaligned real/lreal. So this will work for all my cases.
```
  TxPDO 0x1a06 "Test-record"
    PDO entry 0x6006:01,  1 bit, "bit01"
    PDO entry 0x0000:00,  7 bit, ""
    PDO entry 0x6006:02,  1 bit, "bit02"
    PDO entry 0x0000:00,  7 bit, ""
    PDO entry 0x6006:03, 64 bit, "lreal01"
    PDO entry 0x6006:04,  1 bit, "bit03"
    PDO entry 0x0000:00,  7 bit, ""
    PDO entry 0x6006:05, 32 bit, "real01"

```
Thanks for the quick response to add this.

scottlaird, 2024-01-12:
Great!  I'm going to go ahead and close this for now, feel free to reopen/refile if there's anything else.

closed 2024-01-12 by scottlaird