linuxcnc-ethercat/linuxcnc-ethercat #128: Automatic Deployment of linuxcnc-ethercat at build.opensuse.org with the help of IgH

ighvh, 2024-01-06:
Hi folks, I'm Bjarne from IgH. We use the [OpenSuSE Build Service](https://build.opensuse.org/project/show/science:EtherLab) to distribute our EtherCAT Master. I added linuxcnc-ethercat to our repository, with Sascha's permission.

I see that the main work on this package takes place in this repo, so it would be a good thing to use this repo to distribute linuxcnc-ethercat. We should discuss the details in this issue.

We still need to add some distros at OBS:
 - [ ] Debian Testing
 - [ ] Debian Unstable
 - [ ] Raspbian 12 (not in official OBS repos, so we have to contact SuSE about that).

One important thing: In OBS, we use the upstreamed linuxcnc package. There is no possibility to add the Linuxcnc repositories. But the EtherCAT Master is available in Debian 10 and 11, too.

I see two possibilities:
 - Use [tar_sm](https://github.com/openSUSE/obs-service-tar_scm) OBS service to fetch code via git. This is our standard approach, but we need a quilt debian package format. All files in `debian/` need to be separate and `debian/changelog` cannot be updated automatically.
 - Prepare a debian source package (.dsc, .debian.tar.xz, .orig.tar.gz) in Release pipeline and fetching them with download_url OBS service.
 
In both ways, OBS is triggered by a webhook from GitHub and fetches the updated sources by [Source Service](https://en.opensuse.org/openSUSE:Build_Service_Concept_SourceService). We at IgH have a similar release process.

scottlaird, 2024-01-06:
I don't have strong preferences either way.  I've spent a bit of time trying to a half-hearted `.deb` release process working, and I'd be very happy to rip it all out and let you handle it.

I *think* the `tar_sm` version would be less work.  If I'm reading this right, I'll need to switch `debian/source/format` to `3.0 (quilt)`, along with whatever else that entails, and I'll need to fix the dependencies in `control` to reflect your current Ethercat package naming, rather than the ancient names in there.

ighvh, 2024-01-06:
I set up a [test repo](https://build.opensuse.org/package/show/home:ighvh:branches:science:EtherLab/linuxcnc-ethercat) in OBS, the `tar_scm` approach worked well. I even managed to reuse the files in `debian/`. But #121 and #130 are required to build clean debian packages. Also I learned that `tar_scm` is able to fetch the latest [tag on the default branch](https://github.com/openSUSE/obs-service-tar_scm/blob/e7c0883fcf4afb987947516640e0505a1d4bed1e/tar_scm.service.in#L148), which is quite handy to always ship the last released version.

So, the next steps would be adapting the package in science:EtherLab and implementing the webhook in GitHub.

scottlaird, 2024-01-06:
Great! I'll get to the other PRs in a few hours.

scottlaird, 2024-01-10:
FWIW, the build changes are in, but the webhook isn't yet.  That's next.

ighvh, 2024-01-11:
So, the repository is now switched. Unfortunately, Testing and Unstable have some issues at this moment, I hope they will be fixed soon:
```
[   31s] [199/262] installing systemd-255.2-3
[   31s] ******************************************************************************
[   31s] *
[   31s] * The systemd package cannot be installed because this system does
[   31s] * not have a merged /usr.
[   31s] *
[   31s] * Please install the usrmerge package to convert this system to merged-/usr.
[   31s] *
[   31s] * For more information please read https://wiki.debian.org/UsrMerge.
[   31s] *
[   31s] ******************************************************************************
[   31s] dpkg: error processing archive .init_b_cache/systemd.deb (--install):
[   31s]  new systemd package pre-installation script subprocess returned error exit status 1
[   31s] Errors were encountered while processing:
[   31s]  .init_b_cache/systemd.deb
[   31s] exit ...
[   31s] ### VM INTERACTION START ###
[   31s] [   23.072708][    T1] sysrq: Power Off
[   31s] [   23.073478][    T9] reboot: Power down
[   31s] ### VM INTERACTION END ###
[   31s] 
[   31s] h02-ch1d failed "build ethercat.dsc" at Thu Jan 11 08:14:03 UTC 2024.
[   31s] 
```

Scott, how do we add the webhook? Should I send you the credentials in an encrypted mail?

ighvh, 2024-01-11:
Testing and Unstable are now available in our OBS repos.

scottlaird, 2024-01-11:
email works, scott@sigkill.org is easiest.  I don't think I have a current GPG key anywhere, but I can probably throw something together quickly, unless you have a different preferred encryption setup.

scottlaird, 2024-01-12:
Here's a public key, from `gpg --armor --export $id`:

-----BEGIN PGP PUBLIC KEY BLOCK-----

mQGNBGWg0j8BDAC2FJUM2n3ELEUE1rM67bcvuSRTIFxA9GvJaXF/VwttgZzK9hF6
gxjhkoOhShPbXVW4Sm545fVHDGmkyT56NhGQqV/Wn/rIo1RG8ip9JkoOOIVMuSok
o/m5nAvGauLWoh7NmXWXgS/HhIJrQNkxi+5bQtjuJXtnDNE3taHhgAVCDsWRgZxq
8gq36Iet/up8ZsmwwV1JG53JjCzNE1UffQs0gsaf0tX4Gg6g9HYCgV4f2F+b39sk
LGX+pmBdj06UCnCozA9vAo5abDTy9XwqdOtbGvAFqEIU1B7Ki8q33lTdYoenF29O
nhdw4zsNkv2WRk3U9ti+s9GEdf5eT9JY930iAW4gjnQ+0fNdlfF3REuBMlNSKcxc
UbY2NtmrR/NSsoAFWUknq6bkxb+3BE+5MG884mDKjPruSGYgvBAVn2E6pOIbKIec
9SdAMVaIods9ICK+46cdEiDME02LnX9NGGVpb8WoCKIPakqMFOptQ8ueH1UAAQdt
RoUtQ3Nhmz4Zgv0AEQEAAbQfU2NvdHQgTGFpcmQgPHNjb3R0QHNpZ2tpbGwub3Jn
PokBzgQTAQoAOBYhBPFVLQ0ZN0Yaj+WRWqFzXLozTWHkBQJloNI/AhsDBQsJCAcC
BhUKCQgLAgQWAgMBAh4BAheAAAoJEKFzXLozTWHkLOIMAJ9WM9PByrJM57sCKkNl
OCuAj3sdflfxoyPqlnf50paQiXRLz7iEpvFBkKQvX+FHd/hg8IVC0Xo/j0pV5rFq
TRWaST2XoKiDHfpXC6uR9aYwn9k2sxlZfIA2947yme6OECgRcnQqYaQmq2Paf3FQ
xcdjDETevlHug6wnaBfIIDSGaVKvP0ZgpLfGVYjRQAX7O5U5j29BdKcQmZ0UpC0R
z4iG1/GPP7eHAAHayLVIbSL9rT+uzn1k2zFuP3bg5MU330VBrt1ioScifUf4FwLR
EwSeHdCU0Or8gdbhDN6/aDvqtwVHMPlaCDPDSop/aeSrQXPAC1OfbV5Vbu/qn7Sp
n9EKrj9d36Hv7SiNW0i/3cCwzOEzSvReSXB7R7wE56AL71QKpCT+SCgQ6qKI72oR
GnLVhn4SkGeaKq5nf9tdzh1lpax3M2TuNijM12DQ3Y2CaZdO/SzANxuesQECnNKC
L8OBKTpQ3yncVjsr0tKoZfDxNPihj827YskoG8cC6VFIVbkBjQRloNI/AQwAviqB
1++oMk9OCwJQLaTjiz4JLayVYCTm7vDLRo7aAC1/K9XBhk3DWDN7HgN1P45GMdNr
0SdrRIRXNIch0tuiDUpwEQHXjSq2fxpcBKgqTSNXRIEyKkzF7KoVP8EHADs4WQHE
sLJdgaZlG4VcqAqS9IIqzNCfJOgG2+uYwkOOh/gumjxzHidd7w+J59iPyEQ3ya7q
s9acuefPIKl8hjZMecdRds91zwJ+D6Ve9F7bFb2rS5f+3mU+FZ6cXrxi1NhT1SPs
EF2HeSJ8Hm42g/HBRICN839hOlQOPQgYCTbMKMe187hcZTYA2ks2VsSVIMw/Gn4l
hqOe+G8EpMZUWrDHnVOoI0w20FttXs3JL38G9rNEeOJeVFE0CncjlQnsYKywDy4V
0JJkYM4ynhRoNJ/FNwLWh/gJ6b90+kK41wSrVUlQGP4POFRDYrNWrExo1gcGgso1
zkgYxUmBSqzl0xB/oxNlh4rZRhSiig3m1NiilsX23pKoizAkNUleK1QGmNEfABEB
AAGJAbYEGAEKACAWIQTxVS0NGTdGGo/lkVqhc1y6M01h5AUCZaDSPwIbDAAKCRCh
c1y6M01h5L2uDACSYg/aqMdkFn7fpyW6Cu9HOYUDhvY5K6rQawwZDLH7FZppwL5t
cmQKO5M1uFz7eFjs2Chu80nNMdgZNL4c6lg2tTkWqPkXKAXoxhFpM7EbE1Tsr3iv
ikzdMojkZSjoemgkenKm6zr3tfQdmsTZYCU3g8LW3fMpdbnImPBPPpwn64lo7cYa
Bnr6WutzhNgc6C2IgiB9yymYE9FPYYPniREDv9Nl1qOMNMQITpGDK6QaXldZTky2
TGv9x8Bc7E0mTVjI0dZFE3/CUIptsRYVucYaaK0Kh7QIMAqNvpk1nZwMKk4loLfI
uNBmraNIVvOAgU7mJedD5o610fQux1pOCViZISAlT/HmO9o8+P04bRi+d75ke/iF
CH0qLTwFxK63JZhn7Y/8x4cpHwr1NQRYSoAi+VsiNAiujM05HDvwBz6MMVnOoSC9
8hyglyDcKoZ5l9mgSFyr+XlNC7HeTAbB8aJJgS/wasiLXUjdnyY0JWmZd0gDLISR
tuJ/WIl1iDYcJJw=
=yv5L
-----END PGP PUBLIC KEY BLOCK-----

scottlaird, 2024-01-12:
Thanks!  There was a minor speedbump, but it looks like the webhook is in place and builds are happening now.  I just cut a semi-fake v1.5.3, and it's already appeared on https://build.opensuse.org/package/show/science:EtherLab/linuxcnc-ethercat.  So I think we're good.

closed 2024-01-12 by scottlaird