linuxcnc-ethercat/linuxcnc-ethercat #134: Segfault on nonexisting slave type

jcedrych, 2024-01-09:
Hi, I'm excited that someone is continuing the project and making it easier to add more stuff. I have several cards with drivers written and would like to finally publish them. I will have to do a test with the new version first. Everything looks good so far, but I managed to make a typo in the configuration and lcec_conf crashed. The error always occurs when a non-existent type is entered. I think the error is in the lcec_findslavetype function, but I did not debug. Is this just my problem or is it common?

`<masters>
  <master idx="0" appTimePeriod="1000000" refClockSyncCycles="1">
    <slave idx="0" name="EK" type="EK1100"/>
    <slave idx="1" name="PWM" type="EL2502"/>
    <slave idx="2" name="S300-1" type="KMSS"/>
    <slave idx="3" name="S300-2" type="KMSSS"/>
  </master>
</masters>`

[225867.374819] lcec_conf[296647]: segfault at 0 ip 000055c4c45a5e42 sp 00007fffa2d6bce0 error 4 in lcec_conf[55c4c45a4000+18000] likely on CPU 0 (core 0, socket 0)
[225867.374829] Code: 00 00 00 49 8b 47 08 48 8d 35 83 6a 01 00 49 83 c7 10 48 89 44 24 08 e8 2c e4 ff ff 85 c0 75 d7 48 8b 7c 24 08 e8 aa 10 00 00 <48> 83 38 00 49 89 c5 75 18 48 8d 05 de 63 02 00 48 8b 4c 24 08 48

scottlaird, 2024-01-09:
It's probably a bug in new code.  Give me a minute.

scottlaird, 2024-01-09:
...er, more than a minute.  My test hardware is disassembled and being moved into an enclosure, and I'm having power problems.  I'll look at this tonight.

PR #138 (2024-01-10): fix(lcec_conf): Stop segfaulting on incorrect type= in XML
Oops.  Fixes #134
closed 2024-01-10 by scottlaird