linuxcnc-ethercat/linuxcnc-ethercat #135: Update epocat from FR1000 to FR4000

scottlaird, 2024-01-10:
See #132 and #133.  There are issues with the current code, and the FR1000 is out of date and out of production, with apparently no users.

scottlaird, 2024-01-10:
@abausano I opened an issue to track this, and I'm making the changes discussed on #132 and #133.  Can you verify that it still works and I didn't break anything?

abausano, 2024-01-10:
OK, I've asked it to one of our Customer, but he told me that apparently, in the last version of LinuxCNC is not possible to download the developer package with the tool to compile the components. Are You able to compile with the last version of LinuxCNC?

scottlaird, 2024-01-10:
The easiest way to do this is to download pre-build packages, but they're not *quite* ready yet.  Watch #128 for status.

I'm building against ~2 week old builds of LinuxCNC and Ethercat-Master right now in a couple environments and don't have any issues.

scottlaird, 2024-01-11:
Okay, it looks like that's moving faster than I'd expected.  If you look at https://build.opensuse.org/project/show/science:EtherLab, under linuxcnc-ethercat, you'll see a set of ~current packages.  I don't have directions handy, but this is the preferred way to install new builds of all of the Ethercat libraries as well as LinuxCNC-Ethercat.  I'll update the README here shortly, but there should be directions all over the LinuxCNC forums.

abausano, 2024-01-11:
It works perfectly, also the FR4000. Thank You again. One question: is the "generic" type implemented? Because my Customer told me that it didn't work, don't find it. But in the code, it seems to be present.

scottlaird, 2024-01-11:
I broke it by accident a week or so ago but no one noticed; you're the second person to point this out today.  It should be fixed in the latest release in Github, but it'll take a little bit before it works its way into the Debian packages.  Over the next few days the latency from Github->package repository should drop to ~minutes, but it hasn't happened yet.

abausano, 2024-01-11:
OK.

scottlaird, 2024-01-12:
It looks like the previous fix didn't work, so I spent a bit more time and actually got a test setup with generics working.  Versions >= 1.5.2 should work with generics.  Again, though, I'm not sure when the .deb for this will be produced, as it's still a manual process.

scottlaird, 2024-01-12:
It looks like we have builds working correctly now, and v1.5.3 is on https://build.opensuse.org/package/show/science:EtherLab/linuxcnc-ethercat.  I'm going to close this issue, feel free to reopen (or file a new issue) if there are still things to do.