linuxcnc-ethercat/linuxcnc-ethercat #167: Simplify EL5xxx drivers

scottlaird, 2024-01-13:
We have too many EL5xxx encoder drivers.  At a minimum, it'd be good to merge `el5101`, `el5151`, and `el5152`, as the 5101 and 5151 devices have identical PDOs and the 5152 is almost certainly just a 2-port version of the same thing.

This *may* be kind of painful, though, as the pin names for the el5101 and el5151 aren't identical.  It's not worth breaking existing configs just to make things consistent.

ThomasEg, 2024-01-25:
Please consider EL5112 aswell. I've got one in my setup and i'll be able to test it.

scottlaird, 2024-02-26:
A few notes:

- EL5001 r0: according to the ESI, this is some weird mutant with no PDOs.
- EL5001 r1: still a mutant, but less weird.
- EL5001 r1016+: encoder at `0x6010`, `:01` is data error, `:02` is frame errror, `:03` is power failure, `:11` is counter value.
- EL5002: a superset of the EL5001 r1016+.  It adds a second port at `0x6000`, which is reversed from what you'd expect.  r20+ add additional status bits at `:04`, `:0e`, `:0f`, and `:10:`.
- EL5021:  kind of like the '01, but it's at `0x6000`, not `0x6010`, the status pins are different, and it adds the latch value at `:12`.  r22 adds frequency at `:1b` and/or `13`.
- EL5031: yet another completely different beast.  It's at `0x6000`, and has either 32- or 64-bit counters in up to 49 objects at apparently `:11` and/or `:12`.  This may be a ESI parsing failure, probably best to refer to the manual.  In any case, it's not the same as any of the previous models.
- EL5032: not a 2-channel EL5031, either. If you squint hard enough, you can see the family resemblance, though.  The '32 doesn't include the array output from the '31, and it gains an extra status bit at `:01`.  Also, r28 adds a velocity object at `0x60x8:12`
- EL5042: Pretty much the same as the EL5032, without the velocity object.
- EL5072: Different from all of the above, again.  Position is at `0x60n1:01`, latch is at `:02`.  Status bits are on `0x60n0`.  Also includes a latch timestamp at `0x60n2:01`.
- EL5101 r0/1: 3 objects on `0x6000`.  Totally different from anything above.
- EL5101 r1017+: a superset of the r0/r1, with status values broken out in `0x6010` plus frequency/period/window values in `0x6000`.
- EL5102: similar to a 2-channel EL5101 r1017+, except all of the objects have different numbers.  Value/count went from `:02` to `:11`, the unified status object went away, latch went from `:03` to `:12`, frequency went from `:04` to `:13`, period from `:05` to `:14`, and there are a few new things available hiding in unmapped space.
- EL5112: very similar to an EL5102. 
- EL5122: very similar to an EL5102, but missing some features (no C pin, fewer latch options, etc).
- EL5131: very similar to an EL5102, but 1 channel and with fancy digital outputs.
- EL5151: similar to the EL5101, but maybe different status pins.
- EL5152: looks like a 2-channel EL5151.

So, we may be able to simplify here, but nowhere near as much as I'd hoped when I wrote this.

The good news is that the EL5112 will probably work with a 1-line change to the new EL5102 driver.

scottlaird, 2024-02-26:
To add a bit more data, here's what we currently support:

- EL5002
- EL5032
- EL5101
- EL5102 (new last week)
- EL5151
- EL5152

Those all have their own drivers.  We *might* be able to merge the EL5101, EL5151, and EL5152, if anyone is so inclined.  We can also add the EL5112 and maybe EL5122 to the EL5102 driver and the EL5042 to the EL5032 driver.

These devices *aren't* currently supported and would require some work:

- EL5001 (fairly cheap on eBay, FWIW)
- EL5021 (pricy on eBay)
- EL5031 (0 on eBay)
- EL5072 (0 on eBay)
- EL5131 (0 on eBay)

Notice that I'm *only* looking at the API side of things here.  The physical/electrical side of these devices vary widely.  Also, the maximum count rate also varies widely between devices, from 100 kHz to 5 MHz.

Also, some of these devices (like the EL5102 and EL5112) allow for switching between 5V/12V/24V and even single ended/differential via SDOs that we don't currently support.

To summarize: the EL5xxx devices are a wild grab bag of parts, without a whole lot in common.  We currently support 6 of 14 devices.  3 more would be fairly easy to add, leaving 5 that are unsupported.  4 of those are pricy and/or hard to get on eBay.

ThomasEg, 2024-07-17:
> The good news is that the EL5112 will probably work with a 1-line change to the new EL5102 driver.

I have one of those in my current setup. If you can make the change then i'd be happy to test it out and report back?

ThomasEg, 2024-12-22:
> To add a bit more data, here's what we currently support:
>
> * EL5002
> * EL5032
> * EL5101
> * EL5102 (new last week)
> * EL5151
> * EL5152
>
> Those all have their own drivers. We _might_ be able to merge the [...]
Can i do anything to nudge EL5112 closer to the top of the todo-list?

grandixximo, 2026-10-03:
Second ping (first was 2026-05-01). @ThomasEg - if you can post the EL5112 product code (from `ethercat slaves` or the label), a one-line typelist addition to `lcec_el5102.c` gets you support. The larger EL5101/5151/5152 consolidation stays on hold without demand.