linuxcnc-ethercat/linuxcnc-ethercat #175: Add Rexroth HCS01 indradrive Servo Controller

jc2ktr, 2024-01-13:


scottlaird, 2024-01-13:
Is there an existing HCS01 driver anywhere, or would someone need to start from scratch?

jc2ktr, 2024-01-14:
> Is there an existing HCS01 driver anywhere, or would someone need to start from scratch?

Hi scottlaird,

i´ve uploaded my generic Ethercat xml for the HCS01 based on the https://github.com/aschiffler/linuxcnc/blob/master/configs/roschi_cnc/ethercat_config.xml

from Andreas Schiffler.

i tried to adopt it to cia402 which is also supported by Rexroth but it works not as expected.

[hcs01_generic_xml][(url)](https://github.com/jc2ktr/linuxcnc_hcs01_generic_driver)

scottlaird, 2024-01-31:
Thanks.  Once #180 lands (maybe a week?) this will probably be quite a bit easier.

grandixximo, 2026-10-03:
Second ping (first was 2026-05-01, no reply). The Rexroth HCS01 is CiA 402, and the generic `basic_cia402` driver has shipped since the original request - there is a good chance a pure XML configuration covers your use case without a new driver. @jc2ktr - if you still have the drive, post your XML attempt and we will help tune it.

jc2ktr, 2026-10-04:
> Second ping (first was 2026-05-01, no reply). The Rexroth HCS01 is CiA 402, and the generic `basic_cia402` driver has shipped since the original request - there is a good chance a pure XML [...]

@grandixximo Sorry for the late answer, i am not regular on github. Yes i still have the drives in use, but i am still using them with the Older EtherCAT-SOE protocol. I don't have the knowledge in programming. But this was my last try.

[ethercat_rexroth_29012021.xml](https://github.com/user-attachments/files/33035498/ethercat_rexroth_29012021.xml)
[indradrive_pdo_config.xml](https://github.com/user-attachments/files/33035499/indradrive_pdo_config.xml)

grandixximo, 2026-10-05:
@jc2ktr thanks for posting the XMLs - that was exactly what was needed. Looking at them, your config is already standard CiA 402 (CSP, opmode 8), so no new driver is required.

One question before we invest effort here: which LinuxCNC version are you running, and are you willing to upgrade your setup to test a new configuration? The path forward would be the `basic_cia402` driver with a small XML plus the `cia402` HAL component (which we are about to ship in the package itself, see #534), replacing the generic driver and your hand-written controlword logic.

If your current SoE-based setup works and you would rather leave it alone, that is a perfectly reasonable answer too - in that case I would close this issue, since writing and validating a config nobody will test helps no one. Your call.

jc2ktr, 2026-10-05:
> [@jc2ktr](https://github.com/jc2ktr) thanks for posting the XMLs - that was exactly what was needed. Looking at them, your config is already standard CiA 402 (CSP, opmode 8), so no new driver is required.

However, I have a problem where I am not getting a reliable OP status for my drive using COE-XML, which is why I stuck with the SOE configuration.

> One question before we invest effort here: which LinuxCNC version are you running, and are you willing to upgrade your setup to test a new configuration? The path forward would be the `basic_cia402` [...]

I am using LinuxCNC 2.9.10 with the linuxcnc-ethercat [v1.46.0] on Kernel 6.12.111+deb13-rt-amd64 #1 SMP PREEMPT_RT.
I have 4 of the HCS01 Drives (3 on my CNC-Router and 1 on a Testbench). I could test another Configuration, on my Testbench.

> If your current SoE-based setup works and you would rather leave it alone, that is a perfectly reasonable answer too - in that case I would close this issue, since writing and validating a config nobody will test helps no one. Your call.

I think on my Productive Setup i'll stay on the SOE-Drives, and this issue can be closed, they work very Reliable, my intention with the cia402-driver were an easier setup.

grandixximo, 2026-10-06:
@jc2ktr v1.47.0 is out and ships the `cia402` HAL component in the package itself - no more halcompile of the external component. If you use the apt repository (https://linuxcnc-ethercat.github.io/apt/), it is a plain `apt update && apt upgrade` on your testbench machine.

One important note before you retry the CoE configuration: the first step is most probably in IndraWorks, not LinuxCNC. The HCS01 speaks either SoE or CoE, and the protocol is selected drive-side. Your working setup has the drives in SoE mode, and a generic PDO mapping does not care which protocol the drive speaks - but a CiA 402 configuration only works against a drive that is itself in CoE mode. If the testbench drive has never been switched over in IndraWorks, that is very likely why it never reached OP with your CoE XML.

If it still refuses OP after the switch, post the output of `ethercat slaves` and the dmesg lines from the attempt and we will dig in from there.