linuxcnc-ethercat/linuxcnc-ethercat #180: Add a "generic CiA402" driver for servos/steppers/etc

scottlaird, 2024-01-13:
As I understand it, a lot of newer (?) devices implement CiA402, which standardizes motion control over EtherCAT (and other CAN-family busses).  Having a semi-generic driver that *doesnt'* need custom XML would be great.

GuiHue, 2024-01-23:
I have some rintellent devices, a Hitachi WJ200 and Delta ASDA A2/A3 at my disposal. All of those can play CIA402.
Looking at those XML configs for the generic driver, I am afraid that while they share common PDOs they differ substantially for all the "other" features like DI/DO pins and alarm signals. Samples for CSP Mode at both Delta ASDA-A3 and rintellent ECT can be found here:
https://github.com/GuiHue/linuxcnctests/blob/main/ethercattest/ethercat-conf-A3CSP.xml

I therefore wonder if this makes sense as a driver would only cover the basic elements and everything else would be very ugly to configure (if possible at all).

It may be worth having a cia402-generic if users want to skip advanced features. What do you think? @scottlaird

scottlaird, 2024-01-23:
That makes sense.  

I'm trying to sort out in my head exactly what CiA 402 *is*--is it just the basic PDO spec, a model of what the control algorithm above the PDOs is supposed to look like, or what?

It'd be nice to have a copy of the CiA402 spec somewhere to refer to.  It looks like I could pay EUR 700 to join CiA for the year and get access, but there's probably some other route to legitimate access.  Half of these sorts of standards end up being included in *other* standards which are publically accessible, or at least cheaper.

Maybe https://www.ethercat.org/en/downloads/downloads_733ABB98E11545EA901D80D9A4CA7F80.htm works?  I haven't actually joined the EtherCAT group, but it's supposed to be free.  It also suggests that IEC 61800-7 contains CiA 402.  That's only CHF 375, which is cheaper (and doesn't want to renew annually).  But still more than I want to pay for something that I don't have a personal use for right now.

Hmm.  We probably actually want a `lcec_class_cia402.c` library that implements the basic CiA 401 PDOs.  Then a driver for a basic CiA 402 device would be dead simple.  It'd look like `lcec_el1xxx.c`, but without the big block of PIDs or any loops.  Just an `_init` function with a call to `lcec_cia402_register()` and a read and write function that call `lcec_cia402_read()` and `lcec_cia402_write()`.

Then adding digital ins/outs, etc, would be a simple matter of cloning a basic CiA 402 driver and adding a bit of additional code in `_init`, `_read`, and `_write`.

scottlaird, 2024-01-23:
I just renamed this issue to match the previous comment.

There should be a CiA 402 library, and a small number of device-specific drivers that use it and add din/dout/ain/aout/etc ports, or whatever else their hardware provides.

We'll probably want a generic CiA 402 driver, just as a basis for adding new device-specific drivers, but a tiny amount of work (letting people specify `vid` and `pid` on non-generic drivers) would probably make it at least somewhat usable.

scottlaird, 2024-01-23:
Note that adding din/dout/ain (once #231 lands) ports is basically trivial; I suspect that most people who can configure LinuxCNC's HAL could modify a sample CiA 402 driver to add ports at specific PDOs and contribute back a patch without a lot of work.

aout will show up eventually...

scottlaird, 2024-01-23:
To make the previous bit easier, the generic cia402 driver should probably be configured with 1 CiA 402 servo (but have room for >1 presumably?), and 0 din, 0 dout, 0 ain, and 0 aout, *but* have structures for the in/out ports allocated and read/write calls already in the _read/_write functions.  It'll burn a few extra cycles per loop (checking if a pointer is NULL), but I don't think we're *that* performance-critical.

Then all that needs to happen are calls to `lcec_din_allocate_channels()` for each type used, and a `lcec_din_register_channel()` per channel with the right PDO addresses.

scottlaird, 2024-01-23:
Or we could put `#define DIN_PORTS 0` at the top, then add an `if` in read/write based on that; the compiler would optimize it away entirely.  That'd let me add the `lcec_din_allocate_channels()` call in the generic driver as well, which would make adding ports one step less complex.

scottlaird, 2024-01-23:
One more question: what is the *cheapeast* CiA 402 device out there with a halfway sane implementation?  Maybe a Rtelligent stepper or something?

Probably the ECR60/ECT60?  Wow, $72 from Amazon, even, in the open-loop flavor.  $81 for closed-loop ECT60.  $129 for a kit with the ECT60, a stepper, and a cable.

GuiHue, 2024-01-23:
ECR or ECT60 with closed loop stepper (the open loop ones are harder to find (at least in EU). This is exactly what I currently have on the bench.
Edited previous comment to include correct spelling of the manufacturer.

scottlaird, 2024-01-23:
Thanks.  I ordered a ECT60 and stepper.  Hopefully I can find room for it in my increasingly crowded test box :-).

scottlaird, 2024-01-27:
My ECT60 arrived today, slightly earlier than the "mid March" estimate that was provided :-).  I'll see if I can whip something into shape.

scottlaird, 2024-01-27:
First question: how do multiple servos on a single CiA 402 device work?

Beckhoff makes a few dual-axis CiA 402 servo controllers in the AX8000 line; their crazily detailed PDF (https://download.beckhoff.com/download/document/motion/ax8xxx_objects.pdf) says that they put axis A on 0x6041 and axis B on 0x6841.  See page 2485 (!).  Other PDOs seem similar, offset by 0x800.

At the other end of the budget spectrum, the Oriental Motor AXD4A-KED 4-axis stepper driver uses 6041, 6841, 7041, and 7841.  So that seems like a reasonable model to assume.

We'll want to support single-axis naming schemes (`srv-foo`) and multi-axis naming (`srv-2-foo`).  I'd rather not have `srv-0-foo` on single-axis devices.

So, I'm going to make it possible to have multiple CiA 402 instances per device, at user-provided offsets.

scottlaird, 2024-01-27:
Also, I'm tempted to name the individual channels `axis` instead of `channel` in the code, but the correct English plural of that is `axes`, and that's just going to confuse non-native speakers.  And also native speakers.

scottlaird, 2024-01-27:
I mean, `register_axis` and `register_axes` are even worse than `register_channel` and `register_channels`.

GuiHue, 2024-01-27:
Appears sensible. 
I looked into starting a driver yesterday, but I have not fully worked out the structure of those nested drivers (meaning a class and a driver). I'm a bit far out of my comfort zone with stuff like this.

It may be worth noting: I have not come across a multi-axis application with CIA402 in the real of hobby applications (and thus linuxcnc).

scottlaird, 2024-01-28:
I have a general framework in place, but I hit an issue with sync mappings on the ETC60.  By default it only maps 3 inputs and 2 outputs:

```
$ ethercat pdos -p 22
SM0: PhysAddr 0x1000, DefaultSize   64, ControlRegister 0x26, Enable 1
SM1: PhysAddr 0x1080, DefaultSize   64, ControlRegister 0x22, Enable 1
SM2: PhysAddr 0x1100, DefaultSize    0, ControlRegister 0x64, Enable 1
  RxPDO 0x1600 "RXPDO 1 Mapping Parameter"
    PDO entry 0x6040:00, 16 bit, "Control Word"
    PDO entry 0x6060:00,  8 bit, "Modes of Operation"
    PDO entry 0x607a:00, 32 bit, "Target Position"
SM3: PhysAddr 0x1400, DefaultSize    0, ControlRegister 0x20, Enable 1
  TxPDO 0x1a00 "TXPDO 1 Mapping Parameter "
    PDO entry 0x6041:00, 16 bit, "Status Word"
    PDO entry 0x6061:00,  8 bit, "Modes of Operation Display"
    PDO entry 0x6064:00, 32 bit, "Position Actual Value"
    PDO entry 0x60fd:00, 32 bit, "Digital Inputs"
```

The problem is that I'm not actually sure what I want to map, and it's going to vary by device at least somewhat.  The CiA 402 driver and the device-specific driver will need to know about each other in order to share mapping details.  For instance, for the ETC60, I'm going to want to map the digital output pins, which aren't really part of CiA 402; whatever handles the mapping will need to handle CiA 402 PDOs and device-specific PDOs.  It looks like I'm going to need to set up sync managers on the fly, which I don't *think* we're doing anywhere else in LCEC, *and* that's going to lead to issues with the `pdo_entry_count` parameter in `lcec_typelist_t`, because we may not actually know the answer at compile time in all cases.

Also, trying to monkey with PDO settings via `ethercat download` mostly just spits out errors with the ETC60, but I'm going to bet that that's just a mistake on my part.

scottlaird, 2024-01-28:
Okay, it seems to load if I comment out the 2 PDOs that aren't mapped, although I haven't actually tested anything.  Next up, I'll try setting up a reasonable set of syncs and see what happens.

scottlaird, 2024-01-30:
...and adding a sync setup for the 3 missing PDOs (it's 3, not 2 actually) makes everything work.  Or, at least lets us start up without errors.

So, learning has occurred: if the hardware comes with PDOs/syncs defined that map all of the PDO entries that we care about, then we don't *need* to define PDOs and syncs in the driver.  Just doing `LCEC_PDO_INIT()` is good enough.  OTOH, if the hardware *doesn't* map everything that we care about, then we need to create a mapping.  

I *suspect* that we may want to un-map some of the PDOs and/or PDO entries, just to limit the number of things synced and increase performance, but I don't think that's a major issue for the most part with most drivers.

It's still unclear to me why we need multiple sync PDOs in most cases.  I guess if we're either mapping a huge number of PDO entries, or doing multiple domains, then we'd care, but for for 4-6 PDO entries?  Just do one in and one out and get on with life.  Right?

So, here's our issue as we stand right now: we need to be able to map some number of CiA 402 PDOs *plus* some number of hardware-dependent PDOs, *and* we're probably going to vary the number of CiA 402 PDOs depending on the hardware.  There's likely no point in mapping (for example) 0x60a2 ("jerk factor") on low-end devices like the ECT60, but maybe it'd be useful if we had a Beckhoff AX8000.

So, we have a semi-variable number of CiA 402 PDOs, plus an unknowable (from `lcec_class_cia402.c`'s perspective, at least) number of device-specific PDOs, but we need to set up mappings for all of them.  I see 3 approaches:

1. Define a fixed set of PDO mappings in cia402 code and assign that to the first PDO in/out in the sync, and let the device-specific entry register a second set of mappings.  We'd need a bit of glue code, but they'd be fixed-length and it'd be easy to put together. 
2. Define some way of building a set of mappings from lists of `ec_pdo_entry_info_t`s, and build them on the fly from one or more partial `ec_pdo_entry_info_t`s.  At a minimum, one from the `cia402` code and one from the device-specific code.  We'll *probably* end up needing this at some point because we'll overflow the max length for PDO entries in a single device, or we'll find a device that doesn't actually support a second PDO or something.  But the degenerate case for option 1 and option 2 is pretty close, actually.
3. Go whole-hog and build an LCEC-specific mapping layer that writes `ec_pdo_entry_info_t` and probably deals with `LCEC_PDO_INIT()` while it's at it.  Probably also replace the PDO count field in `lcec_typelist_t`, because it's incompatible with runtime detection of hardware or XML configuration of features.

I suspect we'll end up with option 3 at some point, but option 1 is dead simple and has a reasonable growth path into option 2, and that'll let me learn enough to make better judgements if we ever need option 3.

GuiHue, 2024-01-30:
Starting with 1 sounds like a good approach. I'm afraid this is too much for me to handle for the time being. My lack of C knowledge and driver related programming is affecting the usefulness of my work.

scottlaird, 2024-01-30:
No sweat, this is moving forward reasonably well at this point.  I just added a few RTelligent-specific diagnostic PDOs (voltage, current position, etc), and should be ready to give things a quick test later today or tomorrow.

ThomasEg, 2024-02-01:
I think i have a copy of the "CIA402 standard", but i don't think i'm allowed to post it here in public. It is not the latest version, but maybe it would be of use/interest for you anyways? Can i contact you somehow "privately" in order to send it to you?

scottlaird, 2024-02-03:
> I think i have a copy of the "CIA402 standard", but i don't think i'm allowed to post it here in public. It is not the latest version, but maybe it would be of use/interest for you anyways? Can i contact you somehow "privately" in order to send it to you?

If you have a pointer that you can send to scott@sigkill.org, then I can probably make use of it :-).

Thanks.

scottlaird, 2024-02-07:
[Status update:

*Almost* done with the first iteration through a CiA 402 driver.  There are still missing features, but not as many as you might expect.  My RTelligent ECT60 works in `velocity` mode but not in `position` mode, it's probably something trivial.  Digital in and out pins aren't working either, but I think it's more of a hardware thing than a software thing.  Or, quite likely, both.

Things are somewhat documented, but there are lots of TODOs still.  I probably have 2 days left on this, 1 day's debugging and 1 day's documenting.

I'm not really planning on adding any more pins or modParams in the current PR.  I think it supports everything that the CiA 402 HAL driver supports, so any additional features will need to be added there as well.  That seems like a good dividing line for when I should ship this.

I'm working on getting an Ethercat VFD (for `vl` mode) and a 2-axis stepper controller (from Leadshine, although RTelligent also makes some).  As things stand, we have about 80% of the code needed for multi-axis stepper controllers already in place, and the rest should be pretty easy.

Feel free to comment on the PR (https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/pull/251), and take a look at the code and documentation in there.

Once this is in, we'll want to discuss moving the 3 current drivers that seem to implement pieces of CiA 402 (dems300, omrg5, and deasda) to use the new framework.  This may be traumatic in a few places, but in general it's probably not too bad.  The one downside is that this driver *really* wants to use the CiA HAL component, and people who don't use it will probably end up with a broken config.  Maybe have a `dems300_new` type for now, for example?

(Not that it actually matters for the Delta MS300 -- it's a VFD, and we don't have enough support for VFDs yet to make it worth porting)

scottlaird, 2024-02-10:
I merged the first part of this last night, see https://forum.linuxcnc.org/ethercat/51652-cia-402-in-1-15-0 for some details.

There's still a fair bit of upcoming work.  From the forum post:

- Move a number of the generic CiA 402 modparams (profile velocity, profile acceleration, etc) to be pins instead.  They should have been pins in the first place.
- Figure out my digital in/out aren't working, then add generic digital in/out support into the CiA 402 library.
- Fix the mess that is "pdo_entry_count".  It's surprisingly hard to know how many PDO entries you're going to need to register ahead of time, as almost everything in the CiA 402 spec is optional.  At this point, I've added error reporting when a device requests N PDO entries, and usually just end up changing the count until the errors stop.  It's probably time to make it dynamic and delete "pdo_entry_count" across all drivers.
- Add homing
- Add interpolation mode
- Add vl mode
- Finish multi-axis support.  The framework supports up to 4 axes in theory, but there are a few details still missing, and I don't have any multi-axis hardware to test with.  Both Leadshine and RTelligent make dual-channel stepper drives; I'm hoping to pick up one of the Leadshine models after lunar new year.  Also, I should extend this to 8 axes, as per the actual spec.
- Look at migrating some of the semi-CiA 402 drivers in the tree to use the new CiA 402 framework.

DesertShedStudio, 2024-02-11:
@scottlaird If it would help with multi-axis CiA-402 support / development.  I can loan you one of my Rovix EDS-A6 Multi-axis EtherCAT drivers. (6-axis open loop 12 inputs 4 outputs). I don't currently have project for it, so it is just sitting here.  I did not get any documentation with it when I bought it, however I do have a email out with Rovix requesting a xml and manual. Just let me know, I am also in WA and could have it in the post first thing Monday.

scottlaird, 2024-02-11:
@DesertShedStudio, huh, that's an interesting beast.  A 24V 6-axis stepper controller?  Getting the basic stepper support working shouldn't be too bad, even without docs, probably.  Send me mail at scott@sigkill.org and we can arrange shipping, or if you're in the Seattle area we could probably just meet up someplace.  Thanks!

DesertShedStudio, 2024-02-11:
@scottlaird Email sent..

scottlaird, 2024-02-16:
I'm up to 5.5 PRs pending for this (#307 is *probably* just a proof-of-concept and won't get merged).  I still have a few more things left on the short list:

- Add pins for `tq`, `cst` modes.
- Add pins for `vl` mode.
- Figure out why digital ins/digital outs aren't working on my ECT60.
- Update `basic_cia402` with the flood of new pins.
- Update `rtec` with the new pins that it supports.  There shouldn't be very many.
- Figure out what digital in/digital out look like in `basic_cia402`.
- Figure out how to get PDO mapping to work correctly for multiple axes.  There will be some minor structural changes needed, but nothing major.  Right now, we're mapping PDOs on a per-axis basis, which doesn't make sense because PDOs are a slave-level entity.  It's fine for single-axis devices, but breaks for multiple axes.

I have one multi-axis device in hand (@DesertShedStudio's Rovix ESD-A6), but I'm still waiting on a few cables and connectors before I can plug it in.  I should probably also fire up Ye Olde Three Dee Printer to print a 5x NEMA 17 mount to hold the cheap steppers that go with it.

After that, I'm working to get my hands on a Leadshine 2CS3E-D507, and I'm waiting for an EtherCAT card for an Omron MX2 VFD to make it across the Pacific.

scottlaird, 2024-02-22:
- [ ] Add pins for `tq`, `cst`
- [ ] Add pins for `vl`
- [ ] Add modparams for enabling pins.
- [ ] Update `basic_cia402` with new pins (defaults)
- [ ] Update `rtec` with new pins
- [ ] Add digital in/out to `basic_cia402`
- [ ] Add multiple axis PDO mapping
- [ ] Add driver for Leadshine Ethercat stepper controllers
- [ ] Figure out what's up with my ECT60.  I think I killed it.
- [ ] Add driver for Rovix ESD-A6
- [ ] Add driver for Omron MX2
- [ ] Add driver for RTelligent ECR60x2 (mostly done)
- [ ] Add driver for RTelligent DRV series DC motor controllers
- [ ] Update modparam documentation
- [ ] Update pin documentation
- [ ] Write tool to run `ethercat slaves ; ethercat sdos` and turn that into an XML file, including all of the new `<modParam name="enableActualVoltage" value="true"/>` settings.  Should really be its own issue.  Maybe `lcec_xmlgen`?

scottlaird, 2024-02-22:
Semi-persistent note on hardware status:

- RTelligent ECT60 (@scottlaird): Present, seems to work except for possible hardware problems.  Waiting for replacement.
- Leadshine 2CS3E-D507 (@scottlaird): Hardware ETA 2024-02-24
- Rovix ESD-A6 (@scottlaird): On hand, untested
- Omron MX2 (@scottlaird): Hardware ETA 2024-02-26
- RTelligent ECR60X2 (@scottlaird): OTW no ETA
- RTelligent DRV400E (@scottlaird): OTW no ETA

Other:

- Yaskawa Sigma 7, etc.  Offered on forum for testing, maybe late March?
- Delta MS300: in tree with existing driver
- Delta ASDA: in tree with existing driver
- Omron G5: in tree with existing driver

scottlaird, 2024-02-22:
As for timing, I'd like to merge the giant pile of pending CiA 402 PRs at some point this weekend.  That should include the ~full set of CiA 402 objects from the 2008 draft that I have, covering all of the modes.  Multi-axis won't work yet and documentation will be spotty.  Then I'll start on new hardware after that and work on cleaning up, updating documentation, and fixing bugs in a new set of PRs.

Hopefully I can get things into a decent state before the beginning of March, because I'll be on the road for a big part of the month and my access to hardware, etc, will be limited.

scottlaird, 2024-02-28:
I broke a bunch of the items from this out into their own issues, and have PRs pending for 3 of them.

At this point, the next few actions are:

- Finish the remaining ~8 pins and modParams that are queued up to be added.  This should cover everything in the initial version of CiA 402 that seems important to us.  For now, I'm mostly omitting `ip` mode, as I can't see why we'd want to use it over CSP, etc.  Maybe on low-speed busses?
- Actually get the Leadshine 2CS3E-D507 working.  It inits fine with `basic_cia402` but I haven't got it to move yet.  I've put about 10 minutes into that in total.  I'll shift my focus to that soon.
- Update documentation.  The table at the end of `cia402.md` is up to date, but the rest of it is very stale.
- Create a driver for Leadshine stepper drives, forked from `basic_cia402`.  It'd add Leadshine-specific SDOs as well as hard-coding sane defaults for all of the `enable*` modParams.

scottlaird, 2024-02-28:
In short: we're getting closer.  Right now, we have almost 90k of code+comments in `lcec_class_cia402.[ch]`, make it far and away the largest driver in LinuxCNC-Ethercat.  I have a couple changes in mind that will probably shrink that a bit, but it's still huge.  This is derived from a 140 page PDF, and I think we have most of the core data model implemented at this point.  The biggest pain has been the extreme variability of CiA 402 devices and the flexibility in the standard.  There are *so* many objects, and only ~3 of them are actually required.

scottlaird, 2024-02-29:
I really need to figure out the right way to merge a stack of PRs with Graphite without triggering either N independent release runs or leaving the whole mess lost in a semi-discarded branch.  Over the weekend, I managed the latter, and this time I have the former.

Anyway, I think the last of the boilerplate just went in.  We should have all of the objects that I can see caring about implemented.  There's a chance that there are more typos on addresses or bit sizes, but everything that the Leadshine 2CS3E supports seems fine.  At this point we have:

- Digital in and out added to generic CiA 402.
- A few more pins.
- modparams for a bunch of "ratio" configs, like encoder resolution or steps.  CiA 402 implements these all as ratios with a numerator and a denominator, and you kind of need to provide both.  We support 3 different syntaxes here: "12/7", "12:7", and "1.71428".  All three of those should produce the same result; 12 in the numerator field and 7 in the denominator field.  Yes, even the float. 

I also cleaned up the huge piles of optional feature initialization code.  Annoyingly, the code didn't shrink but I found quite a few inconsistencies (bugs), so the result is better.  We're still just under 90k and just over 1900 lines, *not* counting `basic_cia402`.

scottlaird, 2024-03-24:
Okay, I'm back from travelling for a while, and I've mostly assembled the giant pile of test hardware that I've had sitting around.  I have one more step left, which will break things up into multiple busses and make some testing a bit easier.  But...

```
$ ethercat slaves
 0  0:0   PREOP  +  EK1100 EtherCAT-Koppler (2A E-Bus)
 1  0:1   PREOP  +  EL1018 8K. Dig. Eingang 24V, 10�s
 2  0:2   PREOP  +  EL1018 8K. Dig. Eingang 24V, 10�s
 3  0:3   PREOP  +  EL2008 8K. Dig. Ausgang 24V, 0.5A
 4  0:4   PREOP  +  EL2008 8K. Dig. Ausgang 24V, 0.5A
 5  0:5   PREOP  +  EL2084 4K. Dig. Ausgang 24V, 0.5A, negativ schaltend
 6  0:6   PREOP  +  EL2022 2Ch. Dig. Output 24V, 2A
 7  0:7   PREOP  +  EL2022 2K. Dig. Ausgang 24V, 2A
 8  0:8   PREOP  +  EL2034 4K. Dig. Ausgang 24V, 2A, Diagnose
 9  0:9   PREOP  +  EL2798 8K. Dig. Ausgang 24V AC/DC, 2A, potentialfrei
10  0:10  PREOP  +  EL3068 8K. Ana. Eingang 0-10V
11  0:11  PREOP  +  EL6001 Schnittstelle (RS232)
12  0:12  PREOP  +  EL4004 4K. Ana. Ausgang 0-10V, 12bit
13  0:13  PREOP  +  EL3681 Digital multimeter terminal
14  0:14  PREOP  +  EL3403-0010 3K. Leistungsmessung, 5A
15  0:15  PREOP  +  EL3204 4K. Ana. Eingang PT100 (RTD)
16  0:16  PREOP  +  EL9410 E-Bus Netzteilklemme (Diagnose)
17  0:17  PREOP  +  EL1859 8K. Dig. Eingang 24V, 3ms, 8K. Dig. Ausgang 24V, 0.5A
18  0:18  PREOP  +  EL4032 2K. Ana. Ausgang +/-10V, 12bit
19  0:19  PREOP  +  EL7041-1000 1K. Schrittmotor-Endstufe (50V, 5A, standard)
20  0:20  PREOP  +  EK1110 EtherCAT-Verl�ngerung
21  0:21  PREOP  +  EK1100 EtherCAT-Koppler (2A E-Bus)
22  0:22  PREOP  +  EL1018 8K. Dig. Eingang 24V, 10�s
23  0:23  PREOP  +  EL2008 8K. Dig. Ausgang 24V, 0.5A
24  0:24  PREOP  +  EL6751-0010 (CANopen Slave)
25  0:25  PREOP  +  2CL3-EC507(COE)
26  0:26  PREOP  +  0x00000227:0x00000005
27  0:27  PREOP  +  ECT60V202(COE)
28  0:28  PREOP  +  ECR60(COE)
29  0:29  PREOP  +  DRV400E(COE)
30  8:0   PREOP  +  3G3AX-MX2-ECT EtherCAT Communication Unit for 3G3MX2
31  8:1   PREOP  +  EP9214-0023 4Ch/4Ch Power Distribution for EtherCAT Box Modules
32  8:2   PREOP  +  EP3744-0041 5K. Absolutdruck 6K. Dig. Eingang 10us 2K. Dig. Aus
33  8:3   PREOP  +  EP2308-0002 4 K. Dig. Ein, 3ms, 4 K. Dig. Aus 24V, 0,5A, M12
34  8:4   PREOP  +  EP2338-0002 8 K. Dig. Ein-/Ausgang 24V, 0,5A, M12
35  8:5   PREOP  +  EP4174-0002 4K. Ana. Ausgang +/-10V, 0-10V, 0/4-20mA konfigurie
```

The interesting ones here are slaves 25-30:

- Leadshine 2CL3-EC507 (2 axis CL stepper drive)
- Rovix ESD-A6 6-axis OL stepper drive
- RTelligent ECT60 (CL stepper drive)
- RTelligent ECR60 (2-axis OL stepper drive)
- RTelligent DRV400E (1-axis DC servo drive)
- Omron MX2 VFD

scottlaird, 2024-03-24:
So, the next steps are:

- [ ] Configure the MX2 properly and get it working manually.  I have a ~1 kW spindle connected to a 2.2 kW VFD, but probably only have 1 kW available before I blow a breaker.
- [ ] Get the MX2 working semi-manually with a generic config.
- [ ] Get the MX2 working in VL mode with basic_cia402.
- [ ] Get both 2CL3 steppers turning (at all)
- [ ] Get the Rovix turning (at all).
- [ ] Get the ECT60 stepper turning (at all)
- [ ] Get the ECR60 steppers turning (at all)
- [ ] Get the DRV400E servo turning (at all)
- [ ] Get CSP working for the 2CL3. (Supports PP, PV, HM, CSP)
- [ ] Get CSP and CSV working for the ECT60. (Supports PP, PV, TQ, HM, CSP, CSV)
- [ ] Get CSP and CSV working for the ECR60 X2.  (Support unknown, but probably PP, PV, TQ, HM, CSP, CSV)
- [ ] Get CSP and CSV working for the DRV400E. (Supports PP, PV, TQ, HM, CSP, CSV, CST)
- [ ] Get CSP working for the Rovix. (Supports PP, HM, CSP)
- [ ] Get digital I/O working for each.

scottlaird, 2024-03-24:
Like I said, lots of motors.

[image]

scottlaird, 2024-03-24:
To be clear--my goal here is to get things working well enough to be usable, not to load-test anything or demonstrate the *right* way to mount things.  I'm expecting EM noise from the VFD to cause minor issues, and I'm using a regulated 48V power supply to feed steppers, which *probably* won't be a problem with no load on them, but it's not ideal from a back-EMF perspective.  I just want things to spin at low RPMs under software control.

scottlaird, 2024-03-27:
Note about the Omron MX2: if *none* of the config settings that the manual says to change are visible, then you probably need to change `b037` to `00` or `01`.  Mine was `04`, which wouldn't let me change (for example) basic motor parameters.

scottlaird, 2024-03-27:
I'm having a few issues with the MX2 and EtherCAT still:

```
LCEC: slave 0.omron-mx2: Invalid result size on SDO upload (0x6048:0x00, req: 4, res: 1)
.LCEC: slave 0.omron-mx2: Invalid result size on SDO upload (0x6049:0x00, req: 4, res: 1)
Failed to execute SDO upload: Value too large for defined data type
LCEC: slave 0.omron-mx2: Failed to execute SDO upload (0x6046:0x02, error -75, abort_code 00000021)
.Failed to execute SDO upload: Value too large for defined data type
LCEC: slave 0.omron-mx2: Failed to execute SDO upload (0x6046:0x01, error -75, abort_code 00000021)
```

These are all VL-mode params that were written blindly from the spec and have never seen hardware before.

- `0x6048` is "VL velocity acceleration".  I have it listed as a single U32, while the MX2 sees it as a ratio of a U32 in `:01` and a U16 in `:02`.
- `0x6049` is "VL velocity deceleration".  Exactly the same.
- `0x6046` is "VL velocity minimum".  I have it listed as a ratio of S16:S16.  The MX2 expects a U32:U32.

I'm also having issues with the Rovix:

```
[191031.899201] EtherCAT WARNING 0-26: PDOs configured for SM2, but slave does not provide the sync manager information!
[191031.899207] EtherCAT WARNING 0-26: PDOs configured for SM3, but slave does not provide the sync manager information!
[191031.899208] EtherCAT ERROR 0-26: Failed to determine PDO sync manager for FMMU!
```

I'll look into that after I get other devices happy.

scottlaird, 2024-03-27:
Okay, I mis-read 0x6046.  `:01` and `:02` are min and max, not a ratio.

These are fixed, but now I'm getting mapping errors:

```
[192429.715473] EtherCAT ERROR 0-30: SDO download 0x1600:03 (4 bytes) aborted.
[192429.715477] EtherCAT ERROR 0-30: SDO abort message 0x06090011: "Subindex does not exist".
[192429.715478] EtherCAT WARNING 0-30: Failed to map PDO entry 0x6042:00 (16 bit) to position 3.
[192429.715480] EtherCAT WARNING 0-30: Currently mapped PDO entries: 0x6040:00/16 0x6060:00/8 0x6042:00/16. Entries to map: 0x6040:00/16 0x6060:00/8 0x6042:00/16
[192429.715485] EtherCAT WARNING 0-30: Failed to configure mapping of PDO 0x1600.
[192429.750476] EtherCAT ERROR 0-30: SDO download 0x1A00:03 (4 bytes) aborted.
[192429.750478] EtherCAT ERROR 0-30: SDO abort message 0x06090011: "Subindex does not exist".
[192429.750479] EtherCAT WARNING 0-30: Failed to map PDO entry 0x6043:00 (16 bit) to position 3.
[192429.750480] EtherCAT WARNING 0-30: Currently mapped PDO entries: 0x6041:00/16 0x6044:00/16 0x6043:00/16 0x603F:00/16 0x6061:00/8. Entries to map: 0x6041:00/16 0x6044:00/16 0x6043:00/16 0x603F:00/16 0x6061:00/8
[192429.750487] EtherCAT WARNING 0-30: Failed to configure mapping of PDO 0x1A00.
[192429.766470] EtherCAT ERROR 0-30: Failed to set SAFEOP state, slave refused state change (PREOP + ERROR).
[192429.768473] EtherCAT ERROR 0-30: AL status message 0x0025: "Invalid Output Mapping".
```

I'm not sure what's going on here; both `0x6042:00` and `0x6043:00` appear in `ethercat sdos`.  We could live without 0x6043, but 0x6042 is utterly critical.  That's the speed that we want the VFD to run at.

I'm going to punt on this for tonight and come back after I've made progress on easier devices.

scottlaird, 2024-03-27:
Current status:

```
$ halcmd show | grep slave-oper
     8  bit   OUT          TRUE  lcec.0.ls-2cl3.slave-oper
     8  bit   OUT         FALSE  lcec.0.omron-mx2.slave-oper
     8  bit   OUT         FALSE  lcec.0.rovix-esd.slave-oper
     8  bit   OUT          TRUE  lcec.0.rt-drv400.slave-oper
     8  bit   OUT          TRUE  lcec.0.rt-ecr60x2.slave-oper
     8  bit   OUT          TRUE  lcec.0.rt-ect60.slave-oper
```

The 2cl3 is configured with 2 steppers, while the ecr60x2 only has 1 for now.  Off to try pv or pp modes...

scottlaird, 2024-03-27:
First problem:

```
1$ halcmd show | grep opmode
     8  s32   IN              8  lcec.0.ls-2cl3.srv-1-opmode
     8  s32   OUT             0  lcec.0.ls-2cl3.srv-1-opmode-display
     8  s32   IN              8  lcec.0.ls-2cl3.srv-2-opmode
     8  s32   OUT             0  lcec.0.ls-2cl3.srv-2-opmode-display
     8  s32   IN              2  lcec.0.omron-mx2.srv-opmode
     8  s32   OUT             0  lcec.0.omron-mx2.srv-opmode-display
     8  s32   IN              0  lcec.0.rovix-esd.srv-1-opmode
     8  s32   OUT             0  lcec.0.rovix-esd.srv-1-opmode-display
     8  s32   IN              0  lcec.0.rovix-esd.srv-2-opmode
     8  s32   OUT             0  lcec.0.rovix-esd.srv-2-opmode-display
     8  s32   IN              8  lcec.0.rt-drv400.srv-opmode
     8  s32   OUT             0  lcec.0.rt-drv400.srv-opmode-display
     8  s32   IN              8  lcec.0.rt-ecr60x2.srv-opmode
     8  s32   OUT             0  lcec.0.rt-ecr60x2.srv-opmode-display
     8  s32   IN              8  lcec.0.rt-ect60.srv-opmode
     8  s32   OUT             0  lcec.0.rt-ect60.srv-opmode-display
```

Notice that none of the `opmode-display` fields show a non-zero value?

However:

```
$ ethercat slaves | grep 2CL3
25  0:25  OP     +  2CL3-EC507(COE)
$ ethercat upload -p 25 0x6061 0
0x08 8
```

So that's a bug on our end.

scottlaird, 2024-03-27:
A quick look at the code *looks* ok.  The big issue is that `enable_opmode_display` needs to be set, but if that *isn't*, then the `opmode-display` pin won't be registered.  We see the pin, therefore it's enabled.

Next, I tried setting `*(data->opmode_display) = 123;` in `lcec_cia402_read` before the `FOR_ALL_READ_PDOS_DO(READ_OPT);` line.  `optmode-display` remained 0 for all devices.

Then I moved the extra line after `FOR_ALL_...()`, and all working devices now show an `opmode-display` of 123.

So, it's getting set to 0 explicitly.  Why?  This should just turn into a call to `EC_READ_S8(&pd[data->opmode_display_os])`.  I guess I can try that instead of `123`... and I'm still getting a 0.

So, what is `data->opmode_display_os`?  It should be a small-ish integer offset into `pd`.  But it's not, it's 0.  Which implies that something isn't being initialized right.

Okay -- we call `lcec_pdo_init` for all *writable* PDOs, but never call it for *readable* PDOs.  Oops.  That's going to cause problems.  Fixed and...

```
$ halcmd show | grep opmode
     8  s32   IN              8  lcec.0.ls-2cl3.srv-1-opmode
     8  s32   OUT             8  lcec.0.ls-2cl3.srv-1-opmode-display
     8  s32   IN              8  lcec.0.ls-2cl3.srv-2-opmode
     8  s32   OUT             8  lcec.0.ls-2cl3.srv-2-opmode-display
     8  s32   IN              2  lcec.0.omron-mx2.srv-opmode
     8  s32   OUT             0  lcec.0.omron-mx2.srv-opmode-display
     8  s32   IN              0  lcec.0.rovix-esd.srv-1-opmode
     8  s32   OUT             0  lcec.0.rovix-esd.srv-1-opmode-display
     8  s32   IN              0  lcec.0.rovix-esd.srv-2-opmode
     8  s32   OUT             0  lcec.0.rovix-esd.srv-2-opmode-display
     8  s32   IN              8  lcec.0.rt-drv400.srv-opmode
     8  s32   OUT             8  lcec.0.rt-drv400.srv-opmode-display
     8  s32   IN              8  lcec.0.rt-ecr60x2.srv-opmode
     8  s32   OUT             8  lcec.0.rt-ecr60x2.srv-opmode-display
     8  s32   IN              8  lcec.0.rt-ect60.srv-opmode
     8  s32   OUT             8  lcec.0.rt-ect60.srv-opmode-display
```

So, that's much nicer.

scottlaird, 2024-03-27:
That's all the time I have for tonight.  Hopefully I can get devices into PV or PP mode manually tomorrow and get all of the basic steppers running in one go.

scottlaird, 2024-03-27:
So, I *was* planning on testing with PV or PP modes, because I'd had problems with CSP before, but that was *probably* the read issue that was just fixed.  And nearly everything supports CSP:

```
$ halcmd show | grep supports-mode-csp
     8  bit   OUT          TRUE  lcec.0.ls-2cl3.srv-1-supports-mode-csp
     8  bit   OUT          TRUE  lcec.0.ls-2cl3.srv-2-supports-mode-csp
     8  bit   OUT         FALSE  lcec.0.omron-mx2.srv-supports-mode-csp
     8  bit   OUT          TRUE  lcec.0.rovix-esd.srv-1-supports-mode-csp
     8  bit   OUT          TRUE  lcec.0.rovix-esd.srv-2-supports-mode-csp
     8  bit   OUT          TRUE  lcec.0.rt-drv400.srv-supports-mode-csp
     8  bit   OUT          TRUE  lcec.0.rt-ecr60x2.srv-supports-mode-csp
     8  bit   OUT          TRUE  lcec.0.rt-ect60.srv-supports-mode-csp
```

Everything except the MX2.  CSV is a bit spottier; only the RTelligent devices support it.

All in all, this group of devices covers a lot of territory:

```
$ halcmd show | grep supports-mode | grep TRUE | cut -d. -f4 | sed 's/srv.*supports-//' | sort | uniq -c | sort -rn
      7 mode-pp
      7 mode-hm
      7 mode-csp
      5 mode-pv
      3 mode-tq
      3 mode-csv
      1 mode-vl
      1 mode-cst
```

scottlaird, 2024-03-27:
Note: `-actual-voltage` is currently a U32, which means that LC shows it in hex.  It's the measured voltage in 0.1V increments.  The right presentation would be as a float (divided by 10), but there's no simple way to do that today.  Failing that, switching to a S32 HAL pin would at least get the value in decimal, and it's unlikely that the 1-bit difference would really matter, unless someone's planning on running a multi-megavolt drive using LinuxCNC.

Most of the actual-* pins are somewhat similar.  The position-based ones (position, velocity, others?) are based on implementation-defined units, and the `cia402` HAL code uses them directly, but others may be better as floats.

I should really add a units column to the CiA 402 doc's giant table.

scottlaird, 2024-03-27:
Okay, CSP with the Leadshine 2CL3 still isn't working.  I'm not sure if I haven't provisioned it right (Need to set some `0x2xxx` SDOs?), or if I'm using `cia402` wrong.  More tonight.

scottlaird, 2024-03-27:
OTOH, CSV with the DRV 400 seems to work.  So that's something.

scottlaird, 2024-03-27:
And the DRV isn't working in CSP.  Which almost certainly means that I'm using it wrong.  I'm trying to write directly to `pos-cmd`, and it's not moving.  So it looks like I need to read docs and the `cia402` HAL code.  I'd rather not set up more motion controller infrastructure just to test motors.

scottlaird, 2024-03-27:
Posted https://forum.linuxcnc.org/ethercat/52165-i-need-a-simple-csp-example-for-cia-402#296917 on the LinuxCNC forum, seeing if anyone has a CSP config that they want to donate to the cause :-).  That may save me a day or two of debugging.

scottlaird, 2024-03-27:
Also, before I forget, a few things that need to be addressed before I consider the CiA 402 code to be done and (semi) stable:

- [ ] Rename some pins to be consistent.  We have both `foo_demand` and `demand_bar` in different places.  Pick an order.
- [ ] Consider types for "monitoring" pins.  Things like actual voltage and actual torque are probably better expressed as floats, but they're currently either U32 or S32.  U32 is especially bad because `show` shows the value in hex, which isn't all that useful to most people.
- [ ] Find a good user experience for setting up drives for the common case.  Right now, the mess of `enable*` modparams needed to get a working config is too big, even if we do automate it.  Ideally, we'd be able to say something like `useCSP` (distinct from `enableCSP`) and have it enable the full set of useful pins for CSP, up to the limit of the number of PDO entries that the hardware supports.  This differs from `enableCSP`, because that enables all of the *mandatory* CSP pins.
- [ ] The `cia402` HAL driver needs homing and VL support, at a minimum.  Maybe also CST.

scottlaird, 2024-03-30:
Made a bit of progress on CSP mode not working, but I'm not really any closer to an answer yet.

I tried using a `generic` config for one axis instead of `basic_cia`, using @rodw-au's config, and noticed that the axis moved slightly when powering on LinuxCNC, before erroring out with a following error.  During this time, the motor was actually moving slightly. 
 It looked like `target-position` was moving by 5, and `actual-position` was moving to match.  Switching back to the `basic_cia` driver continued the same (good!) behavior.

Unfortunately, after a few iterations, it stopped its move-and-error process.  LC stayed enabled, but any attempt to move the axis (via homing or jogging) resulted in a following error with 0 movement.

So, why did it work at first, what was it doing, and why did it stop?

Looking at the CiA 402 spec, the first figure in CSP mode shows that it takes 0x607a (target position), plus 0x60b0 (position offset), 0x60b1 (velocity offset), and 0x60b2 (torque offset).  This results in movement and updated values in 0x6064 (actual position), 0x606c (actual velocity), and 0x6077 (actual torque).

The following figure shows that 0x607a and 0x60b0 are compared to 0x607b and 0x607d (range limits), then multiplied by 0x607e (polarity) to feed into the "drive control function".  This also takes 0x6065 and 0x6066 (following error controls), 0x6080 (max motor speed), 0x60b1 (max velocity), 0x6085 (quick stop decel), 0x605a (quick stop option code), 0x6086 (motion profile type), 0x60c2 (interpolation time period), 0x60b2 (torque offset), 0x6076 (motor rated torque), and 0x6072 (max torque).

So, that means that these standard objects matter:

- 0x605a (quick stop option code) (value: 2)
- 0x6065 (following error) (not supported on ECT60)
- 0x6066 (following error) (not supported on ECT60)
- 0x6072 (motor max torque) (not supported on ECT60)
- 0x6075 (motor rated torque) (not supported on ECT60)
- 0x607a (target position) (varies)
- 0x607b (range limit) (not supported on ECT60)
- 0x607d (range limit) min=max=-1 (?)  *verify*
- 0x607e (polarity) (not supported on ECT60?  Odd)
- 0x6080 (max motor speed) (not supported on ECT60)
- 0x6085 (quick stop decel) (value: 500,000)
- 0x6086 (motion profile) (not supported on ECT60)
- 0x60b0 (position offset) (not supported on ECT60)
- 0x60b1 (velocity offset) (value: 0)
- 0x60b2 (torque offset) (value: 0)
- 0x60c2 (interpolation time) (value: :01 is 1, :02 is -3
 
In addition, we depend on a bunch of device-specific settings, but I wouldn't expect these to cause problems with CSP and not CSV.  We may also depend on input pins.  If one of them is set to be estop (or similar), and it's reading as positive, then I'd expect constant failures, although I wouldn't expect CSV to ever work, so that's weird.

Looking at these, 0x607d *might* be a problem.  It's set to -1 for both min and max, which might be special and might be bogus.  The spec says that the defaults are implementation-defined and that it's a 32-bit signed int.  If we were hitting range limits, then I'd expect a different failure mode than multi-second following errors, but maybe not?

The [ECT60  manual](http://www.rtelligent.net/upload/wenjian/Stepper/ECT%20Series%20User%20Manual.pdf) doesn't limit 0x607d at all.

I set it to -1000000, +1000000, and saw no difference when using `basic_cia402`.  However, `generic` seems to sort of work -- I was able to home from -950 to -387 before getting another following error, *but* the following errors were on axes 0 and 1, not 2.  So maybe that's working?  FWIW, the generic config sets `0x2000:00=1770` (6A peak current), `0x2007:06=05` (input 6 is estop), `0x2007:05=0` (input 5 is general use), `0x2011:0=0001` (closed loop mode).  Re-enabling the device and only jogging axis 2 seems to work fine with no errors.  So, good?

Flipping back to `basic_cia`, it doesn't move at all again.  So that's a clear difference between `generic` and `basic_cia`.  While using `basic_cia`, I ran `watch -n 0.1 ethercat upload -p 27 0x607a 0` and observed that target position is changing, so we're not failing to write.

scottlaird, 2024-03-30:
Continuing...

Running `watch -n 0.1 'ethercat upload -p 27 0x607a 0 ; ethercat upload -p 27 0x6040 0 ; ethercat upload -p 27 0x6041 0 ; ethercat upload -p 27 0x6060 0; ethercat upload -p 27 0x6061 0' to see what's actually happening under the hood w/ target position, controlword, statusword, opmode, and opmode_display.

With `basic_cia`, but LinuxCNC in `off` mode:
```
0x00000225 549
0x0004 4
0x1a50 6736
0x08 8
0x08 8
```

With `basic_cia`, but LC in `on` mode:
```
0x00000226 550
0x000f 15
0x1a37 6711
0x08 8
0x08 8
```

Then attempting to jog shows the first value changing but none of the others.  Seems reasonable.

Now, back to `generic`.  First, with LC running but not enabled:

```
0x00000225 549
0x0004 4
0x0a50 2640
0x08 8
0x08 8
```

Hmm.  Statusword is different.  We'll come back to that.

Now enabled:

```
0x00000225 549
0x000f 15
0x1637 5687
0x08 8
0x08 8
```

Also different.  Odd, because these values should be completely controlled via the `cia402` HAL module.

Finally, after seeking a bit:

```
0x0000034e 846
0x000f 15
0x1637 5687
0x08 8
0x08 8
```

So, statusword is weird.  With basic, we went from `0x1a50` to `0x1a37` when enabled, while `generic` went from `0x0a50 ` to `0x1637`.  So, let's decode those.

The low-order byte is always 0x50 when disabled and 0x37 when enabled.  0x50 = 0101_0000 and 0x37 = 0011_0111.  The first is is ready-to-switch-on=0, switch-on=0, operation-enabled=0, fault=0, voltage-enabled=1, quickstop=0, switch-on-disabled=1.  The second is operation-enabled/switch-on/ready-to-switch-on=1, fault=0, voltage enabled = 1, quickstop=1 (which is false; 0 means stopped due to quickstop).  Those both seem fine.

The next byte is 0x1a for `basic_cia` and 0x0a->0x16 for generic.

- 0x1a: 0001_1010
- 0x0a: 0000_1010
- 0x16: 0001_0110

I'm just going to pay attention to the bits that differ.  

- 0x10 (bit 12) is "oms" -- operation mode specific.  I'll have to look that up in a minute.
- 0x08 (bit 11) is "ila" -- internal limit active.
- 0x04 (bit 10) is "tr" -- target reached

So, ignoring the operation-mode difference, `basic_cia` says that it's hit an internal limit and hasn't reached its target, while `generic` says that  there's no limit and it's reached the target.  Which seems like a good description of what's visibly happening, but maybe doesn't tell me as much as I'd like.

Looking at the "oms" bit -- bit 12==0 is "target position ignored" and bit 12==1 is "target position shall be used as input to position control loop".  So not very exciting, although why does the idle value differ for `basic_cia`?

scottlaird, 2024-03-30:
I flipped back from `generic` to `basic_cia402`, and I'm still seeing the same results.  0x1a50 -> 0x1a37 when enabled.  Jogging changes the target location but doesn't change the status code.  Huh.

scottlaird, 2024-03-30:
Arghhhhh.

Found it, I think.  I'd screwed up my DC config for basic_cia.  I copied the `generic` DC config and now I can job with basic_cia.

Such a fun bug hunt :-(.

scottlaird, 2024-03-30:
I'm working on a doc that explains DC configs, because it was a mystery to me.  Also, posted https://forum.linuxcnc.org/ethercat/52200-default-dcconf-settings-for-cia-402-devices#297240 with a proposed solution for CiA devices.  Basically, for *known* devices (not generic, not `basic_cia402`), we'll add a default `<dcConf>` config that looks roughly like `<dcConf assignActivate="300" sync0Cycle="*1" sync0Shift="0" sync1Cycle="*1" sync1Shift="0"/>`.  The precise value of `assignActivate` needs to come from the per-device ESI file, so we'll need to code that into each driver, which is why we can't do this generically.  Normal values seem to be 0x3xx or 0x7xx, where xx are in (00, 10, 20, 30).

In general, it looks like I'm unblocked now, though.

scottlaird, 2024-03-31:
Something is still weird with the Rovix.  Since I have absolutely no docs on the thing, I'm not sure quite what's up, but it has issues with the way we're configuring PDOs and SMs.  Not really a high priority.

I think the next few steps are:

- [ ] Get Omron MX2 working (notes above)
- [ ] Clean up inconsistent pin names
- [ ] Update rtec driver to match basic_cia402
- [ ] Add `dcConf` to rtec driver
- [ ] Add 2-channel ECRx2 to rtec driver
- [ ] Test digital I/O with rtec driver.  It *probably* wasn't working before due to DC issues.

scottlaird, 2024-04-08:
I've also developed problems with the ECR60x2, it feels like DCs broke again, which doesn't make a lot of sense.  I also have a couple issues with the device; it doesn't support `ethercat sdos`, and it only seems to have 2 TX PDOs, vs 3 for the ECT60.  Having 2x as many objects and fewer PDOs isn't a great mix.

Fortunately, an ECT60x2 just arrived, so I can compare the two and see what it looks like.

scottlaird, 2024-04-11:
The ECT60x2 is here, but I'm not quite ready to test it yet, though.  I gave it a quick check, though, to see how it differs from the ECR60x2, besides the obvious encoder interfaces.

Good:
- `ethercat sdos` works, unlike the ECR60x2.

Bad:
- Still only 2 RX and 2 TX PDOs listed in `ethercat sdos`, and they're 0x1[6A]00 and 0x1[6A]10, not 0x1[6A]01.

scottlaird, 2024-04-11:
On a first pass, the ECT60x2 didn't work any better than the ECR60x2.  They both failed to actually *do* anything.

To minimize the differences between the ECT60 and ECT60x2, I created single-axis version of the ECT60x2 config and tried using that.  No luck, same error.  When looking at it via 'ethercat sdos`, the 0x16xx and 0x1cxx SDOs (PDO mappings) looked slightly weird, so I tried power-cycling the device to clear them, and now it works fine.  So it's possible that I'm breaking something as part of the init of the second channel.  Fortunately, I now seem to have a working and a non-working config, so it's hopefully not to hard to step through the differences between the two.

A quick review of PDO mappings from working and non-working configs doesn't show anything obviously exciting.

The 2-channel config still doesn't work.

Reverting back to the previously-working 1-channel config without power-cycling the ECT60x2 leaves it broken.

Power-cycling makes it work again.

Here's the diff between the SDOs between broken and working:

```
--- ect60x1-broken.txt  2024-04-10 19:07:04.310331487 -0700
+++ ect60x1-working.txt 2024-04-10 19:08:17.725811687 -0700
@@ -25,7 +25,7 @@
   0x10f1:01, rwrwrw, uint32, 32 bit, "Local Error Reaction"                        0x00000001 1
   0x10f1:02, rwrwrw, uint16, 16 bit, "Sync Error Counter Limit"                    0x0004 4
 SDO 0x10f8, "Timestamp Object"
-  0x10f8:00, rwrwrw, uint64, 64 bit, "Timestamp Object"                            0x0aa1cad118bcabe8 766116411041688552
+  0x10f8:00, rwrwrw, uint64, 64 bit, "Timestamp Object"                            0x0aa1cae235e3a1e6 766116484545225190
 SDO 0x1600, "RXPDO 1 Mapping Parameter"
   0x1600:00, rwrwrw, uint8, 8 bit, "SubIndex 000"                                  0x04 4
   0x1600:01, rwrwrw, uint32, 32 bit, "SubIndex 001"                                0x60400010 1614807056
@@ -43,9 +43,9 @@
 SDO 0x1610, "RXPDO 1 Mapping Parameter"
   0x1610:00, rwrwrw, uint8, 8 bit, "SubIndex 000"                                  0x04 4
   0x1610:01, rwrwrw, uint32, 32 bit, "SubIndex 001"                                0x68400010 1749024784
-  0x1610:02, rwrwrw, uint32, 32 bit, "SubIndex 002"                                0x68600008 1751121928
-  0x1610:03, rwrwrw, uint32, 32 bit, "SubIndex 003"                                0x687a0020 1752825888
-  0x1610:04, rwrwrw, uint32, 32 bit, "SubIndex 004"                                0x68fe0120 1761476896
+  0x1610:02, rwrwrw, uint32, 32 bit, "SubIndex 002"                                0x687a0020 1752825888
+  0x1610:03, rwrwrw, uint32, 32 bit, "SubIndex 003"                                0x68600008 1751121928
+  0x1610:04, rwrwrw, uint32, 32 bit, "SubIndex 004"                                0x00000008 8
   0x1610:05, rwrwrw, uint32, 32 bit, "SubIndex 005"
   0x1610:06, rwrwrw, uint32, 32 bit, "SubIndex 006"
   0x1610:07, rwrwrw, uint32, 32 bit, "SubIndex 007"
@@ -69,15 +69,15 @@
   0x1a00:0b, rwrwrw, uint32, 32 bit, "SubIndex 011"
   0x1a00:0c, rwrwrw, uint32, 32 bit, "SubIndex 012"
 SDO 0x1a10, "TXPDO 1 Mapping Parameter "
-  0x1a10:00, rwrwrw, uint8, 8 bit, "SubIndex 000"                                  0x08 8
-  0x1a10:01, rwrwrw, uint32, 32 bit, "SubIndex 001"                                0x68410010 1749090320
-  0x1a10:02, rwrwrw, uint32, 32 bit, "SubIndex 002"                                0x68f40020 1760821280
+  0x1a10:00, rwrwrw, uint8, 8 bit, "SubIndex 000"                                  0x06 6
+  0x1a10:01, rwrwrw, uint32, 32 bit, "SubIndex 001"                                0x683f0010 1748959248
+  0x1a10:02, rwrwrw, uint32, 32 bit, "SubIndex 002"                                0x68410010 1749090320
   0x1a10:03, rwrwrw, uint32, 32 bit, "SubIndex 003"                                0x68640020 1751384096
-  0x1a10:04, rwrwrw, uint32, 32 bit, "SubIndex 004"                                0x68770020 1752629280
-  0x1a10:05, rwrwrw, uint32, 32 bit, "SubIndex 005"                                0x686c0020 1751908384
-  0x1a10:06, rwrwrw, uint32, 32 bit, "SubIndex 006"                                0x683f0010 1748959248
-  0x1a10:07, rwrwrw, uint32, 32 bit, "SubIndex 007"                                0x68610008 1751187464
-  0x1a10:08, rwrwrw, uint32, 32 bit, "SubIndex 008"                                0x68fd0020 1761411104
+  0x1a10:04, rwrwrw, uint32, 32 bit, "SubIndex 004"                                0x68fd0020 1761411104
+  0x1a10:05, rwrwrw, uint32, 32 bit, "SubIndex 005"                                0x68610008 1751187464
+  0x1a10:06, rwrwrw, uint32, 32 bit, "SubIndex 006"                                0x00000008 8
+  0x1a10:07, rwrwrw, uint32, 32 bit, "SubIndex 007"
+  0x1a10:08, rwrwrw, uint32, 32 bit, "SubIndex 008"
   0x1a10:09, rwrwrw, uint32, 32 bit, "SubIndex 009"
   0x1a10:0a, rwrwrw, uint32, 32 bit, "SubIndex 010"
   0x1a10:0b, rwrwrw, uint32, 32 bit, "SubIndex 011"
@@ -187,7 +187,7 @@
   0x2007:03, rwrwrw, uint16, 16 bit, "IN3 function"                                0x0003 3
   0x2007:04, rwrwrw, uint16, 16 bit, "IN4 function"                                0x0006 6
 SDO 0x2008, "Inputs Polarity"
-  0x2008:00, rwrwrw, uint16, 16 bit, "Inputs Polarity"                             0x0000 0
+  0x2008:00, rwrwrw, uint16, 16 bit, "Inputs Polarity"                             0x003f 63
 SDO 0x2009, "Filter Time"
[... 39 lines cut ...]
   0x6077:00, r-r-r-, int16, 16 bit, "Torque Actual Value"                          0x0000 0
[... cut ...]
   0x68f4:00, r-r-r-, int32, 32 bit, "Following error actual value"                 0x00000000 0
 SDO 0x68fd, "Digital Inputs"
-  0x68fd:00, r-r-r-, uint32, 32 bit, "Digital Inputs"                              0x00000007 7
+  0x68fd:00, r-r-r-, uint32, 32 bit, "Digital Inputs"                              0x00000000 0
 SDO 0x68fe, "Digital Output"
   0x68fe:00, r-r-r-, uint8, 8 bit, "SubIndex 000"                                  0x02 2
   0x68fe:01, rwrwrw, uint32, 32 bit, "Physical Outputs "                           0x00000000 0

[32 more replies]