linuxcnc-ethercat/linuxcnc-ethercat #184: lcec: dlopen: /usr/lib/linuxcnc/modules/lcec.so: undefined symbol: hal_export_funct

ighvh, 2024-01-16:
I started a [small blinking test](https://gitlab.com/igh-vh/linuxcnc_ec_blinking/) but I couldn't start it:
```
$ halrun -I -f blink.hal
rtapi_shmem_new failed due to shmget(key=0x48484c34): Permission denied
rtapi_init: could not open shared memory for uuid
HAL_LIB: ERROR: rtapi init failed
hal_lib: rtapi_app_main: Invalid argument (-22)
iopl() failed: Operation not permitted
cannot gain I/O privileges - forgot 'sudo make setuid' or using secure boot? -parallel port access is not allowed
Note: Using POSIX non-realtime
lcec: dlopen: /usr/lib/linuxcnc/modules/lcec.so: undefined symbol: hal_export_funct
blink.hal:3: waitpid failed /usr/bin/rtapi_app lcec
blink.hal:3: /usr/bin/rtapi_app exited without becoming ready
blink.hal:3: insmod for lcec failed, returned -1
halcmd: objdshmctl IPC_LOCK: Cannot allocate memory
shared memory segment not locked as requested
iopl() failed: Operation not permitted
cannot gain I/O privileges - forgot 'sudo make setuid' or using secure boot? -parallel port access is not allowed
Note: Using POSIX non-realtime
```

I installed linuxcnc and linuxcnc-ethercat from the upstream repos.
```
$ dpkg -l | grep linuxcnc
ii  linuxcnc-doc-en                      2.9.0~pre1+git20230208.f1270d6ed7-1 all          motion controller for CNC machines and robots (English documentation)
ii  linuxcnc-ethercat                    1.6.0.ge9b56ec-0                    amd64        LinuxCNC EtherCAT HAL driver
ii  linuxcnc-uspace                      2.9.0~pre1+git20230208.f1270d6ed7-1 amd64        motion controller for CNC machines and robots
ii  linuxcnc-uspace-dev                  2.9.0~pre1+git20230208.f1270d6ed7-1 amd64        PC based motion controller for real-time Linux
```

Also happens when installing linuxcnc-ethercat (1.6.0) from source. Any ideas?

lcec.so seems to be linked correctly:
```
ldd /usr/lib/linuxcnc/modules/lcec.so 
        linux-vdso.so.1 (0x00007ffe217b6000)
        liblinuxcnchal.so.0 => /lib/liblinuxcnchal.so.0 (0x00007faf31b59000)
        libethercat.so.1 => /lib/x86_64-linux-gnu/libethercat.so.1 (0x00007faf31b3e000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007faf3195d000)
        /lib64/ld-linux-x86-64.so.2 (0x00007faf31b9f000)
```

But liblinuxcnchal does not contain `hal_export_funct`.

ighvh, 2024-01-16:
Ah, this time I ran linuxcnc in a docker container with an openSUSE host. I will try again in a virtual Machine.

ighvh, 2024-01-16:
I installed Debian 12 on a physical machine and it works very well. Sorry for the noise.

closed 2024-01-16 by ighvh

scottlaird, 2024-01-16:
I'm curious how that symbol was left out of the installed version.  Not that it really matters, I guess.

Your "blink test" gets more useful if you add an EL1xxx and wire an output of the EL2xxx to the EL1xxx, so you can read it directly.  I'm slowly working through a pile of ELxxxx hardware, cross-wiring them, and adding tests in https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/master/tests/haltest.sh (and the `.hal` and `.xml` files in the same directory).  I bought a block of random ELxxxx devices off of eBay a few years ago, including a zillion EL1xxx and EL2xxx devices for practically nothing, so this is cheap for me.

The latest addition is having 2x EL1xxx and 2x EL2xxx, and using `<slave type="generic">` for one of each.  Then I can verify that generic inputs and outputs work as well.  Since I killed `generic` during refactoring and didn't notice, this is good to have :-).

Next up will be EL3xxx and EL4xxx, doing the same thing with analog, and an EL7041 driving a small stepper with an encoder, verifying that the encoder shows movement.  I'm debating picking up a second encoder and an EL5xxx, just for redundancy there (since the EL7041 has a "lie about the encoder" mode), but used EL5xxxs are too expensive right now.

scottlaird, 2024-01-16:
I'd really *like* to run my test suite as a pre-commit check, but that opens a ton of security risks w/ Github, and I haven't had the time to mitigate any of them yet.  Right now, my test hardware is just sitting on the floor of my office wired into my home network; at an absolute minimum I need to move it to its own isolated network and add some firewall rules.

See https://johnstawinski.com/2024/01/11/playing-with-fire-how-we-executed-a-critical-supply-chain-attack-on-pytorch/ for a partial discussion of the issues.

Also: I have a Raspberry Pi 4 with a couple more ELxxxx modules for doing ARM64 tests, but I haven't even installed an OS on it yet.