linuxcnc-ethercat/linuxcnc-ethercat #199: Add Support for Delta ASD-A3 series drives (new PID)

GuiHue, 2024-01-18:
modify existing deasda without breaking legacy to add PID for A3 (PR to follow)

thomam04, 2024-01-18:
It would be great if we can have modparam or flags? for this driver, as the Delta A2 and A3 are a bit different, besides the basic PDOs. 

Delta A2 motor 20-bit resolution (1280000 pulses/revolution)
Delta A3 motor 24-bit (16777216 pulses/revolution)

There are also some enhancements I can think of that require additional PDOs for the A3 and those might not be available for the A2 or the mapping is different. 

Not sure if it makes sense to create two drivers or try to combine everything into a single driver?

GuiHue, 2024-01-18:
@thomam04  has a great point that stupid me missed. The resolution must became a variable for setup as a3 drivers can be used with a2 and a3 series servos. Will see to that.

GuiHue, 2024-01-25:
Closed with #200

closed 2024-01-25 by GuiHue