linuxcnc-ethercat/linuxcnc-ethercat #201: Add Support for CSP mode for Delta ASDA Drives

GuiHue, 2024-01-18:
Currently, the DeASDA driver uses CSV (continuous sync. velcocity) mode only. This is fine when using external encoders, but not great when not using them. Current version requires closing the position loop in linuxcnc based on the encoder data provided by the drive through ethercat. This requires PID tuning in linuxcnc and cannot be a better way than just have the drive do its thing. Accordingly, having CSP mode would be great.

This is done by setting 6060h to 8 (CSP) instead of 9 (CSV).

The responsible lines are:
```
  // set to cyclic synchronous velocity mode
  if (ecrt_slave_config_sdo8(slave->config, 0x6060, 0x00, 9) != 0) {
    rtapi_print_msg (RTAPI_MSG_ERR, LCEC_MSG_PFX "fail to configure slave %s.%s sdo velo mode\n", master->name, slave->name);
  }
```
This is NOT to be changed during runtime as the drive will likely go haywire. Furthermore, major hal changes are required anyways. I assume that it therefore makes sense to pass a parameter from the XML file instead of using a halpin. 

Question: Is there a way to pass an additional parameter in the slave entry in the ethercat-conf.xml? Basically, I'd imagine an additional tag such as "mode" in: <slave idx="0" type="DeASDA3" name="A3" />

Is this possible?

GuiHue, 2024-01-18:
In addition to my own comment: Changing the mode requires changes to a number of further things within the driver which all would have to be switched (e.g. the write cycle, the exposed hal pins).

Is it maybe better to have DeASDA and DeASDA-CSP as drivers?

scottlaird, 2024-01-18:
In general, this is what `<modParam>`s are for.  They let you add semi-arbitrary key-value sets in the XML that get passed through to the driver.

That doesn't answer the "does this fit in a single driver" question, though.

GuiHue, 2024-01-20:
With the input on #199 I have made some progress. I now have a basic working CSP mode driver for DEASDA (tested only on A3, according to the manual they should perform the same).
Since I have not written more than a few very basic lines of C in my life, progress is a bit slow and painful, but I am getting there.

With regard to the does this fit a single driver question:
I think it does and will aim accordingly. The ambition is to have A2 and A3 series drives in one driver and being able to switch between the two modes CSV and CSP at INIT. I will work with modparam and a couple of other helpers. The driver file becomes a lot longer (lines of code), since declarations, read and write functions need to be split and init needs to handle a couple of cases. The actual write and read remain lean so it should be ok. 
PR will take some more hairpulling;).
I have started an *.md for it as well.

GuiHue, 2024-01-25:
Closed with merge of fixes into master. #200 
https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/pull/200

closed 2024-01-25 by GuiHue