linuxcnc-ethercat/linuxcnc-ethercat #258: Beckhoff EL3102: ERROR 0 0:11: PDO entry 0x6000:11 is not mapped

thomam04, 2024-01-30:
With release 1.13.1 I am getting an: "ERROR 0 0:11: PDO entry 0x6000:11 is not mapped" with my EL3102.

[Jan30 19:53] EtherCAT: Requesting master 0...
[ +0.000002] EtherCAT: Successfully requested master 0.
[ +0.384079] EtherCAT ERROR 0 0:1: PDO entry 0x7000:11 is not mapped.
[ +0.000026] EtherCAT 0: Releasing master...
[ +0.000035] EtherCAT 0: Released.

scottlaird, 2024-01-31:
Can you send a copy of `ethercat pdos -p 11` when LinuxCNC isn't running?  Everything that I can see makes it look like 0x6000:11 should be mapped by default.  It shouldn't be too hard to fix, but I don't understand why it needs to be fixed.

scottlaird, 2024-01-31:
Also, was this working before, or is this the first attempt with this hardware?

thomam04, 2024-01-31:
Yes, it was working previously. 

```
cnc@lcnc:~/ethercat$ ethercat pdos -p 11
SM0: PhysAddr 0x1800, DefaultSize  246, ControlRegister 0x26, Enable 1
SM1: PhysAddr 0x18f6, DefaultSize  246, ControlRegister 0x22, Enable 1
SM2: PhysAddr 0x1000, DefaultSize    0, ControlRegister 0x24, Enable 1
SM3: PhysAddr 0x1100, DefaultSize    0, ControlRegister 0x20, Enable 1
  TxPDO 0x1a00 "TxPDO 001 mapping"
    PDO entry 0x3101:01,  8 bit, "Status"
    PDO entry 0x3101:02, 16 bit, "Value"
  TxPDO 0x1a01 "TxPDO 002 mapping"
    PDO entry 0x3102:01,  8 bit, "Status"
    PDO entry 0x3102:02, 16 bit, "Value"
```

scottlaird, 2024-01-31:
Okay, that should be fixed now.  Sorry about that, the EL31x2 is *really* weird, and is probably one of the earliest EtherCAT devices.  It's odd that the EL31x1 and EL31x4 are "normal" but the EL31x2 isn't.  I should have caught this in the code.

thomam04, 2024-01-31:
Thanks Scott. All working now!

PR #261 (2024-01-31): fix(el31x2): revert EL31x2 to old driver
It turns out that the EL31x2 devices *aren't* the same as the rest of the EL3xxx family.   They use totally different PDOs.

Fixes #258
closed 2024-01-31 by scottlaird