linuxcnc-ethercat/linuxcnc-ethercat #274: Improve error messages

scottlaird, 2024-02-01:
Our error messages when we fail to start up aren't very clear today; they largely tell devs where things went wrong without providing much user guidance.  Audit error messages and improve the user experience where possible.

GuiHue, 2024-02-02:
To me this raises another point. The error messages are handled by rtapi which is fine, but that is generally set up so that it only displays the actual errors. One can change this in code using rtapi_set_msg_level https://linuxcnc.org/docs/stable/html/man/man3/rtapi_set_msg_level.3rtapi.html but I consider this to be a cumbersome affair for someone working with linuxcnc-ethercat instead of building.
Is there a simple way we can recommend users to increase the rtapi_set_msg_level? Maybe as a flag during the load process of the driver?
(I really hope that such a thing doesn't exist yet and I make a fool of myself).

scottlaird, 2024-02-03:
Maybe `halcmd debug`?