linuxcnc-ethercat/linuxcnc-ethercat #286: lcec_deasda Delta ASD setting E-gear

zmrdko, 2024-02-06:
Would it be possible and worth it to make a parameter(s) for setting E-gear ratio via SDO?

[image]

scottlaird, 2024-02-06:
Is this just an SDO setting that needs to be set before starting, or is this something that gets modified from time to time?

It'd probably be best to add this to the (new, not yet merged) CiA 402 library, and then use that in `lcec_deasda.c`, but that's a non-trivial amount of work and will probably end up requiring changes to people's `.hal` files as well.

zmrdko, 2024-02-06:
I have Delta B3 EtherCAT drives and there is parameter P3.012 which gives two options. Either choose to have default values for certain parameters or use stored values for parameters. And E-gear is one of them. I have to use E-gearing due to having long axis and I overflown 32bit integer value for position in LinuxCNC. I liked idea of having it configured via HAL/ini. I can have a look at the code and try to come up with something simple.

zmrdko, 2024-02-06:
And I am not sure, if E-gearing is part of the cia402 standard.

GuiHue, 2024-02-06:
In my opinion: Egear appears to be something that is highly specific to the vendor and should be handled through their configuration interface. To me its similar to setting up the direction of rotation and servo tuning and thus not be handled from linuxcnc.

thomam04, 2024-02-06:
I would agree with @GuiHue, E-gearing should be pre-configured in the servo amp during setup / tuning or via the configuration tool.

zmrdko, 2024-02-07:
using sdoConfig is sufficient for setting E-gear

scottlaird, 2024-02-25:
FWIW, the e-gear (and 4 or 5 other ratio settings) are on my short list for near-term CiA 402 changes.

Since you're playing with this, I'd like to run my rough plan by you guys and see if you have any comments or objections.

In general, there are 3 ways to configure device settings with LinuxCNC-Ethercat:

1.  Add an sdoConfig to the XML.  Completely generic, very flexible, but not user-friendly.  Also, it can only be changed at startup.
2. Create a modParam in the XML file.  More user-friendly, users don't have to look up SDO object numbers and formats, but they're still fixed after startup.  As a plus, we can do unit conversion, etc, here.
3.  Create a new HAL pin.  Can change without restarting, but incurs a small cost to servo cycle time, and odds are a lot of devices have more SDO objects than we could actually map and update via PDOs at runtime.

For CiA 402 settings, I keep going back and forth between modParams and pins for a lot of things.  I started out making lots of modParams, but I've moved a lot of them to be pins over time, because they're not really much more difficult to code and they're more discoverable and generally useful.

I'm leaning towards making a pile of numerator/denominator modParams right now, for:

- vl velocity quick-stop speed (604a)
- vl velocity set-point (604b)
- vl velocity dimension (604c)
- encoder resolution (608f)
- velocity encoder resolution (6090)
- gear ratio (6091)
- feed ratio (6092)
- e-gear ratio (6093)

These are ~all implemented in 2 parts, with a s32/u32 numerator and a u16/u32 denominator (they vary a bit).  I was thinking about implementing these as modParams that take strings, and then doing a bit of cleverness under the hood.  So, you could say `<modParam name="egear" value="2.5"/>`, or `<modParam name="egear" value="31/17"/>`.  If you provide a single number, then it's interpreted as a floating point number, and we pick a numerator and denominator to get as close as possible.  And if you provide two numbers with a `/` between them, then they're set directly.

Would that work okay?  Are there any reasonable use cases where we're want to change any of these on the fly, or where this setup would be confusing?

pippin88, 2024-02-25:
I am trying to understand how the scaling works for deasda.

Delta A3 servos.
Using deasda driver
metric / mm setup
5mm pitch ball screws

setp %s.%s.%s.srv-pulses-per-rev 16,777,216
Results in 1 revolution of servo = 1mm on linuxcnc DRO
(or 300mm/min jogging = 300rpm)

But I need to scale for 5mm pitch ball screws.

Tried:
setp %s.%s.%s.srv-pulses-per-rev 3355443.2
(16,777,216 / 5)
But it won't take a floating point / decimal number

Using 3355443 (dropping the .2) does work, but obviously introduces some inaccuracy.

I will be using the full closed loop function of the A3 servos and it seems to want to set egearing to 1:1 for that, but I could be wrong.

zmrdko, 2024-02-26:
Thanks for coming back @scottlaird!
Since its E-gear - I don't think its necessary to change it on the fly. Using modParam is totally sufficient.
I would not worry about calculations, just set one value according to motor specs (1280000 or 16777216) and the other value which would be directly parameter as pulses per revolution. This would apply for direct mount. If you would want to be extra nice, you can add secondary gearing (for belts coupler lets say). @pippin88  you can do easy math here: for 5mm pitch ballscrew I would throw 5000 pulses per revolution that would split one millimeter of your axis travel to 1000 pulses.
If you would need anything else, let me know..

zmrdko, 2024-02-26:
i know gearing is nice to specify with ratio, but i think its more convenient to specify gearing via desired pulses per revolution, and let the servo drive do the math..
Let me put it this way: If I specify <modParam name="egear" value="2.5"/> how many pulses per revolution would it be?
I think it will be 5,510,886.4 pulses per revolution, which is not very convenient and you have to do rounding. since you have to send integers to nominator and denominator parameters.
Another issue would be axis with long travel, which would overflow encoder position, which is 32bit integer in linuxcnc after some distance. Thats why you want to set E-gearing by specifying pulses per revolution, so that you can scale it down to lets say 500k pulses per revolution instead of 5.5M..

scottlaird, 2024-02-26:
Part of the problem is that different CiA 402 hardware uses different "base" units, as (at least in my spec) none of the rotation or position metrics have a unit specified.

That's why I'm interested in ratios -- we can say `value="16777216/5"`, and it'll put 16M in the numerator and 5 in the denominator.  Odds are we'll end up sharing the denominator across most of the settings.

scottlaird, 2024-02-28:
Okay, I'm working on this in #355 now, for generic CiA 402 support.  It won't help lcec_deasda today, but it probably will at some point.

So far, I just have the code that converts strings as described above into a ratio.  I have 4 test cases right now:

- "1/2"
- "3"
- "5.2"
- "3355443.2"

Somewhat impressively, it is actually able to turn those into (1,2), (3,1), (26,5), and (16777216,5).

Of course, now that I've actually looked through `cia402.md`, it turns out that they're using a mix of 32/32, 32/16, and 16/16 bit ints, with and without signs.  Of course.  I'll go adjust my code for 32/32, with a configurable limit on the denominator.

zmrdko, 2024-02-28:
I 100% agree on ratios. You made it as flexible as it can be. Let me know if you want me to test it!

closed 2025-01-05 by zmrdko