linuxcnc-ethercat/linuxcnc-ethercat #294: Support for Beckhoff EL6021

GuiHue, 2024-02-09:
Support [Beckhoff EL6021 ](https://www.beckhoff.com/en-us/products/i-o/ethercat-terminals/el6xxx-communication/el6021.html) communication module.

It's a 1-channel communication interface for serial using RS422 and RS485.

Would be a fantastic way to get many VFDs into EtherCAT and not having one of those nasty USB to serial things. ATC router spindles can be a bit tricky since they often have a 4 or more poles which requires VFD capable of driving >400Hz. This limits the choice in the market substantially and often collides with export limitations which mean that non corporate legal entities cannot buy them new. Many established VFDs are not available with EtherCAT (or are stupid expensive).

I have one sitting here.

scottlaird, 2024-02-09:
I have an EL6001, which I *think* is the same basic hardware, just single-ended instead of differential.  Not that useful for driving VFDs, but easier for testing and probably register-compatible.  IIRC our EtherCAT library has code for EtherCAT-attached serial devices, so it's probably not too hard to code up.

The problem is that I have no idea what they should look like in LinuxCNC.  I don't think that writing a "VFD-brand-X-over-Beckhoff-serial-model-Y-over-EtherCAT" driver is the right way to go :-).  I don't see a "generic serial" or "generic RS485" layer that we could attach into.  The best modbus VFD example that I can see just uses `/dev/ttyXXX` directly.  I guess we *could* present a fake serial port to Linux, and that'd be semi-useful outside of LinuxCNC.  The problem is that I don't *think* there's an actual way to create a serial device from user space.  We could use a PTY, which is close, but you couldn't change baud rate, etc, via ioctls.

Alternately, it looks like there's a `mb2hal` module that can currently talk to both serial and Modbus TCP devices.  I can't remember if Modbus TCP is literally just MB over TCP, or if it's more complicated.  I've only used it once, and it was a few years ago.  We could maybe just listen on a TCP port and feed that through to the EtherCAT serial port?  It'd be kind of weird, structurally, but I don't see why it wouldn't work, and it'd be somewhat flexible.  Of course, then you have issues around networks and security, but they're manageable, mostly.

Do you want RS-422/485, or do you want Modbus/Canbus/etc?

scottlaird, 2024-02-09:
There is an EL60xx *kernel* driver in https://gitlab.com/etherlab.org/ethercat/-/tree/master/examples/tty.  I'm not sure that we could share a master between this and LCEC, but at least it's an example.

grandixximo, 2026-05-01:
@GuiHue if you're still active (also pinging you on #435), is the EL6021 still on your bench? This one is genuinely hard because it needs to expose serial to userspace as a tty so existing modbus/VFD components can talk to it. Etherlab has a tty kernel example but it's unclear how to share the master. Lower priority unless you have appetite for a deeper dive.

rodw-au, 2026-09-05:
Unrelated but  it may serve as a guide, I just shared a MODBUS RTU configuration for the ethercat catio-a IO module thats available on aliexpress. I have this controlling my VFD now.
Forum: https://forum.linuxcnc.org/ethercat/59180-catio-a-spindle-control-with-modbus-over-ethercat
This device supports MODBUS RTU and custom serial comms. The docs for it are on the forum too but I can share them here if you like. The Google translation from Chinese is not easy to read.

The approach it takes is it has a set of PDOS that data is written to and sent from. Each of these PDO's has a set of SDOS that define the MODBUS register to read from or write to. These SDOs are a bit funny because once they are written to they are read only until a power cycle so I used a bash script to set the environment up before running LinuxCNC.

Once configured, the hal side (which I also shared) it is basically the same as mb2hal (except it uses ints and mb2hal uses floats) so I would recommend you get your VFD going with mb2hal before starting this. I found my VFD could not run higher than 38400 baud so check this with mb2hal to avoid wasting time.

The two parameters that are critical is the MODBUS timeout and the MODBUS cycle time (time between updates). I have not explored in detail but you need to use the fastest baud rate you can given the disparity between the servo thread and MODBUS timings. I did get timeut errors reported but its working perfectly for me now even with these errors. I wondered if the lcec.write_all command in the hal file loop was causing the serial port to flood but have no way to checking this. The device did not work at 9600 baud (well it did but you could not change the frequency ever without really long cycle times that were unuseable).... If the driver could have a way of configuring each register when loaded, (even with a C header file to start with)  your idea might work quite well.

grandixximo, 2026-10-03:
Solved by #525 (on master, shipped in 1.46.0): the EL6001/EL6021 driver exposes the terminal's serial port as a CUSE character device (`/dev/<ttyName>`), so mb2hal/pymodbus/minicom work directly, with termios settings applied to the terminal at runtime. Hardware-verified with Modbus RTU over mb2hal. Docs: documentation/el6021.md. Closing; thanks @GuiHue for the original request and @rodw-au for pushing it forward.

closed 2026-10-03 by grandixximo