linuxcnc-ethercat/linuxcnc-ethercat #327: Support Leadshine 2CS3E-D507 (and other Leadshine EtherCAT stepper drives)

scottlaird, 2024-02-25:
Okay, my Leadshine 2CS3E-D507 has arrived.  That's a 2-axis EtherCAT closed-loop stepper drive.

Here's what I see when I wave `lcec_configgen` over it:

```xml
    <slave idx="21" type="basic_cia402" vid="0x00004321" pid="0x00002100" name="D22">
      <!--2CL3-EC507(COE)-->
      <modParam name="enablePP" value="true"/>
      <modParam name="enablePV" value="true"/>
      <modParam name="enableHM" value="true"/>
      <modParam name="enableCSP" value="true"/>
      <modParam name="enableActualCurrent" value="true"/>
      <modParam name="enableActualFollowingError" value="true"/>
      <modParam name="enableActualTorque" value="true"/>
      <modParam name="enableActualVelocitySensor" value="true"/>
      <modParam name="enableActualVoltage" value="true"/>
      <modParam name="enableFollowingErrorTimeout" value="true"/>
      <modParam name="enableFollowingErrorWindow" value="true"/>
      <modParam name="enableHomeAccel" value="true"/>
      <modParam name="enableInterpolationTimePeriod" value="true"/>
      <modParam name="enableMaximumAcceleration" value="true"/>
      <modParam name="enableMaximumCurrent" value="true"/>
      <modParam name="enableMaximumDeceleration" value="true"/>
      <modParam name="enableMaximumMotorRPM" value="true"/>
      <modParam name="enableMaximumTorque" value="true"/>
      <modParam name="enableMotorRatedCurrent" value="true"/>
      <modParam name="enableMotorRatedTorque" value="true"/>
      <modParam name="enablePolarity" value="true"/>
      <modParam name="enableProfileAccel" value="true"/>
      <modParam name="enableProfileDecel" value="true"/>
      <modParam name="enableProfileEndVelocity" value="true"/>
      <modParam name="enableProfileMaxVelocity" value="true"/>
      <modParam name="enableProfileVelocity" value="true"/>
      <modParam name="enableTargetTorque" value="true"/>
      <modParam name="enableTorqueDemand" value="true"/>
      <modParam name="enableTorqueProfileType" value="true"/>
      <modParam name="enableTorqueSlope" value="true"/>
      <modParam name="enableVelocityDemand" value="true"/>
      <modParam name="enableVelocityErrorTime" value="true"/>
      <modParam name="enableVelocityErrorWindow" value="true"/>
      <modParam name="enableVelocitySensorSelector" value="true"/>
      <modParam name="enableVelocityThresholdTime" value="true"/>
      <modParam name="enableVelocityThresholdWindow" value="true"/>
    </slave>
```

And here's a stock PDO on first power-up:

```
SM0: PhysAddr 0x1000, DefaultSize  128, ControlRegister 0x26, Enable 1
SM1: PhysAddr 0x1100, DefaultSize  128, ControlRegister 0x22, Enable 1
SM2: PhysAddr 0x1200, DefaultSize    0, ControlRegister 0x64, Enable 1
  RxPDO 0x1600 "Axis 1 Receive PDO 1 Mapping Parameter"
    PDO entry 0x6040:00, 16 bit, "Control Word"
    PDO entry 0x607a:00, 32 bit, "Profile Target Position"
    PDO entry 0x60b8:00, 16 bit, "Touch Probe Function"
  RxPDO 0x1610 "Axis 2 Receive PDO 1 Mapping Parameter"
    PDO entry 0x6040:00, 16 bit, "Control Word"
    PDO entry 0x607a:00, 32 bit, "Profile Target Position"
    PDO entry 0x60b8:00, 16 bit, "Touch Probe Function"
SM3: PhysAddr 0x1400, DefaultSize    0, ControlRegister 0x20, Enable 1
  TxPDO 0x1a00 "Axis 1 Transmit PDO 1 Mapping Parameter"
    PDO entry 0x603f:00, 16 bit, "Last Error Code"
    PDO entry 0x6041:00, 16 bit, "Status Word"
    PDO entry 0x6061:00,  8 bit, "Modes of Operation Display"
    PDO entry 0x6064:00, 32 bit, "Actual Motor Position"
    PDO entry 0x60b9:00, 16 bit, "Touch Probe Status"
    PDO entry 0x60ba:00, 32 bit, "Touch Probe 1 Positive Value"
    PDO entry 0x60fd:00, 32 bit, "Digital Inputs"
  TxPDO 0x1a10 "Axis 2 Transmit PDO 1 Mapping Parameter"
    PDO entry 0x603f:00, 16 bit, "Last Error Code"
    PDO entry 0x6041:00, 16 bit, "Status Word"
    PDO entry 0x6061:00,  8 bit, "Modes of Operation Display"
    PDO entry 0x6064:00, 32 bit, "Actual Motor Position"
    PDO entry 0x60b9:00, 16 bit, "Touch Probe Status"
    PDO entry 0x60ba:00, 32 bit, "Touch Probe 1 Positive Value"
    PDO entry 0x60fd:00, 32 bit, "Digital Inputs"
```

scottlaird, 2024-02-25:
Making some progress:

```
     8  bit   OUT         FALSE  lcec.0.D22.slave-online
     8  bit   OUT         FALSE  lcec.0.D22.slave-oper
     8  bit   OUT         FALSE  lcec.0.D22.slave-state-init
     8  bit   OUT         FALSE  lcec.0.D22.slave-state-op
     8  bit   OUT         FALSE  lcec.0.D22.slave-state-preop
     8  bit   OUT         FALSE  lcec.0.D22.slave-state-safeop
     8  s32   OUT             0  lcec.0.D22.srv-1-actual-current
     8  u32   OUT    0x00000000  lcec.0.D22.srv-1-actual-following-error
     8  s32   OUT             0  lcec.0.D22.srv-1-actual-position
     8  s32   OUT             0  lcec.0.D22.srv-1-actual-torque
     8  s32   OUT             0  lcec.0.D22.srv-1-actual-velocity
     8  s32   OUT             0  lcec.0.D22.srv-1-actual-velocity-sensor
     8  u32   OUT    0x00000000  lcec.0.D22.srv-1-actual-voltage
     8  u32   IN     0x00000000  lcec.0.D22.srv-1-cia-controlword
     8  u32   OUT    0x00000000  lcec.0.D22.srv-1-cia-statusword
     8  u32   IN     0x00000064  lcec.0.D22.srv-1-following-error-timeout
     8  u32   IN     0x00002710  lcec.0.D22.srv-1-following-error-window
     8  u32   IN     0x0007A120  lcec.0.D22.srv-1-home-accel
     8  s32   IN             19  lcec.0.D22.srv-1-home-method
     8  u32   IN     0x00002710  lcec.0.D22.srv-1-home-velocity-fast
     8  u32   IN     0x00001388  lcec.0.D22.srv-1-home-velocity-slow
     8  u32   IN     0x00000002  lcec.0.D22.srv-1-interpolation-time-period
     8  u32   IN     0x05F5E100  lcec.0.D22.srv-1-maximum-acceleration
     8  u32   IN     0x00000000  lcec.0.D22.srv-1-maximum-current
     8  u32   IN     0x05F5E100  lcec.0.D22.srv-1-maximum-deceleration
     8  u32   IN     0x00000BB8  lcec.0.D22.srv-1-maximum-motor-rpm
     8  u32   IN     0x00000BB8  lcec.0.D22.srv-1-motor-rated-current
     8  u32   IN     0x00000BB8  lcec.0.D22.srv-1-motor-rated-torque
     8  s32   IN              0  lcec.0.D22.srv-1-opmode
     8  s32   OUT             0  lcec.0.D22.srv-1-opmode-display
     8  u32   IN     0x00000000  lcec.0.D22.srv-1-polarity
     8  u32   IN     0x00002710  lcec.0.D22.srv-1-profile-accel
     8  u32   IN     0x00002710  lcec.0.D22.srv-1-profile-decel
     8  u32   IN     0x00000000  lcec.0.D22.srv-1-profile-end-velocity
     8  u32   IN     0x7FFFFFFF  lcec.0.D22.srv-1-profile-max-velocity
     8  u32   IN     0x00002710  lcec.0.D22.srv-1-profile-velocity
     8  u32   OUT    0x000000A5  lcec.0.D22.srv-1-supported-modes
     8  bit   OUT          TRUE  lcec.0.D22.srv-1-supports-mode-csp
     8  bit   OUT         FALSE  lcec.0.D22.srv-1-supports-mode-cst
     8  bit   OUT         FALSE  lcec.0.D22.srv-1-supports-mode-csv
     8  bit   OUT          TRUE  lcec.0.D22.srv-1-supports-mode-hm
     8  bit   OUT         FALSE  lcec.0.D22.srv-1-supports-mode-ip
     8  bit   OUT          TRUE  lcec.0.D22.srv-1-supports-mode-pp
     8  bit   OUT          TRUE  lcec.0.D22.srv-1-supports-mode-pv
     8  bit   OUT         FALSE  lcec.0.D22.srv-1-supports-mode-tq
     8  bit   OUT         FALSE  lcec.0.D22.srv-1-supports-mode-vl
     8  s32   IN              0  lcec.0.D22.srv-1-target-position
     8  s32   IN              0  lcec.0.D22.srv-1-target-torque
     8  s32   IN              0  lcec.0.D22.srv-1-target-velocity
     8  s32   OUT             0  lcec.0.D22.srv-1-torque-demand
     8  u32   IN     0x00000BB8  lcec.0.D22.srv-1-torque-maximum
     8  s32   IN              0  lcec.0.D22.srv-1-torque-profile-type
     8  u32   IN     0x00000064  lcec.0.D22.srv-1-torque-slope
     8  s32   OUT             0  lcec.0.D22.srv-1-velocity-demand
     8  u32   IN     0x00000064  lcec.0.D22.srv-1-velocity-error-time
     8  u32   IN     0x00002710  lcec.0.D22.srv-1-velocity-error-window
     8  s32   IN          65535  lcec.0.D22.srv-1-velocity-sensor-selector
     8  u32   IN     0x00000064  lcec.0.D22.srv-1-velocity-threshold-time
     8  u32   IN     0x00002710  lcec.0.D22.srv-1-velocity-threshold-window
     8  s32   OUT             0  lcec.0.D22.srv-2-actual-current
     8  u32   OUT    0x00000000  lcec.0.D22.srv-2-actual-following-error
     8  s32   OUT             0  lcec.0.D22.srv-2-actual-position
     8  s32   OUT             0  lcec.0.D22.srv-2-actual-torque
     8  s32   OUT             0  lcec.0.D22.srv-2-actual-velocity
     8  s32   OUT             0  lcec.0.D22.srv-2-actual-velocity-sensor
     8  u32   OUT    0x00000000  lcec.0.D22.srv-2-actual-voltage
     8  u32   IN     0x00000000  lcec.0.D22.srv-2-cia-controlword
[... 21 lines cut ...]
     8  u32   IN     0x00002710  lcec.0.D22.srv-2-profile-velocity
     8  u32   OUT    0x000000A5  lcec.0.D22.srv-2-supported-modes
     8  bit   OUT          TRUE  lcec.0.D22.srv-2-supports-mode-csp
     8  bit   OUT         FALSE  lcec.0.D22.srv-2-supports-mode-cst
     8  bit   OUT         FALSE  lcec.0.D22.srv-2-supports-mode-csv
     8  bit   OUT          TRUE  lcec.0.D22.srv-2-supports-mode-hm
     8  bit   OUT         FALSE  lcec.0.D22.srv-2-supports-mode-ip
     8  bit   OUT          TRUE  lcec.0.D22.srv-2-supports-mode-pp
     8  bit   OUT          TRUE  lcec.0.D22.srv-2-supports-mode-pv
     8  bit   OUT         FALSE  lcec.0.D22.srv-2-supports-mode-tq
     8  bit   OUT         FALSE  lcec.0.D22.srv-2-supports-mode-vl
     8  s32   IN              0  lcec.0.D22.srv-2-target-position
     8  s32   IN              0  lcec.0.D22.srv-2-target-torque
     8  s32   IN              0  lcec.0.D22.srv-2-target-velocity
[... cut ...]
     8  u32   IN     0x00000064  lcec.0.D22.srv-2-torque-slope
     8  s32   OUT             0  lcec.0.D22.srv-2-velocity-demand
     8  u32   IN     0x00000064  lcec.0.D22.srv-2-velocity-error-time
     8  u32   IN     0x00002710  lcec.0.D22.srv-2-velocity-error-window
     8  s32   IN          65535  lcec.0.D22.srv-2-velocity-sensor-selector
     8  u32   IN     0x00000064  lcec.0.D22.srv-2-velocity-threshold-time
     8  u32   IN     0x00002710  lcec.0.D22.srv-2-velocity-threshold-window
```

scottlaird, 2024-02-26:
So, I ran into a fun issue here, the 2CS3E-D507 (man, that just rolls off the tongue) has some pretty serious limitations on the number of PDO entries that can be mapped.  It supports 4 read PDOs and 2 transmit PDOs, with 8 entries each. 

If you look above, you'll notice *way* more than 32 pins.  As things stand, the device won't go into OP mode, and it spits out a difficult-to-parse error message in `dmesg` when it fails to add a 9th PDO entry to a PDO.

It looks like all of the Leadshine EtherCAT devices share the same basic limitation, not just the 2CS3Es.  I spot-checked the single-axis CS3E, the CS3E-B, the EM3E open-loop drives, the EL7 servo drive, and the new EL8 servo drive.  They're all limited to 8 PDO entries per PDO.  Some models *may* have 4 TX PDOs, but that's not a major win in my book. 

Right now, we're adding axis 1 to the first PDO and each additional axis goes into the next PDO.  Device-specific drivers will usually use the next PDO.  Remember that CiA 402 has a number of mandatory or near-mandatory PDOs (controlword, opmode).  Just adding all of the required `csp` and `csv` settings puts us close to the limit, much less homing, etc.

So, I think we have a few options here:

1.  Detect the number of allowed PDO entries per PDO and report a better error.  Then it'd be up to the user to disable settings until their device works.  Not great, but at *some* point this will end up being unavoidable.
2.  Stop putting PDO entries into PDOs by axis, and auto-overflow into the next PDO when one fills up.  This would get us ~2x as many PDOs on the 2CS3E and 4x as many on 1-axis devices.  I *think* it'd work?
3. Unmap a bunch of the rarely-written "config" pins and switch to writing to them as SDOs in `_write`.  The `home-accel` setting doesn't *really* need to be updated at 1 kHz, does it?  I think I can do this fairly easily.  We're allowed to write to SDOs in realtime mode, right?
4. Move all of the config pins back to modParams.  I'm not a big fan of this for UX reasons.
5. Bug Leadshine and see if 8 is just a dumb software limitation, or if their hardware is actually capped there.  Then get everyone to do a firmware upgrade.  Probably worth doing this anyway, but I don't expect quick payoffs.

I think we'll probably end up having to do options 1, 2, and 3, plus make an attempt at 5.  1 and 2 are somewhat interrelated (they both need to know how many PDO entries you can have per PDO), while 3 is completely independent.  I don't expect 5 to pay off, but it's worth asking.

scottlaird, 2024-02-26:
Ugh, option 3 isn't happening.  From https://gitlab.com/etherlab.org/ethercat/-/blob/master/include/ecrt.h#L1480:

> This method has to be called in non-realtime context before ecrt_master_activate().

That's going to make this a *lot* of fun.  Practically speaking, it's going to make the configuration of Leadshine devices more work, because they'll have to be configured for a ~single mode via the XML file, rather than allowing all modes there and then letting the user sort it out at runtime.

jcedrych, 2024-02-27:
Hi, I use this piece of code lcec_class_rt_sdo.c  mentioned here:
https://forum.linuxcnc.org/24-hal-components/48407-changing-ethercat-sdos-using-user-panel

[github.com/steup-engineering/linuxcnc-et.../lcec_class_rt_sdo.c](https://github.com/steup-engineering/linuxcnc-ethercat/blob/add-nanotec-pd4e/src/lcec_class_rt_sdo.c)

[github.com/steup-engineering/linuxcnc-et.../src/lcec_nanopd4e.c](https://github.com/steup-engineering/linuxcnc-ethercat/blob/add-nanotec-pd4e/src/lcec_nanopd4e.c)

scottlaird, 2024-02-27:
Ah, interesting!  Yeah, that's the same basic process that I'm using, except there's a difference in the way that we do writes.  I *think* mine is right, although that depends on the interpretation of how you're supposed to use `ecrt_sdo_request_state()`.

I'm doing (with minor editing; the original in `lcec_class_cia402.c` is in a preprocessor macro):

```c
      if (*(data->pin_value) != data->pin_value_old) { 
        if (ecrt_sdo_request_state(data->pin_sdorequest) != EC_REQUEST_BUSY ) { 
	  data->pin_value_old = *(data->pin_value); 
	  uint8_t *sdo_tmp = ecrt_sdo_request_data(data->pin_sdorequest);	
	  EC_WRITE_S8(sdo_tmp, data->pin_old); 
	  ecrt_sdo_request_write(data->pin_sdorequest); 
        }								
      }
    } 
```

The [code that you linked](https://github.com/steup-engineering/linuxcnc-ethercat/blob/add-nanotec-pd4e/src/lcec_class_rt_sdo.c#L108) looks more like this (also reformatted, sticking to the S8 case for clarity):

```c
  if (!hal_data->do_write) {
    if (hal_data->s_parameter != hal_data->s_parameter_old) {
      EC_WRITE_S8(ecrt_sdo_request_data(hal_data->sdo_request_data), hal_data->s_parameter);
      hal_data->do_write = 1; 
      ecrt_sdo_request_write(hal_data->sdo_request_data); 
    }

  switch (ecrt_sdo_request_state(hal_data->sdo_request_data)) {
    case EC_REQUEST_BUSY:
      break;
    case EC_REQUEST_SUCCESS:
      if (hal_data->do_write) {
        hal_data->s_parameter_old = hal_data->s_parameter;
        hal_data->do_write = 0;
      }
    break;
  }    
```

Basically, I check `EC_REQUEST_BUSY` *before* writing to `ecrt_sdo_request_data()`, and delay doing the write if it's busy.  Once I do the write, I trust that Etherlab's code gets it to the device eventually.  This is based on the assumption that `EC_REQUEST_BUSY` actually means that there's a pending write for this SDO, and if I overwrite it then (a) the previous write will vanish and (b) there's the chance of a race condition where the slave gets some bytes from the current write and some from the previous write.  That's rather unlikely with 8/16/32 bit values, but if we were writing to string SDOs then it'd be a worry. 

The linked code uses `EC_REQUEST_BUSY` as an indicator that the request hasn't completed yet, which I *think* misses the point of why it's there.  It's not to flag "the write is complete" -- it's there to say "the data is in use, please don't modify."

Looking at the `ecrt_sdo_request_write` and `ecrt_sdo_request_state` code, it looks like the kernel actually does writes using the user-mode data buffer as a source, and `request_state` asks the kernel about the current state.  That's probably not how I'd have implemented this, at least for <=64 bit data blobs, but it's the only reasonable way to implement longer writes.  That means that (a) `request_state()` is actually kind of expensive, and (b) it may take a while (hundreds of usec?) before each write clears and we can do the next one.  Which means, effectively, that we *should* be clear if we only do one SDO write per update cycle (per SDO).  

My version basically just skips writes when we get `EC_REQUEST_BUSY`, and tries again during the next write cycle.  For config-like objects, that's probably fine.  I really wouldn't want motion control to depend on that, though.  The linked version is a bit worse; if you manage to do a write more than every ~other cycle, then you run the risk of silently losing the second one.  Again, for config-like things that's not very likely, but it's possible.  I could absolutely see a case where a .hal file sets the same pin twice during initialization, and sometimes the timing would result in the second write simply never happening, as `        hal_data->s_parameter_old = hal_data->s_parameter` would end up skipping the second write when cleaning up from the first.  My version should be safe; two writes to a pin will either result in 1 or 2 SDO writes, depending on timing, but you shouldn't see writes *vanish*.

Concurrency is fun, isn't it?

scottlaird, 2024-02-27:
At this point, I've moved the bulk of the config settings to SDOs, and that *mostly* keeps us under 8 PDOs per channel on the 2CS3E.  Unfortunately, there are still a couple settings to add, so it may break again.

FWIW, RTelligence allows 12 entries per PDO.  The Delta ASDA A3 is also 8 per PDO.  So 8 isn't unheard of here.  I guess I've been use to Beckhoff devices which frequently allow a ton of PDO entries.

scottlaird, 2024-02-27:
To be clear, we currently have 16 different output PDOs defined in `lcec_cia402_add_output_sync`:

- controlword
- digital_output
- home_method
- interpolation_time_period
- motion_profile
- opmode
- polarity
- profile_velocity
- target_position
- target_torque
- target_velocity
- target_vl
- torque_profile_type
- torque_slope
- vl_accel
- vl_decel

Of these, we can *probably* move home_method, interpolation_time_period, polarity, vl_accel, and vl_decel to SDOs.  That leaves 11, but I doubt there's a lot of overlap on real hardware -- I don't think I've ever seen hardware that supports vl and pp modes, for instance.  We may be able to move motion_profile, torque_profile_type, and torque_slope as well, but I'd want to research a bit more.  That'd get us to 8, which would ~always be safe, even on a device that supported every single option.  Anyway, I should move this to its own issue.

scottlaird, 2024-03-02:
Okay, with the current state in #343, I have it connected, in state OP, with no errors in `dmesg`.

I fired it up with `halrun -f fulltest.hal` (it's checked in, although I have minor tweaks vs the current rev in git).

I ran `halcmd setp cia402.0.enable true` and `halcmd setp cia402.0.pos-cmd 10000`, and nothing happens.  Here are the `cia402.0` pins:

```
    11  u32   OUT    0x0000000F  cia402.0.controlword ==> rtec-controlword
    11  s32   IN              0  cia402.0.drv-actual-position <== rtec-drv-act-pos
    11  s32   IN              0  cia402.0.drv-actual-velocity <== rtec-drv-act-velo
    11  bit   OUT         FALSE  cia402.0.drv-fault
    11  s32   OUT         10000  cia402.0.drv-target-position ==> rtec-drv-target-pos
    11  s32   OUT             0  cia402.0.drv-target-velocity
    11  bit   IN           TRUE  cia402.0.enable
    11  bit   IN          FALSE  cia402.0.fault-reset
    11  bit   I/O         FALSE  cia402.0.home
    11  s32   OUT             8  cia402.0.opmode ==> rtec-modes-of-operation
    11  bit   OUT         FALSE  cia402.0.opmode-cyclic-position
    11  bit   OUT         FALSE  cia402.0.opmode-cyclic-velocity
    11  s32   IN              0  cia402.0.opmode-display <== rtec-opmode-display
    11  bit   OUT         FALSE  cia402.0.opmode-homing
    11  bit   OUT          TRUE  cia402.0.opmode-no-mode
    11  float IN          10000  cia402.0.pos-cmd
    11  float OUT             0  cia402.0.pos-fb
    11  s32   OUT           122  cia402.0.read-all.time
    11  bit   OUT         FALSE  cia402.0.stat-fault
    11  bit   OUT         FALSE  cia402.0.stat-homed
    11  bit   OUT         FALSE  cia402.0.stat-homing
    11  bit   OUT          TRUE  cia402.0.stat-op-enabled
    11  bit   OUT          TRUE  cia402.0.stat-quick-stop
    11  bit   OUT          TRUE  cia402.0.stat-remote
    11  bit   OUT          TRUE  cia402.0.stat-switched-on
    11  bit   OUT         FALSE  cia402.0.stat-switchon-disabled
    11  bit   OUT          TRUE  cia402.0.stat-switchon-ready
    11  bit   OUT         FALSE  cia402.0.stat-target-reached
    11  bit   OUT          TRUE  cia402.0.stat-voltage-enabled
    11  bit   OUT         FALSE  cia402.0.stat-warning
    11  u32   IN     0x00001637  cia402.0.statusword <== rtec-statusword
    11  float IN              0  cia402.0.velocity-cmd
    11  float OUT             0  cia402.0.velocity-fb
    11  s32   OUT            46  cia402.0.write-all.time
                         <== cia402.0.controlword
                         ==> cia402.0.drv-actual-position
                         ==> cia402.0.drv-actual-velocity
                         <== cia402.0.drv-target-position
                         <== cia402.0.opmode
                         ==> cia402.0.opmode-display
                         ==> cia402.0.statusword
    11  bit   RW           TRUE  cia402.0.auto-fault-reset
    11  bit   RW           TRUE  cia402.0.csp-mode
    11  float RW              1  cia402.0.pos-scale
    11  s32   RW           1466  cia402.0.read-all.tmax
    11  bit   RO          FALSE  cia402.0.read-all.tmax-increased
    11  float RW              1  cia402.0.velo-scale
    11  s32   RW           2514  cia402.0.write-all.tmax
    11  bit   RO          FALSE  cia402.0.write-all.tmax-increased
```

Notice that `pos-fb` (and `drv-actual-position` are 0.  The `D22.srv-1` pins might be more useful:

```
     8  s32   OUT             0  lcec.0.D22.srv-1-actual-position ==> rtec-drv-act-pos
     8  s32   OUT             0  lcec.0.D22.srv-1-actual-velocity ==> rtec-drv-act-velo
     8  u32   IN     0x0000000F  lcec.0.D22.srv-1-cia-controlword <== rtec-controlword
     8  u32   OUT    0x00001637  lcec.0.D22.srv-1-cia-statusword ==> rtec-statusword
     8  bit   OUT         FALSE  lcec.0.D22.srv-1-din-0
     8  bit   OUT          TRUE  lcec.0.D22.srv-1-din-0-not
     8  bit   OUT         FALSE  lcec.0.D22.srv-1-din-1
     8  bit   OUT          TRUE  lcec.0.D22.srv-1-din-1-not
     8  bit   OUT         FALSE  lcec.0.D22.srv-1-din-10
[... 101 lines cut ...]
- (0) ready start is true
- (1) start true
- (2) permitted operation true
- (3) error false
- (4) power on true
- (5) quickstop true ???
- (6) not started false
- (7) reserved (false)
- (8) quick stop (csp) false
- (9) distance true
- (10) invalid true ???
- (11)   limit validity false (I think this is "we hit the hardware limits")
- (12) following effective true
- (13) invalid (false)
- (14) invalid (false)
- (15) invalid (false)

So, why is bit 10 true?  And what does bit 5 mean?

Checking additional status bits from the Leadshine documentation:

- 0x3ffe:01 is an internal error code.  Reads as 0x0000
- The LED on the thing is blinking `2-E` and either `81b` or `816`.  The manual is not helpful here, at least not yet.  Neither of these numbers match what I'd expect to see.

I wonder if I skipped a few steps setting this thing up?  I'll go through LS's directions and see if that helps.

scottlaird, 2024-03-02:
I should also just put it into PV mode manually and see what happens.  It's less complex than CSP.

scottlaird, 2024-03-10:
FWIW, I have my motor test bench torn apart right now, and I'm re-assembling it over the next few days.  So I'm not making any progress on the 2CS3E in the short term, until I'm able to power it back up again, and I can't do that until I get a few things that are due on Wednesday.

scottlaird, 2024-03-27:
Set #180 for updates.  Things are re-assembled.  I have CSV working on CSV-capable devices, but the 2CS3 doesn't do CSV.  CSP isn't working right for anything at the moment.

scottlaird, 2024-04-20:
Okay, a basic "leadshine_stepper" driver is included.  It's kind of feature-sparse compared to the RTelligent driver, but it works for me.  I can't actually find EtherCAT PIDs for all of Leadshine's EtherCAT devices in their assorted ESI files, *and* they can't always decide what the "correct" name is for their devices.  Case in point, my 2CS3E-D507 calls itself a 2CL3-EC507 via `ethercat slaves`.  Since I bought it directly from Leadshine, I'm pretty sure it's not a counterfeit :-).

scottlaird, 2024-04-20:
The only remaining task for this is to add modparams and/or pins for a handful of Leadshine-specific features (motor amps, for example).  In general, Leadshine follows the CiA 402 spec more closely than RTelligent, so there won't be as many device-specific configs needed.

scottlaird, 2024-04-21:
Okay, I think that's it.  I'm sure there are still bugs, but it's complete enough for now.  We can deal with features, etc, in future issues.

closed 2024-04-21 by scottlaird