linuxcnc-ethercat/linuxcnc-ethercat #342: Add a "minimum cycle time" test

scottlaird, 2024-02-27:
I should write a "minimum cycle time" test that takes a known XML/HAL config with known hardware, and then starts modifying the cycle time (`loadrt threads name1=servo-thread period1=XXX` and `<master idx="0" appTimePeriod="XXX000" refClockSyncCycles="1000">`) to see how fast we're able to run.

There are three goals here:

1. Capture regressions, where we do something that makes everything slower.
2. Make sure that our handling of cycle-missed / slave-leaving-OP is reasonable.
3. Understand (and document) how fast we're actually able to go, at least on limited hardware.  Bonus points for identifying things that make it get slower.

jlama, 2024-03-20:
Note that for RT-PREEMPT systems refClockSyncCycles should be -1.
This way the master clock is the first slave supporting DC, and LinuxCNC syncs to this clock.

When refClockSyncCycles is >= 0, the master clock is LinuxCNC which is then synced to all the slaves. The problem with this is that the LinuxCNC host jitter is propagated and the DC will slowly drift, possibly causing sync issues.

scottlaird, 2024-03-20:
Ah, thanks.  refClockSyncCycles seems to be kind of a mystery to a lot of people.  I'll refine this a bit and add it to the docs.