linuxcnc-ethercat/linuxcnc-ethercat #357: Beckhoff EL6751 support

zmrdko, 2024-02-28:
Hi,
i have this Beckhoff EL6751 CANopen Master/Slave communication channel.
I was hoping to connect CANopen VFD to it and somehow control VFD using LinuxCNC.
But I don't know how to configure it.
Does anybody have more info on this, how to make it work?

scottlaird, 2024-02-28:
I have an EL6751 as well (part of a "big block of EL* parts" lot on eBay), but nothing to plug into it.  *Supposedly* it's possible with `generic` and TwinCAT, but it's a pain.  It might be interesting to see if we could reduce the "pain" side of things.

Does anyone have any pointers to cheap CANopen devices that it could be used with?

zmrdko, 2024-02-28:
Hi, I have Cumark ES580 VFD which supposedly should have CANopen interface:
[image]
screenshot from manual.

however cheapest device I could found was this:
https://www.aliexpress.com/item/1005003589127614.html

zmrdko, 2024-02-28:
Delta makes CANopen interfaces for their VFDs. If you happen to have any.

scottlaird, 2024-02-28:
The problem here, of course, is that this starts to make the CiA 402 driver look like it should sit on top of a generic CANopen layer, not the CoE layer that we're using now.  Then we could probably talk to these things using nothing but a cheap USB RS-485 adapter like the modbus folks use.

scottlaird, 2024-02-28:
Also: I'm not writing a generic CANopen layer.

scottlaird, 2024-02-28:
OTOH, EL6751->CiA 402 *might* not be too terrible.

zmrdko, 2024-03-07:
Do you have any suggestions, where to start? I have wired EL6751 to my Cumark VFD, which has CANopen interface.
I have TwinCAT installed and EL6751 being detected.

scottlaird, 2024-03-07:
I've been looking through EL6751 manuals and random forum posts, and I'm still somewhat confused.  It looks like it can bridge PDOs and SDOs between EtherCAT and CANopen, but it's not clear exactly how that works, especially when you have multiple CANopen devices behind the EL6751.  Does it allocate new SDO numbers on the EtherCAT end?

Can you talk to the VFD from TwinCAT?

Can you run at least `ethercat sdos` against the EL6751?  Maybe also `lcec_configgen`.  I'm curious what it sees.

scottlaird, 2024-03-09:
One thing about CiA 402 over CANopen that may be weird, even if we deal with all of the other issues (PDO/SDO mapping, etc) is *speed*.  That will probably pose a couple problems:

- First, it's entirely possible that CSV/CSP modes won't really make sense, and we'd need to fall back to PV/PP modes, which are designed for slower communication mediums.  Probably not a problem for a VFD, though.
- Even at 1 Mbps, the max speed that the EL6751 supports, we'd only be able to move 1% of the data that EtherCAT supports.  That might mean that we'd have to slow down the servo thread from 1 kHz to something slower.

The second one there is actually kind of interesting.  I don't think LCEC uses many global variables, so we could *probably* use a servo thread per master, which would let us run one master as 1 kHz and another at something slower.  I have no idea how well LinuxCNC itself would support that, though.  I haven't seen anything that really makes me think that it has enough locking in place to make that a good idea.

scottlaird, 2024-04-09:
FWIW, I now have a CANopen stepper drive on hand.  It's not very high on my to-do list, though.  And I doubt it's really possible to get it working without TwinCAT, which never works very well for me.  But it'll be an interesting rainy-day project.

[image]

zmrdko, 2024-05-06:
I might be getting canopen communication card for delta vfd to see, if it gets detected by twincat via el6751.

Some guys on linuxcnc forum seem to got it working, but its not explained how..