linuxcnc-ethercat/linuxcnc-ethercat #368: Build time very long

vr6111, 2024-03-02:
Hey,
I was trying to build the repo with an additional driver but the build time exceeded >12h and I needed to stop. I simply ran "make -j 8" on the master branch with the new driver added to src/devices. Does anyone have an idea why this happens? Maybe I'm missing something here.

scottlaird, 2024-03-02:
It's probably stuck in a dependency-mapping loop; the actual build shouldn't take more than a few seconds on current hardware.  IIRC the N100 that I'm testing on is fast enough that I just do `make clean; make -j all` every time.

If you edit `src/Makefile` and comment out the line `include $(all-srcs:.c=.d)`, does it work?  Without that line, incremental builds are kind of broken (because it screws up dependency tracking), but a fresh build should be fine.  

I've had the same thing happen a couple times here when deleting files, so it's probably worth figuring out how to fix this.  I'm fairly confident that something is going wrong with dependency generation, which leads to make recursively re-loading the Makefile and re-starting dependency generation.  So all we need to do is break that loop, or possibly even just report that it's happening so it's not silently wedged.

scottlaird, 2024-03-02:
So, to reproduce, if I run `make clean; make -j` on a clean tree, that works fine.  Then, I create a file `src/devices/foo.c` with the one line `#include "foo.h"` and re-run make, and I get an endless string of 

```
/home/scott/git/linuxcnc/src/Makefile.modinc:107: warning: ignoring old recipe for target 'install'
Makefile:80: warning: overriding recipe for target 'install'
/home/scott/git/linuxcnc/src/Makefile.modinc:107: warning: ignoring old recipe for target 'install'
Makefile:80: warning: overriding recipe for target 'install'
/home/scott/git/linuxcnc/src/Makefile.modinc:107: warning: ignoring old recipe for target 'install'
Makefile:80: warning: overriding recipe for target 'install'
/home/scott/git/linuxcnc/src/Makefile.modinc:107: warning: ignoring old recipe for target 'install'
Makefile:80: warning: overriding recipe for target 'install'
```

scottlaird, 2024-03-02:
Removing `-MG` from the `%.d: %.c` rule seems to fix things.  I'll check it in in a minute.

closed 2024-03-02 by scottlaird

vr6111, 2024-03-04:
Great, it works now. Thank you very much.

PR #369 (2024-03-02): build: Fix hanging build issue
Closes #368