linuxcnc-ethercat/linuxcnc-ethercat #372: Delta DEASDA Digital Output support

pippin88, 2024-03-08:
Delta DEASDA does not currently support digital outputs.

Can digital output be added the driver please?

Delta A3 manual:
```
60FEh Sub1 Physical Outputs
Bit - DO - Description
0-15 - reserved
16 - D01 - 0: off, 1: on
17 - D02 - 0: off, 1: on
18 - D03 - 0: off, 1: on
19 - D04 - 0: off, 1: on
20-31  - reserved
```

```
60FEh Sub2 Bit Mask
Bit - DO - Description
0-15 - reserved
16 - D01 - 0: disable physical outputs, 1: enable
17 - D02 - 0: disable physical outputs, 1: enable
18 - D03 - 0: disable physical outputs, 1: enable
19 - D04 - 0: disable physical outputs, 1: enable
20-31  - reserved
```

Unfortunately I have very little coding experience and don't understand these drivers and so I can't implement it myself.

scottlaird, 2024-03-08:
This shouldn't be that hard.  There are two ways to do this:

1.  Try the `basic_cia` driver with the A3.  It requires a very different XML config, you can probably run `lcec_configgen` to get a start and then remove things that you don't need.  Downsides: (a) it uses different pin names than the `deasda` driver (b) it really wants you to use the `cia402` HAL component alongside it (c) it may not actually work on the A3.
2. Add `class_dout` support to `lcec_deasda.c`.  You'll need to `#include "lcec_class_dout.h"` at the top, add 0x60fe:01 to the sync mappings in `lcec_deasda_pdos_out_csv` and `lcec_deasda_pdos_out_csp`, add `lcec_class_dout_channels_t *dout` to `lcec_deasda_data_t`, initialize it in `lcec_deasda_init` with one call to `hal_data->dout = lcec_dout_allocate_channels(4)` and 4 calls like `hal_data->dout->channel[0] = lcec_dout_register_channel_packed(slave, 0x60fe, 1, 16, "dout-d01")` (change the 16 to 17, 18, 19 and the `dout-d01` to 02, 03, 04), manually set SDO `0x60fe:02` to `0x000f0000`, and finally call `lcec_dout_write_all(hal_data->dout)`  in `lcec_deasda_write`.

I need to get on a plane in a few minutes, or I could probably just whip this up for you.  Almost all of the infrastructure is already there.

scottlaird, 2024-03-09:
Okay, give v1.26.0 a spin.  No guarantees that it works, but it compiles cleanly and looks okay at a first glance.

If it crashes, then send me as much debugging output as possible (especially dmesg output).

pippin88, 2024-03-09:
Thanks Scott.

Unfortunately crashes when I try to start LinuxCNC. I have not changed my hal / ini / xml at all for outputs yet -> just tried to load previous working config with the updated 1.26 ethercat

```
[  860.998506] EtherCAT ERROR 0 0:0: PDO entry 0x60F3:01 is not mapped.
```

I guess this is a simple typo? e vs 3? Should be 0x60FE:01?
My knowledge is so limited I don't really understand hex...
I'm just using linuxcnc installed from live-usb and so I don't think I'm able to edit the source and fiddle myself (sorry)

```
mill@mill:~$ linuxcnc
LINUXCNC - 2.9.2
Machine configuration directory is '/home/mill/linuxcnc/configs/asda3_outputs'
Machine configuration file is 'ethercat-test.ini'
Starting LinuxCNC...
(time=1709955165.663712,pid=4569): Registering server on TCP port 5005.
(time=1709955165.664262,pid=4569): running server for TCP port 5005 (connection_socket = 3).
emc/iotask/ioControl.cc 786: can't load tool table.
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX realtime
Found file(REL): ./ethercat-test.hal
Failed to register PDO entry: No such file or directory
LCEC: master 0 PDO entry registration failed
LCEC: failure, clearing config
LCEC: exiting
LCEC: returning -EINVAL
lcec: rtapi_app_main: Invalid argument (-22)
./ethercat-test.hal:4: waitpid failed /usr/bin/rtapi_app lcec
./ethercat-test.hal:4: /usr/bin/rtapi_app exited without becoming ready
./ethercat-test.hal:4: insmod for lcec failed, returned -1
Shutting down and cleaning up LinuxCNC...
(time=1709955166.187605,pid=4569): Deleting 5 channels from the NML_Main_Channel_List.
(time=1709955166.187626,pid=4569): Deleting emcCommand NML channel from NML_Main_Channel_List.
(time=1709955166.187634,pid=4569): deleting NML (1)
(time=1709955166.187642,pid=4569):  delete (CMS *) 0x56547f7b3300;
(time=1709955166.187652,pid=4569): rcs_shm_close(shm->key=1001(0x3E9),shm->size=8192(0x2000),shm->addr=0x7fa282d60000)
(time=1709955166.187671,pid=4569): deleting CMS (emcCommand)
(time=1709955166.187680,pid=4569): free( data = 0x56547f7b4030);
(time=1709955166.187689,pid=4569): Leaving ~CMS()
(time=1709955166.187696,pid=4569):  CMS::delete(0x56547f7b3300)
(time=1709955166.187704,pid=4569):  CMS::delete successful.
(time=1709955166.187713,pid=4569): Leaving ~NML()
(time=1709955166.187720,pid=4569): NML channel deleted from NML_Main_Channel_List
(time=1709955166.187730,pid=4569): Deleting emcStatus NML channel from NML_Main_Channel_List.
(time=1709955166.187738,pid=4569): deleting NML (2)
(time=1709955166.187745,pid=4569):  delete (CMS *) 0x56547f7b9510;
(time=1709955166.187755,pid=4569): rcs_shm_close(shm->key=1002(0x3EA),shm->size=20480(0x5000),shm->addr=0x7fa282d5b000)
(time=1709955166.187770,pid=4569): deleting CMS (emcStatus)
(time=1709955166.187778,pid=4569): free( data = 0x56547f7ba220);
(time=1709955166.187801,pid=4569): Leaving ~CMS()
(time=1709955166.187818,pid=4569):  CMS::delete(0x56547f7b9510)
(time=1709955166.187827,pid=4569):  CMS::delete successful.
(time=1709955166.187834,pid=4569): Leaving ~NML()
(time=1709955166.187841,pid=4569): NML channel deleted from NML_Main_Channel_List
(time=1709955166.187848,pid=4569): Deleting emcError NML channel from NML_Main_Channel_List.
(time=1709955166.187855,pid=4569): deleting NML (3)
(time=1709955166.187861,pid=4569):  delete (CMS *) 0x56547f7bf900;
(time=1709955166.187870,pid=4569): rcs_shm_close(shm->key=1003(0x3EB),shm->size=8192(0x2000),shm->addr=0x7fa282d59000)
(time=1709955166.187884,pid=4569): deleting CMS (emcError)
(time=1709955166.187893,pid=4569): free( data = 0x56547f7c0620);
(time=1709955166.187901,pid=4569): Leaving ~CMS()
(time=1709955166.187907,pid=4569):  CMS::delete(0x56547f7bf900)
(time=1709955166.187914,pid=4569):  CMS::delete successful.
(time=1709955166.187921,pid=4569): Leaving ~NML()
(time=1709955166.187928,pid=4569): NML channel deleted from NML_Main_Channel_List
(time=1709955166.187935,pid=4569): Deleting toolCmd NML channel from NML_Main_Channel_List.
(time=1709955166.187942,pid=4569): deleting NML (4)
[... 60 lines cut ...]
[  726.902227] EtherCAT: Successfully requested master 0.
[  727.058491] EtherCAT ERROR 0 0:0: PDO entry 0x60F3:01 is not mapped.
[  727.058512] EtherCAT 0: Releasing master...
[  727.058515] EtherCAT 0: Released.
[  790.606118] EtherCAT: Requesting master 0...
[  790.606122] EtherCAT: Successfully requested master 0.
[  790.754528] EtherCAT ERROR 0 0:0: PDO entry 0x60F3:01 is not mapped.
[  790.754553] EtherCAT 0: Releasing master...
[  790.754556] EtherCAT 0: Released.
[  842.199154] EtherCAT: Requesting master 0...
[  842.199157] EtherCAT: Successfully requested master 0.
[  842.350485] EtherCAT ERROR 0 0:0: PDO entry 0x60F3:01 is not mapped.
[  842.350509] EtherCAT 0: Releasing master...
[  842.350512] EtherCAT 0: Released.
[  860.844842] EtherCAT: Requesting master 0...
[  860.844845] EtherCAT: Successfully requested master 0.
[  860.998506] EtherCAT ERROR 0 0:0: PDO entry 0x60F3:01 is not mapped.
[  860.998529] EtherCAT 0: Releasing master...
[  860.998533] EtherCAT 0: Released.
[ 1202.164265] EtherCAT: Requesting master 0...
[ 1202.164268] EtherCAT: Successfully requested master 0.
[ 1202.306478] EtherCAT ERROR 0 0:0: PDO entry 0x60F3:01 is not mapped.
[ 1202.306502] EtherCAT 0: Releasing master...
[ 1202.306506] EtherCAT 0: Released.
```

scottlaird, 2024-03-09:
Yeah, a typo on my part.  1.26.1 should be out in a minute.

pippin88, 2024-03-09:
Thanks Scott.

Hasn't come through apt update yet.

I have installed a run in place (RIP) version of LinuxCNC, but so far have not managed to get ethercat going with that.
Only instructions I can find are for older repos etc etc.
How do you install it for development?

Is there a donate button somewhere? I would be very happy to contribute, e.g. to costs of buying hardware for testing / development

scottlaird, 2024-03-09:
Weird.  1.26.1 didn't trigger a build for some reason.  I don't see an error.  I'll see what I can do about that...

scottlaird, 2024-03-09:
Okay, 1.26.2 seems to be building now.  It should be out in ~5 minutes or so.

pippin88, 2024-03-10:
Thanks Scott!

1.26.2 is working fine.

Digital outputs work on a Delta A3

scottlaird, 2024-03-10:
Great!  Let me know if you run into any more problems.

zmrdko, 2024-03-10:
Sorry to interrupt, but B3 does not support 60FE PDO:
```
Debug file information:
Note: Using POSIX realtime
HAL: ERROR: duplicate component name 'lcec_conf'
lcec_conf: ERROR: hal_init failed
./hallib/00_lcec.hal:2: waitpid failed lcec_conf lcec_conf
./hallib/00_lcec.hal:2: lcec_conf exited without becoming ready
Failed to execute SDO download: Input/output error
LCEC: slave 0.Y2: Failed to execute SDO download (0x60fe:0x02, size 4, byte0=0, error -5, abort_code 06020000)
LCEC: failed to configure slave 0.Y2 sdo for enabling digital output ports 1-4
LCEC: failure in proc_init for slave 0.Y2
LCEC: failure, clearing config
LCEC: exiting
LCEC: returning -EINVAL
lcec: rtapi_app_main: Invalid argument (-22)
./hallib/00_lcec.hal:3: waitpid failed /usr/bin/rtapi_app lcec
./hallib/00_lcec.hal:3: /usr/bin/rtapi_app exited without becoming ready
./hallib/00_lcec.hal:3: insmod for lcec failed, returned -1
HAL: ERROR: function 'lcec.read-all' not found
6240
Stopping realtime threads
Unloading hal components
Note: Using POSIX realtime
```
I have also checked the manual for B3 and there is no 60FE object in entire manual, unlike in A3 manual.

scottlaird, 2024-03-10:
OK, I'll flag it off for the b3, and check the a2.

pippin88, 2024-03-10:
Delta B3 manual that I have has 60FE (60FEh)

https://filecenter.deltaww.com/Products/download/06/060201/Manual/DELTA_IA-ASD_ASDA-B3_UM_EN_20231025.pdf
Manual from 2023-10-25 (same in older versions that I also have)

Object 60FEh Digital Outputs
Page 12-91 (page 828 of 1059)

Identical to A3 manual

Have you set the parameters to enable software control of digital outputs?
P2.018 = 0x0130
etc
See page 12-92 of B3 manual

scottlaird, 2024-03-10:
I just checked in a possible fix, although it's a bit more complicated than I'd like, so there's a chance that I've broken something else.

Once this finishes building, can you two check 1.26.3 and let me know if it works for you?

I had to replace the static sync config with one that's generated on the fly.  It's probably better this way, but there's a chance that I broke something in the process.

scottlaird, 2024-03-10:
I wish there was a good way to check and see if a slave supports a specific SDO, because we could flag this based on dynamically looking at the hardware.  Unfortunately, the "good" way to do this is an internal-only API in the Etherlab EtherCAT code, and the "bad" way (trying to read/write to the SDO and see what happens) generates errors in `dmesg` and could conceivably drop the device into an error state that we'd have to reset our way out of.

scottlaird, 2024-03-10:
Mind you, this basic problem is mostly why `lcec_configgen` exists.  There are so many options for CiA 402 that it's actually hard to use without some automated way of flagging things on and off.  It ended up being much easier to have an external tool do processing and then write an XML file that describes the hardware, as none of the alternatives actually looked like something that I'd be willing to live with.

pippin88, 2024-03-10:
The manual for B3 has the same PDO / information for Digital Outputs (60FE) as the A3 manual - so are we sure on what the issue is here?

zmrdko, 2024-03-10:
> Delta B3 manual that I have has 60FE (60FEh)
> 
> https://filecenter.deltaww.com/Products/download/06/060201/Manual/DELTA_IA-ASD_ASDA-B3_UM_EN_20231025.pdf Manual from 2023-10-25 (same in older versions that I also have)
> 
> Object 60FEh Digital Outputs Page 12-91 (page 828 of 1059)
> 
> Identical to A3 manual
> 
> Have you set the parameters to enable software control of digital outputs? P2.018 = 0x0130 etc See page 12-92 of B3 manual
> 
> Please share your xml and hal files

I was looking into this version: DELTA_IA-ASD_ASDA-B3_UM_EN_20220505

It is possible, that newer version drives (and firmware) could support outputs as well. I have 4 drives with 3 different firmwares, so I might check one by one, to verify my drives.

scottlaird, 2024-03-10:
The error from above is what I'd expect to see if the device doesn't support digital outputs, or if the EtherCAT modules doesn't *know* that it should support digital outputs.

Out of curiosity, do you have 0x60fe:01?  The error is about 0x60fe:02, which is the digital output enable/disable object.  It's probably possible to have :01 (the actual output object) without :02 (the enable object).  What does 'ethercat sdos' show?

zmrdko, 2024-03-10:
> I just checked in a possible fix, although it's a bit more complicated than I'd like, so there's a chance that I've broken something else.
>
> Once this finishes building, can you two check 1.26.3 and [...]

Thanks, it looks like it working again now. LinuxCNC opens again without any issues.
However it looks like I have been looking into outdated manual, so I have to try changing P2.018 = 0x0130 as @pippin88  wrote. So what do I have to reconfigure to make B3 load with digital outputs, if I want to try settings @pippin88 mentioned?

zmrdko, 2024-03-10:
> The error from above is what I'd expect to see if the device doesn't support digital outputs, or if the EtherCAT modules doesn't _know_ that it should support digital outputs.
>
> Out of curiosity, do [...]

I think its possible to read/write PDO's via terminal somehow, but I have to look it up. So I should try @pippin88 settings and try to read digital output pdo manually.

scottlaird, 2024-03-10:
The easiest way would be to edit `lcec_deasda.c` and add `FLAG_DOUT` to B3 as well.  Failing that, check and see if `ethercat sdos` shows either 0x60fe:01 or 0x60fe:02, and I can whip up something that should work pretty quickly.  Worse case, I'll just add types for B3 with and without digital outs.

zmrdko, 2024-03-10:
> The error from above is what I'd expect to see if the device doesn't support digital outputs, or if the EtherCAT modules doesn't _know_ that it should support digital outputs.
>
> Out of curiosity, do [...]

[ethercat_sdos.txt](https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/files/14551907/ethercat_sdos.txt)

zmrdko, 2024-03-10:
Looks like 2 out of 4 drives support outputs and 2 does not.

pippin88, 2024-03-11:
Have you tried updating the firmware of the drives?

zmrdko, 2024-03-11:
I did contact delta support twice already regarding firmware files. They dont want to provide it since they claim, that it’s risky procedure. They redirected me to my local reseller. And I am waiting for response from local reseller now. Anyway, I have already contacted Delta support and described this issue, that I have missing PDOs, depending on firmware versions and I mentioned pdf manual differences as well.

pippin88, 2024-03-11:
Unfortunately they do operate on a support from local distributor model. Probably good if you are a large business user, but not good for hobby users.

I bought a B3 servo (not ethercat :( ) in the past. I asked Delta (main website contact) what is the maximum input voltage (my supply goes up to 253V at times). They did not give me an answer, just referred me to my local country distributor.
My local country distributor did not really want to help, because I had bought the drives direct from China. I did eventually get some sort of answer, but it took a long time and phone calls etc etc for what should have been a very simple email reply.

pippin88, 2024-03-11:
Scott - what about a hal pin that can determine whether outputs should be enabled or disabled?

E.g.
{HAL_BIT, HAL_IN, offsetof(lcec_deasda_data_t, switch_on), "%s.%s.%s.digital-outputs-on"},

User can 
> setp digital-outputs-on 0
to disable outputs
Or 
> setp digital-outputs-on 1
to enable

Would this be a reasonable idea in the overall structure for LinuxCNC-Ethercat?
(I see you are doing a lot of work on overall structure and consistency)

scottlaird, 2024-03-11:
It's too late by then, we have to set up PDOs before the pins are really accessible.  The usual way to handle this is via modParams.  In this case, we could also just define two different types of B3, though -- one with analog outputs and one without.  It'd be kind of nice if we had an easy way to tell them apart, though :-(.

@zmrdko, are the 2 "good" and 2 "bad" B3s all the same basic hardware?  Does 'ethercat slaves -v' show anything different between the two?

zmrdko, 2024-03-11:
@scottlaird The drives are all ASD-B3-0421-E,
however there are slight differences:

Drive 1 and Drive 2: version on label 1.00; firmware version 1.0361, SDO 60FE present
Drive 3: version on label 1.00; firmware version 1.0161, SDO 60FE missing
Drive 4: version on label 1.04; firmware version 1.0462, SDO 60FE missing (this one has ribbed aluminium side, compared to flat on previous 3)

pippin88, 2024-03-11:
This delta branch on youtube might be helpful:
Delta product firmware request: https://www.youtube.com/watch?v=3DfAI0a133s
Delta ASD-A3/B3 firmware upgrade: https://www.youtube.com/watch?v=8t8Yl8JYjE8

zmrdko, 2024-03-11:
Thanks @pippin88, I have already seen that, and I already contacted my local dealer with firmware upgrade request.

GuiHue, 2024-04-21:
Guess i have an A2 that doesn’t like it. After wiring everything for EtherCAT, I realised that the previously working (2 months ago) A2 crashed during start: 
`Note: Using POSIX realtime Found file(REL): ./grootec.hal Failed to execute SDO download: Input/output error LCEC: slave 0.a-axis: Failed to execute SDO download (0x60fe:0x02, size 4, byte0=0, error -5, abort_code 06020000) LCEC: failed to configure slave 0.a-axis sdo for enabling digital output ports 1-4 LCEC: failure in proc_init for slave 0.a-axis LCEC: failure, clearing config LCEC: exiting LCEC: returning -EINVAL lcec: rtapi_app_main: Invalid argument (-22) ./grootec.hal:9: waitpid failed /usr/bin/rtapi_app lcec ./grootec.hal:9: /usr/bin/rtapi_app exited without becoming ready ./grootec.hal:9: insmod for lcec failed, returned -1 Shutting down and cleaning up LinuxCNC... Note: Using POSIX realtime LinuxCNC terminated with an error. You can find more information in the log: /home/groot/linuxcnc_debug.txt `

can we add a modparam to explicitly disable DO mapping?

zmrdko, 2024-04-21:
i think it was already removed, latest version does not crash for me.
(2 of 4 drives does support Digital Outputs PDOS and 2 does not)

scottlaird, 2024-04-21:
I added a modParam, using the same syntax as the CiA 402 driver.

Mind you, we could almost certainly make the ASDA devices use the CiA driver now, but it'd result in a number of pins changing names, and I don't have the hardware to test them.

Try adding `<modParam name="enableDigitalOutput" value="false"/>` and see if that does the trick.

GuiHue, 2024-04-22:
@scottlaird Works for me! Thanks a lot!

closed 2024-04-22 by scottlaird

zmrdko, 2024-05-03:
I have just had all my drives updated to the newest firmware version 2.1603 sub 7385 and 60FE seems to be there.
How do I enable outputs for B3 in xml config please?

adding <modParam name="enableDigitalOutput" value="true"/> throws an error:

Invalid modparam 'enableDigitalOutput'