linuxcnc-ethercat/linuxcnc-ethercat #381: About PDO mappings doc

jlama, 2024-03-21:
Hi,

In pdos-and-syncs.md you wrote:

> The one possible issue here is performance--I suspect (but haven't verified) that our EtherCAT library will look for existing PDO sync mappings and pick the smallest one that covers the PDO objects that we care about, and then sync everything listed in that PDO.

This is incorrect. If `slave->sync_info` is set, any existing PDO assignments and mappings will be simply overwritten.

scottlaird, 2024-03-21:
Ah, yeah, I should have been clearer that I meant "only if `sync_info` is unset".  I'll update the doc in a bit.

Is the SM2/SM3 case specific to Beckhoff, or is that a generic Ethercat thing?  I suppose I can dig through the Etherlab code to see how what they do...

Thanks for the review.  I'm still working my way through things here and keep finding things that are utterly undocumented and confusing, and I'm trying to document as I go and not be *too* wrong.

jlama, 2024-03-21:
This is how the Etherlab master seems to do things. I just looked at the code. The relevant function is `ecrt_slave_config_reg_pdo_entry`.

scottlaird, 2024-03-23:
Okay, I have some proposed changes in #382.  Care to take a look?

jlama, 2024-03-23:
LGTM

closed 2024-03-24 by scottlaird