linuxcnc-ethercat/linuxcnc-ethercat #396: DS402,modular prifile, dynamic protocol (dual axis drive)

claudiolorini, 2024-03-31:
Hello, i am trying to test under x86_64 platform using linuxcnc, some drives that i have up and running with a previous version of lcec using ARM platform (AMD Zynq ultrascale) but having some issues.

The drive uses the DS402 modular profile (since is a dual axis motor controller) and the DP profile that enables dynamically switching between CST/CSV/CSP profiles. 

This configuration uses a relevant number of pdos (see [ethercat-conf-CSTVP.xml.txt](https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/files/14816950/ethercat-conf-CSTVP.xml.txt))
this configuration is OK, i can reach OP state.

When i try to 'expand' the pdo in it's bit constituents using the halType="complex" construct (see 
[ethercat-conf-CSTVP.xml.txt](https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/files/14817004/ethercat-conf-CSTVP.xml.txt) )

i get this error: 
```
halcmd: loadusr -W  lcec_conf ethercat-conf-CSTVP.xml
halcmd: loadrt      lcec
Note: Using POSIX realtime
LCEC: lcec_pdo_init() failed for slave 0:0; lcec_pdo_entry_reg_t is full, with 128 of 128 entries used
LCEC: lcec_pdo_init() failed for slave 0:0; lcec_pdo_entry_reg_t is full, with 128 of 128 entries used
...
```

i'm not sure if this is related to the different version or the different platform, and i think i can continue experimentation with the working config, but if there is some interest in trying ironing out this, i can try to do some testing and/or provide further info.
Feel free to close the issue at leisure if not in the scope of the project.

Claudio

scottlaird, 2024-03-31:
Combined CST/CVT/CSP profile?  Interesting.  Vendor ID 664, so "Istituto Italiano di Tecnologia"?  Sounds fun.

The error that you're hitting just says that we've filled up a statically-sized array.  IIRC until a couple month ago, it'd just crash instead, at least we have an error now.  I didn't want to swap something more dynamic in in its place yet.  I'm trying to limit how many hard-to-debug problems I cause per month :-).

You should be able to increase LCEC_MAX_PDO_REG_COUNT in `lcec.h` and recompile.  I have a device here that has hit the same limit, but it also has a flaky EtherCAT interface and I don't have it syncing yet, so I haven't bothered to raise the default yet.  I don't think there's any particular downside to increasing it, other than increased memory use.

scottlaird, 2024-03-31:
Also, I should mention #180, which adds a whole framework for multi-axis CiA 402 devices.  I don't know that it'll actually help you at all today, but it's under active development, and I have several multi-axis devices here to test.  It's not particularly user-friendly yet, but it's hopefully not *that* far away.

scottlaird, 2024-03-31:
I'll go ahead and increase the limit anyway.  It's not *that* much RAM.

scottlaird, 2024-03-31:
Version 1.27.4 should show up in a few minutes with this limit raised.

closed 2024-03-31 by scottlaird

claudiolorini, 2024-04-01:
> Vendor ID 664, so "Istituto Italiano di Tecnologia"? Sounds fun.

Yep, it is (mostly)

> Version 1.27.4 should show up in a few minutes with this limit raised.

i can confirm. i'm up and running!

[image]

> Also, I should mention https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/issues/180, which adds a whole framework for multi-axis CiA 402 devices.

Yes, i'm following (in stealth mode) the development. My idea is to test as much as i can of your work using our drives now that there is a big impulse of work in the project; i think can be useful to 'strike while the iron it's hot'.
I'll let you know if i find some 'strange behavior' and i'll try to un-stealth if i can be of some assistance.

thanks for the good work, it makes things so much easier.
C.

PR #397 (2024-03-31): fix: Increase the number of PDOs allowed per device from 128 to 256
Fixes #396, and is also needed for #328.