linuxcnc-ethercat/linuxcnc-ethercat #414: Support Omron MX2 VFD w/ EtherCAT

scottlaird, 2024-04-20:
The Omron MX2 is a common VFD, and it has an EtherCAT option available.  It supports the VL CiA 402 profile.

There's one weird thing with it that will end up requiring some additional CiA 402 infrastructure work -- it looks like it really only wants 1 PDO entry per PDO, *but* it supports 256 RX and TX PDOs.

Our usual model is to cram 8-12 entries into one RX and one TX PDO per axis.

The sane way to handle this is to model CiA PDO entries at a slightly higher level, and then "overflow" into the next PDO when we fill one up.  This will help with Leadshine devices, which only support ~8 PDO entries but have 3+ PDOs available.  It's easier to do this now when we only have 3 uses of `lcec_class_cia402` rather than wait until there are more to update.

scottlaird, 2024-04-27:
Hey, success!  I was able to "manually" enable the MX2 via the CiA 402 control word and set the spindle speed via `srv-target-vl`.  Setting it to 10000 made it go reasonably fast.  Setting it to 0 stopped it.

I'm not entirely convinced that it was running at 10k RPM, but that's not an *EtherCAT* problem as much as it is a MX2 setup problem.

scottlaird, 2024-04-27:
For the record, here what I did to set it up.  YMMV, these aren't necessarily correct even for my hardware, much less yours, etc.

I reset the MX2's settings using the procedure in the manual.  Then I had to set B037 to 0, otherwise I could only see a subset of all of the config settings.  This allegedly defaults to 0, but not on mine, apparently.

Then I set A004 to 400 (the maximum frequency that my test spindle supports) and A003 to 400.

I reduced B012 (the maximum power percentage), but I'm not convinced my value is right; the device display didn't match the manual.

Then I set H004 to 4 (number of motor poles) and B086 to 60 (multiplier for frequency -> RPM display).

Finally, I set A001 and A002 to 4 (EtherCAT control, not front panel control)

I had to power-cycle the MX2, or it came up in an error state.  The manual says that this is normal after re-initializing the config.

Then I fired up LinuxCNC.  Here's the XML snippet for the MX2:

```
    <slave idx="10" type="OMMX2" name="omron-mx2"/>
```

Not very complex :-).

My HAL config doesn't do anything with the MX2 yet.  I fired up LinuxCNC and then ran the following `halcmd` commands:

- `halcmd setp lcec.1.omron-mx2.srv-opmode 2` -- set to "VL" mode, which is the only mode that the MX2 supports.  Verified via `halcmd show  |grep mx2`.
- `halcmd setp lcec.1.omron-mx2.srv-cia-controlword 7`.  I *think* this is the first step for enabling it. 
- `halcmd setp lcec.1.omron-mx2.srv-cia-controlword 15`.  This should be the second step of the state machine dance.  It should now be enabled.
-  `halcmd setp lcec.1.omron-mx2.srv-target-vl 10000`.  It spins.
- `halcmd setp lcec.1.omron-mx2.srv-target-vl 0`.  It stops spinning.

The `controlword` dance should really be handled in code with error checking, but that's a higher-level HAL thing, not a MX2 CiA 402 driver thing.

scottlaird, 2024-04-27:
Going to mark this as closed.  The MX2 driver works fine, but we need VL support in general.  I'll open a new issue for that.

closed 2024-04-27 by scottlaird

rodw-au, 2024-04-27:
Great! Sounds like you need a cheap Digital Laser Tachometer to check speeds. eBay et. al. are awash with them for $20 or so. 
I found RS485 commands yielded very accurate speeds with one of these tachos.... Hopefully Ethercat will be the same.

scottlaird, 2024-04-27:
I have one, it's just not handy at the moment.  In any case, that'd be a problem with device setup, not really a CiA/driver problem, so I'm not too concerned about it.

Which is to say that I wouldn't try to mill something without resolving this, but as a software test bench, it's fine.