linuxcnc-ethercat/linuxcnc-ethercat #433: Delta MS300 support

zmrdko, 2024-05-08:
Hello,
I have tried to hook up Delta MS300 with EtherCAT communication card. The drive gets to OP state but I am missing hal pins like run forward, to control VFD. Did I miss something?

GuiHue, 2024-06-17:
Same here. That driver does not appear to be working or - at the very least - is missing a working reference config. I am at a similar point: see here https://www.forum.linuxcnc.org/ethercat/52845-ms300-ethercat-not-going-into-op-state-with-cmm-02-rev-131328?start=10#303148

I briefly looked at: https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blowb/master/src/devices/lcec_dems300.c
To me it appears that the driver does indeed set all the bits in 6040h to 1 as required. 
Looking at See: https://deltaacdrives.com/Delta-CMM-EC02-EtherCAT-Comm-Card-Operation-Manual.pdf --> Chapter 5, Specifically 5.2 as well as Page 59
I wonder if the sequence of Bits 0-3 is important? Then again: My layman's understanding of the code takes the status into consideration before changing the controlword. It does so for all relevant bits.

With regard to forward/ backward: I assume this is done through absolute rpm?

scottlaird, 2024-06-17:
I've been able to drive an Omron MX2 VFD over EtherCAT using the generic driver with a bit of help.  We *should* have VL support at this point.  The MX2 is kind of weird, in that it only allows 2 PDO entries per PDO, but it has 512 available PDOs.  Prior to this, the smallest I'd seen was 8 PDO entries, so some extra code was needed.

What does lcec_configgen show if you run it against the hardware?

GuiHue, 2024-06-17:
All it produces is:
    <slave idx="9" type="DeMS300" name="D10"/>

scottlaird, 2024-06-17:
Ah, yeah -- it recognized that it has a driver already in the tree.  I keep meaning to add a flag to disable that.

As for the MX2, it merged in #418, so it's in v1.35 and up.

scottlaird, 2024-06-17:
Okay, I actually *did* add the flag -- try `--typedb=false`.  It will generate ignore everything but generic and basic CiA 402 drivers.

GuiHue, 2024-06-17:
I’ll look into that. There is also always the generic driver and dbraun1981 cia402. 

My problem may be elsewhere?, full write up here: https://forum.linuxcnc.org/ethercat/52982-ms300-with-cmm-ec01-ec02-op-but-not-acting-as-expected

scottlaird, 2024-06-17:
I'll try to look into it soon, but I'm swamped in work right now and my Linux dev box in the middle of being RMAed due to an apparent motherboard failure, so everything's kind of slow at the moment.

GuiHue, 2024-06-18:
Shame on me...
I had linked x.x.fault-reset to machine on --> reviewing the state machine code line by line: This means a certain control bit never reaches 1. That solves the issue of the spindle not starting. What remains is the RPM thing (I am a factor of 10 too high...) and @zmrdko point of direction. The former can be addressed by the scale factor. According to the manual, MS300 accepts negative RPM, but linuxcnc doesn't create that. I guess I'll use a hal comp triggered by spindle.0.reverse to multiply by  -1.

GuiHue, 2024-06-18:
Update:
@zmrdko The drive takes the RPM as a signed integer and linuxcnc provides the spindle rpm signed as well. There is no need for forward/reverse pins. It really is simple if you don't mess it up as I did.

zmrdko, 2024-06-18:
Thanks!! I will have look into it as soon as possible. Do you have working hal config uploaded on github? I was recently occupied with making EL6751 work with canopen VFD.

GuiHue, 2024-06-18:
> Thanks!! I will have look into it as soon as possible. Do you have working hal config uploaded on github? I was recently occupied with making EL6751 work with canopen VFD.

It is part of my config that I am currently building up. Spindle part is working:
https://github.com/GuiHue/grootec
Specifically:
https://github.com/GuiHue/grootec/blob/main/lib/spindle.hal

zmrdko, 2024-06-20:
got it working - THANKS!!

closed 2025-02-10 by zmrdko