linuxcnc-ethercat/linuxcnc-ethercat #435: ADD Velocity Mode for EL70x1 series stepper drivers

GuiHue, 2024-06-18:
If I understood both the existing driver and the manual correctly, then:
- the current driver does not support velocity mode
- The hardware has 7010:21 Set Velocity as an option.

Would it be possible to create a velocity mode for the Drivers? Applications are stuff like stepper driven pumps for MQL etc where velocity is used instead of positions.

Alternatively: Are there other ways to create pos commands from velocity inputs?

scottlaird, 2024-06-18:
It's probably not that hard, but IIRC my EL70x1 drives don't support velocity mode.  Various members of the family add and remove various features.

It'd probably be worth merging the existing 2 (or was it 3) drivers and getting feature support synced up right.

scottlaird, 2024-06-18:
Mind you, you could probably also have something write a constant delta into the position pin every cycle.  It's not ideal in a lot of ways, but it'd work in a pinch.

GuiHue, 2024-06-19:
Scott, thanks as always. I was thinking about this today. 
I find the manual to be surprisingly difficult to read. Lots of stuff covered but I was unable to find what I was looking for.

By the driver lcec.0.EL7031.srv-pos-cmd is tied to 7010h:11 which according to the manual is UINT32. The pin is declared as s32 within the driver (and float in linuxcnc).

This raises two questions: 
1) What does srv-pos-cmd expect? An absolute position? Or an incremental position to be driven within the next cycle?
2) How is direction towards the Driver handled? If 7010:11 is UINT32, it doesnt have direction, right? In the write function of the driver I could not identify a bit that handles direction.

If 1) is absolute, than handling an extended period of constant velocity may lead to an overflow of the variable in which case I do not know what the driver will do. Admittedly UINT32 is large and the pump is really slow.

scottlaird, 2024-06-19:
So, er, we kind of do support velocity mode for these, in an annoyingly weird way.

The [EL70x1 driver is position-only](https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/4299f7a10c8890e7e968fd736b20aa628e2fd0e2/src/devices/lcec_el70x1.c#L189), while the [EL7041 drive is velocity-only](https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/4299f7a10c8890e7e968fd736b20aa628e2fd0e2/src/devices/lcec_el7041.c#L302).

Odds are you could do a quick hack and remove your device's PID from the 70x1 driver and add it to the 7041 driver and it'd mostly just work.

These should really just be a single driver, with selectable position/velocity settings, except that might be fun if we need to default to position or velocity mode depending on which driver originally supported the device, just for config compatibility reasons.  I haven't read through the two drivers enough to see what that'd take today, but I doubt it's *that* much work.

grandixximo, 2026-10-03:
Second ping (first was 2026-04-30, no reply). The offer stands: merge `lcec_el70x1.c` and `lcec_el7041.c` with a `mode` modParam (position/velocity) if someone with the hardware can test. @GuiHue - still on your radar? koppi's EL7041-velocity workaround above is the interim path.