linuxcnc-ethercat/linuxcnc-ethercat #436: Leadshine AC servo drive EL6-xxx series - could not control

dpflexvn, 2024-07-22:
Hello,
I just made question on LCNC forum: https://forum.linuxcnc.org/ethercat/53307-ethercat-leadshine-el6-series-could-read-but-not-control
I think that ESI of EL6-xxx and EL7-xxx is same. Please refer ESI file from Leadshine.
Can you advise to solve ?.
Thank you.
[EL6-EC_EL7-ECN ESI (xml file).zip](https://github.com/user-attachments/files/16335959/EL6-EC_EL7-ECN.ESI.xml.file.zip)

scottlaird, 2024-07-22:
Can you add your XML config here?  I'll try to look at it later today or possibly tomorrow.

dpflexvn, 2024-07-23:
Hello Scottlaird,
Thank you for your quick feedback.
Today I found root cause which is default setting of 6092. I revised it to "10000" and it works.
I close this ticket. Thank you again.