linuxcnc-ethercat/linuxcnc-ethercat #437: Deviation between sittner/linuxcnc-ethercat and linuxcnc-ethercat/linuxcnc-ethercat

Bouni, 2024-08-30:
Hi,

if I remember right, when you started this repo you had contact with @sittner and the idea was to continue developement here in this repo.

Now I've seen that in [sittner/linuxcnc-ethercat](https://github.com/sittner/linuxcnc-ethercat/commits/master/) have been a number of commits this year, most of them last month.
The last commit in this repo has been 4 months ago. Is this repo still active or do you plan on merge the changes of @sittner into this repo as well?
I find it very nice that this can be installed via apt and has not to be compiled from source every time.

This is not meant to be negative criticism or put preassure on you, I just have seen the difference and wondered why this is ....

You did an awesome job so far!

scottlaird, 2024-09-03:
I'm not sure what's up with his tree.  I tried to merge a couple of his changes, but Git has diverged too far to make it easy.

I've been kind of underwater with "real" work for a while now, and I just haven't had the energy to work on more code after working on paying code all day.  I keep thinking it'll slow down a bit, but then something else comes up that needs another month of creativity and attention, over and over again.  Which is fine, it makes for a fun job, but it's made it hard to get a lot done here.

August was *finally* slightly better, but I spent my free time doing a bit of vacationing.

Bouni, 2024-09-03:
As I said, no preassure, I just wondered what's going on. I know that kind of problems all to well 😄

closed 2024-09-03 by Bouni