linuxcnc-ethercat/linuxcnc-ethercat #438: Bugs in ECT60 and CIA402 drivers. Can't set some modParams

rodw-au, 2024-09-01:
I am a bit stuck here. Finally tried to get homemod working.

If I use the basic_cia402 driver, I am unable to set
```
      <modParam name="peakCurrent_amps" value="2.0"/>
      <modParam name="controlMode" value="closedloop"/>	
```
6 amps into a 2 amp motor may not be wise.

If I use the ECT60 driver, I am unable to set
```
<modParam name="homeMode" value="17"/>
```
In either case, I can't play with homing further.

I might revert to the generic XML driver but I am out of time tonight.

The good news is the drive started to home!

scottlaird, 2024-09-02:
There's a chance that homemode isn't plumbed through right for RT steppers.  I'll take a look later today.

scottlaird, 2024-09-03:
Ah, okay -- `homeMode` isn't there anywhere.  That'd be the problem.  I'll need to add it.  I'll take a look tonight, it shouldn't be much work, but I don't have time right now.

rodw-au, 2024-09-03:
Thanks I did look at the code but had no  idea where to start. Its good to find these things...

scottlaird, 2024-09-03:
Yeah, the CiA spec is messy enough with almost everything being optional that the code ends up being kind of complex. It's probably only ~20 lines of code, but they'll be kind of weird. C isn't a great language for doing this sort of thing cleanly.