linuxcnc-ethercat/linuxcnc-ethercat #441: Change Omron G5 type names

Bouni, 2024-09-04:
The repo contains an [example](https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/master/examples/omrg5/omrg5.xml) for Omron G5 Servos:

```xml
<masters>
	<master idx="0" appTimePeriod="1000000" refClockSyncCycles="1">
		<slave idx="0" type="R88D-KN04H-ECT">
			<dcConf assignActivate="300" sync0Cycle="*1" sync0Shift="0"/>
			<watchdog divider="2498" intervals="1000"/>
		</slave>
	</master>
</masters>
```

But this does not corespond with the type names used in the actual code:

https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/af7af3396be0b6765d871287393ea1a4d0913e85/src/devices/lcec_omrg5.c#L30-L52

I suggest that we change the kind of cryptic `OmrG5_KN04H` for `R88D-KN04H-ECT` as shown in the example.
Obviously for all the types, not just the 400W model 😉 
Its much better readable an makes more sense in my opinion.

I'll submit a PR if you agree.

scottlaird, 2024-09-08:
That sounds reasonable to me, but I'd rather not break old configs that depend on the old names.  I doubt there are very many of them, but still.  Could you add the new names while keeping the old names and just add a comment that they're deprecated?

Bouni, 2024-09-09:
Sure, sound like a good idea.

Maybe we should add some kind of deprecation warning tat gets printed every time one of the old typenames is used wth a mesage like "The used typname xxxx is going to be deprecated soon, please replace it with yyyy".