linuxcnc-ethercat/linuxcnc-ethercat #445: Links in documetation are broken

Bouni, 2024-09-16:
I realized that the docs for `rtdrv` are broken:

[image]

When clicking the link to `Improve this page`, that link is broken as well:

[image]

The link behind `Improve this page` is `https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/edit/master/rtdrv.md` but should be `https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/edit/master/documentation/rtdrv.md` (the documentation subfolder is missing).

That is true for all `Improve this page` links. However I was not able to figure out where these come from in order to provide a PR to fix this issue ...

scottlaird, 2024-09-23:
This is the version of the docs from https://linuxcnc-ethercat.github.io/linuxcnc-ethercat/?

It passes through [.github/workflow/pages.yml](https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/master/.github/workflows/pages.yml) before turning into webpages; there's probably an option wrong there somewhere, although I don't see it at first glance?

Bouni, 2024-09-24:
I think that the root cause is that the files must be in the root directory:

[image]

In the GH action the source argument is set to documentation:

https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/f6c6af992dbc892b4371656654887a6ce0edd10e/.github/workflows/pages.yml#L41

Do you insist on using GH pages with the builtin Jekyll stuff or would you accept a PR with a change to something slightly more flexible?

I have no ide what to use at the moment (hugo?) but I could check out a few options and do a small test on my fork if you are ok with that

scottlaird, 2024-09-24:
I just used Jekyll because it produced decent looking results with ~0 effort. Having actually-correct pages would (of course) be better.

IIRC there are a few complex tables lurking in there that I'd like to avoid modifying if possible. So whatever we use, it'd be nice if it stuck with GitHub's flavor of Markdown as much as possible.

Bouni, 2024-09-25:
I will try to use the markdown files as they are and build something with hugo that renders nice looking pages out of it.
Not sure about the time I need for that but we're not in a hurry, right 😏

Bouni, 2024-09-26:
@scottlaird I have a first draft working (still local, no GH actions involved):

https://dev.bouni.de/

What do you think? 

I had to tweak the the index.md a bit and the markdown files are now in a subdirectory, but apart from thart, nobig changes were needed.

Here's my branch if you're interested https://github.com/Bouni/linuxcnc-ethercat/tree/hugo-docs/documentation/content