linuxcnc-ethercat/linuxcnc-ethercat #446: Add support for Beckhoff EL6224 and EL9110

kworm83, 2024-09-23:
Hi, I'm planning to add support for the following Beckhoff modules:

EL6224 (IO Link Master)
EL9110 Power Supplier with LED (24V, Diagnostics)

https://www.beckhoff.com/en-us/products/i-o/ethercat-terminals/el6xxx-communication/el6224.html
https://www.beckhoff.com/en-us/products/i-o/ethercat-terminals/el9xxx-system/el9110.html

I'm thinking it would be two new drivers since I don't see anything similar to EL6224 and the lcec_9410.c is not really the same type of adapter other than power related.  Let me know if you have any other suggestions.

scottlaird, 2024-09-23:
Looking at those two devices:

- The EL6224 looks fairly different than anything else in the tree, so it should be its own driver.  I'm not entirely sure what IO Link is; presumably it's a serial sensor interface.  You'd probably need to do a bit of work to get the IO Link interface working, and then you'd need to decode the IO Link protocol and add IO Link devices from there, so it could be kind of interesting.
- The EL9110 is pretty trivial.  It looks like it has a single bit at `0x6000:01`, presumably true when the device is healthy and false when it isn't.  Most likely the EL9110 and EL9160 could share the same driver, and the EL9210, EL9260, and maybe a few others are pretty close, with a single extra bit for  "Fuse Error."

scottlaird, 2024-09-23:
Also, you could probably just ignore the EL9110  entirely and just leave it out of the XML file, and it'd work fine, you'd just be missing the diagnostic data.

kworm83, 2024-09-23:
Hi Scott,

For the EL9110 I'll share the EL9160 driver and rename it to el91xx or
should I make it el9xxx for any future ones?  Then there would still be the
more specific el9410 and el5xx.

I will have to get familiar with IO link myself but basically it's a
multiplexed i/o and in my case is used to connect several Festo
pneumatic solenoids...ie 16 port valve block with just one or two pair
cable to it for control.  From the PDOs and SDOs it looks like it's pretty
simple in that each channel is just a map.  Apparently each channel can
have 64 inputs or outputs so the hardest part may be figuring out a way to
limit the number of hal pins generated to what is actually in use.  I'm not
familiar enough yet with LCNC to know if having a large number of hal pins
is an issue but if they were all created then it looks like there would be
something like 512 pins.

---PDO---
=== Master 0, Slave 1 ===
SM0: PhysAddr 0x1000, DefaultSize  256, ControlRegister 0x26, Enable 1
SM1: PhysAddr 0x1100, DefaultSize  256, ControlRegister 0x22, Enable 1
SM2: PhysAddr 0x1200, DefaultSize    0, ControlRegister 0x24, Enable 0
  RxPDO 0x1600 "IO RxPDO-Map Outputs Ch.1"
  RxPDO 0x1601 "IO RxPDO-Map Outputs Ch.2"
  RxPDO 0x1602 "IO RxPDO-Map Outputs Ch.3"
  RxPDO 0x1603 "IO RxPDO-Map Outputs Ch.4"
SM3: PhysAddr 0x1300, DefaultSize    6, ControlRegister 0x20, Enable 1
  TxPDO 0x1a00 "IO TxPDO-Map Inputs Ch.1"
  TxPDO 0x1a01 "IO TxPDO-Map Inputs Ch.2"
  TxPDO 0x1a02 "IO TxPDO-Map Inputs Ch.3"
  TxPDO 0x1a03 "IO TxPDO-Map Inputs Ch.4"
  TxPDO 0x1a80 "DeviceState TxPDO-Map Inputs"
    PDO entry 0xf100:01,  8 bit, "State Ch1"
    PDO entry 0xf100:02,  8 bit, "State Ch2"
    PDO entry 0xf100:03,  8 bit, "State Ch3"
    PDO entry 0xf100:04,  8 bit, "State Ch4"
  TxPDO 0x1a81 "DeviceState TxPDO-Map Inputs"
    PDO entry 0x0000:00, 12 bit, "Gap"
    PDO entry 0xf101:0d,  1 bit, "Device Diag"
    PDO entry 0x0000:00,  2 bit, "Gap"
    PDO entry 0xf101:10,  1 bit, "Device State"

---SDO for channel 1 output---
SDO 0x1600, "IO RxPDO-Map Outputs Ch.1"
  0x1600:00, rwr-r-, uint8, 8 bit, "SubIndex 000"
  0x1600:01, rwr-r-, uint32, 32 bit, "SubIndex 001"
  0x1600:02, rwr-r-, uint32, 32 bit, "SubIndex 002"
  0x1600:03, rwr-r-, uint32, 32 bit, "SubIndex 003"
  0x1600:04, rwr-r-, uint32, 32 bit, "SubIndex 004"
  0x1600:05, rwr-r-, uint32, 32 bit, "SubIndex 005"
  0x1600:06, rwr-r-, uint32, 32 bit, "SubIndex 006"
  0x1600:07, rwr-r-, uint32, 32 bit, "SubIndex 007"
  0x1600:08, rwr-r-, uint32, 32 bit, "SubIndex 008"
  0x1600:09, rwr-r-, uint32, 32 bit, "SubIndex 009"
  0x1600:0a, rwr-r-, uint32, 32 bit, "SubIndex 010"
  0x1600:0b, rwr-r-, uint32, 32 bit, "SubIndex 011"
  0x1600:0c, rwr-r-, uint32, 32 bit, "SubIndex 012"
  0x1600:0d, rwr-r-, uint32, 32 bit, "SubIndex 013"
  0x1600:0e, rwr-r-, uint32, 32 bit, "SubIndex 014"
  0x1600:0f, rwr-r-, uint32, 32 bit, "SubIndex 015"
  0x1600:10, rwr-r-, uint32, 32 bit, "SubIndex 016"
  0x1600:11, rwr-r-, uint32, 32 bit, "SubIndex 017"
  0x1600:12, rwr-r-, uint32, 32 bit, "SubIndex 018"
  0x1600:13, rwr-r-, uint32, 32 bit, "SubIndex 019"
  0x1600:14, rwr-r-, uint32, 32 bit, "SubIndex 020"
  0x1600:15, rwr-r-, uint32, 32 bit, "SubIndex 021"
  0x1600:16, rwr-r-, uint32, 32 bit, "SubIndex 022"
  0x1600:17, rwr-r-, uint32, 32 bit, "SubIndex 023"
  0x1600:18, rwr-r-, uint32, 32 bit, "SubIndex 024"
  0x1600:19, rwr-r-, uint32, 32 bit, "SubIndex 025"
  0x1600:1a, rwr-r-, uint32, 32 bit, "SubIndex 026"
  0x1600:1b, rwr-r-, uint32, 32 bit, "SubIndex 027"
  0x1600:1c, rwr-r-, uint32, 32 bit, "SubIndex 028"
  0x1600:1d, rwr-r-, uint32, 32 bit, "SubIndex 029"
  0x1600:1e, rwr-r-, uint32, 32 bit, "SubIndex 030"
  0x1600:1f, rwr-r-, uint32, 32 bit, "SubIndex 031"
  0x1600:20, rwr-r-, uint32, 32 bit, "SubIndex 032"
  0x1600:21, rwr-r-, uint32, 32 bit, "SubIndex 033"
  0x1600:22, rwr-r-, uint32, 32 bit, "SubIndex 034"
  0x1600:23, rwr-r-, uint32, 32 bit, "SubIndex 035"
  0x1600:24, rwr-r-, uint32, 32 bit, "SubIndex 036"
  0x1600:25, rwr-r-, uint32, 32 bit, "SubIndex 037"
  0x1600:26, rwr-r-, uint32, 32 bit, "SubIndex 038"
  0x1600:27, rwr-r-, uint32, 32 bit, "SubIndex 039"
  0x1600:28, rwr-r-, uint32, 32 bit, "SubIndex 040"
  0x1600:29, rwr-r-, uint32, 32 bit, "SubIndex 041"
  0x1600:2a, rwr-r-, uint32, 32 bit, "SubIndex 042"
  0x1600:2b, rwr-r-, uint32, 32 bit, "SubIndex 043"
  0x1600:2c, rwr-r-, uint32, 32 bit, "SubIndex 044"
  0x1600:2d, rwr-r-, uint32, 32 bit, "SubIndex 045"
  0x1600:2e, rwr-r-, uint32, 32 bit, "SubIndex 046"
  0x1600:2f, rwr-r-, uint32, 32 bit, "SubIndex 047"
  0x1600:30, rwr-r-, uint32, 32 bit, "SubIndex 048"
  0x1600:31, rwr-r-, uint32, 32 bit, "SubIndex 049"
  0x1600:32, rwr-r-, uint32, 32 bit, "SubIndex 050"
  0x1600:33, rwr-r-, uint32, 32 bit, "SubIndex 051"
  0x1600:34, rwr-r-, uint32, 32 bit, "SubIndex 052"
  0x1600:35, rwr-r-, uint32, 32 bit, "SubIndex 053"
  0x1600:36, rwr-r-, uint32, 32 bit, "SubIndex 054"
  0x1600:37, rwr-r-, uint32, 32 bit, "SubIndex 055"
  0x1600:38, rwr-r-, uint32, 32 bit, "SubIndex 056"
  0x1600:39, rwr-r-, uint32, 32 bit, "SubIndex 057"
  0x1600:3a, rwr-r-, uint32, 32 bit, "SubIndex 058"
  0x1600:3b, rwr-r-, uint32, 32 bit, "SubIndex 059"
  0x1600:3c, rwr-r-, uint32, 32 bit, "SubIndex 060"
  0x1600:3d, rwr-r-, uint32, 32 bit, "SubIndex 061"
  0x1600:3e, rwr-r-, uint32, 32 bit, "SubIndex 062"
  0x1600:3f, rwr-r-, uint32, 32 bit, "SubIndex 063"
  0x1600:40, rwr-r-, uint32, 32 bit, "SubIndex 064"

Thanks,

Kevin

kworm83, 2024-09-23:
Yeah I know it will work if I just ignore it but I like things to be
complete and contribute when/where I can.

kworm83, 2024-09-23:
I was able to get the eni file contents for the EL6224s from the old
control...not sure if hal pins support array of bytes...

  <Slave Name="EL6224_1001[001-BC10.1]" PhysAddr="1001">
   <Description>
    <VendorId>2</VendorId>
    <ProductCode>#x18503052</ProductCode>
    <RevisionNo>#x00140000</RevisionNo>
   </Description>
   <PreviousPort>
    <PhysAddr>1000</PhysAddr>
    <Port>B</Port>
   </PreviousPort>
   <Dc Id="FreeRun"/>
   <Pdo>
    <Edit>
     <Entry Type="Tx" Name="TxPDO_0x1a05" Index="#x1a05"/>
     <Entry Type="Tx" Name="TxPDO_0x1a04" Index="#x1a04"/>
     <Entry Type="Tx" Name="TxPDO_0x1a01" Index="#x1a01"/>
     <Entry Type="Rx" Name="RxPDO_0x1601" Index="#x1601"/>
     <Entry Type="Tx" Name="TxPDO_0x1a02" Index="#x1a02"/>
     <Entry Type="Rx" Name="RxPDO_0x1602" Index="#x1602"/>
     <Entry Type="Tx" Name="TxPDO_0x1a03" Index="#x1a03"/>
     <Entry Type="Rx" Name="RxPDO_0x1603" Index="#x1603"/>
    </Edit>
   </Pdo>
   <PdoEntry>
    <Add>
     <Entry SubIndex="#x0001" Size="32" DataType="ARRAY [0..3] OF BYTE"
PdoIndex="#x1601" Name="Ventilinsel VTUG 16[460-V1]" Index="#x7010"/>
     <Entry SubIndex="#x0001" Size="32" DataType="ARRAY [0..3] OF BYTE"
PdoIndex="#x1602" Name="Ventilinsel VTUG 16[111-V2]" Index="#x7020"/>
     <Entry SubIndex="#x0001" Size="48" DataType="ARRAY [0..5] OF BYTE"
PdoIndex="#x1603" Name="Ventilinsel VTUG 24[361-V3]" Index="#x7030"/>
    </Add>
    <Edit>
     <Entry SubIndex="13" PdoIndex="#x1a05" Name="Device_diag"
Index="#xf101"/>
     <Entry SubIndex="16" PdoIndex="#x1a05" Name="Device_state"
Index="#xf101"/>
     <Entry SubIndex="1" PdoIndex="#x1a04" Name="State_channel1"
Index="#xf100"/>
     <Entry SubIndex="2" PdoIndex="#x1a04" Name="State_channel2"
Index="#xf100"/>
     <Entry SubIndex="3" PdoIndex="#x1a04" Name="State_channel3"
Index="#xf100"/>
     <Entry SubIndex="4" PdoIndex="#x1a04" Name="State_channel4"
Index="#xf100"/>
    </Edit>
   </PdoEntry>
   <ExcludePdo>
    <Add>
     <Entry Index="#x1a00"/>
     <Entry Index="#x1600"/>
    </Add>
   </ExcludePdo>
   <InitCmd>
    <Edit>
     <Entry SubIndex="1" Data="0000000000000000100108000083000000000300"
Index="#x8010" Ccs="1"/>
     <Entry SubIndex="1" Data="0000000000000000100108000083000000000300"
Index="#x8020" Ccs="1"/>
     <Entry SubIndex="1" Data="0000000000000000100108000085000000000300"
Index="#x8030" Ccs="1"/>
    </Edit>
   </InitCmd>
  </Slave>
  <Slave Name="EL6224_1002[001-BC10.2]" PhysAddr="1002">
   <Description>
    <VendorId>2</VendorId>
    <ProductCode>#x18503052</ProductCode>
    <RevisionNo>#x00140000</RevisionNo>
   </Description>
   <PreviousPort>
    <PhysAddr>1001</PhysAddr>
    <Port>B</Port>
   </PreviousPort>
   <Dc Id="FreeRun"/>
   <Pdo>
    <Edit>
     <Entry Type="Tx" Name="TxPDO_0x1a05" Index="#x1a05"/>
     <Entry Type="Tx" Name="TxPDO_0x1a04" Index="#x1a04"/>
     <Entry Type="Tx" Name="TxPDO_0x1a01" Index="#x1a01"/>
     <Entry Type="Rx" Name="RxPDO_0x1601" Index="#x1601"/>
     <Entry Type="Tx" Name="TxPDO_0x1a02" Index="#x1a02"/>
     <Entry Type="Rx" Name="RxPDO_0x1602" Index="#x1602"/>
     <Entry Type="Tx" Name="TxPDO_0x1a03" Index="#x1a03"/>
     <Entry Type="Rx" Name="RxPDO_0x1603" Index="#x1603"/>
    </Edit>
   </Pdo>
   <PdoEntry>
    <Add>
     <Entry SubIndex="#x0001" Size="16" DataType="ARRAY [0..1] OF BYTE"
PdoIndex="#x1a01" Name="EPI1809_0021[001-A10]" Index="#x6010"/>
     <Entry SubIndex="#x0001" Size="16" DataType="ARRAY [0..1] OF BYTE"
PdoIndex="#x1a02" Name="EPI1809_0021[001-A11]" Index="#x6020"/>
     <Entry SubIndex="#x0001" Size="48" DataType="ARRAY [0..5] OF BYTE"
PdoIndex="#x1a03" Name="HF08[001-A12]" Index="#x6030"/>
    </Add>
    <Edit>
     <Entry SubIndex="13" PdoIndex="#x1a05" Name="Device_diag"
Index="#xf101"/>
     <Entry SubIndex="16" PdoIndex="#x1a05" Name="Device_state"
Index="#xf101"/>
     <Entry SubIndex="1" PdoIndex="#x1a04" Name="State_channel1"
Index="#xf100"/>
     <Entry SubIndex="2" PdoIndex="#x1a04" Name="State_channel2"
Index="#xf100"/>
     <Entry SubIndex="3" PdoIndex="#x1a04" Name="State_channel3"
Index="#xf100"/>
     <Entry SubIndex="4" PdoIndex="#x1a04" Name="State_channel4"
Index="#xf100"/>
    </Edit>
   </PdoEntry>
   <ExcludePdo>
    <Add>
     <Entry Index="#x1a00"/>
     <Entry Index="#x1600"/>
    </Add>
   </ExcludePdo>
   <InitCmd>
    <Edit>
     <Entry SubIndex="1" Data="0000000000000000110105001000000000000300"
Index="#x8010" Ccs="1"/>
     <Entry SubIndex="1" Data="0000000000000000110105001000000000000300"
Index="#x8020" Ccs="1"/>
     <Entry SubIndex="1" Data="0000000000000000110328008400000000000300"
Index="#x8030" Ccs="1"/>
    </Edit>
   </InitCmd>
  </Slave>

kworm83, 2024-09-28:
I have the driver for the EL9110 done and testing a little more.  Will have a pull request soon..

grandixximo, 2026-10-03:
Second ping (first was 2026-04-30, no reply). @kworm83 you mentioned in 2024 that the EL9110 driver was done and a PR was coming - if the branch still exists, push it somewhere and we will take it from there (review, rebase, merge). EL6224 IO-Link remains a separate, larger scope.

kworm83, 2026-10-07:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Hi,

Unfortunately I had to abandon work on this due to time limitations.  I
ended up just using Twincat since it was much easier to implement.