linuxcnc-ethercat/linuxcnc-ethercat #447: Add support for KEB G6 in VL mode

kworm83, 2024-09-23:
I'm planning to add support for a KEB G6 VFD in VL mode.  I will probably copy the Omron driver that supports VL unless you have another suggestion with some of your work on VL/Cia402/etc.

https://www.kebamerica.com/products/g6-open-loop-vfd/

grandixximo, 2026-10-03:
Second ping (first was 2026-04-30, no reply). The offer stands: extend `lcec_class_cia402` with VL mode (relates to #425/#426), then a thin KEB G6 driver on top. We need someone with the hardware to test against. @kworm83 - still interested? If the need has gone away, say so and we will close this.

kworm83, 2026-10-07:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Hi Luca,

Unfortunately I had to abandon work on this due to time limitations on this
project machine.   So you can close it.

closed 2026-10-07 by grandixximo