linuxcnc-ethercat/linuxcnc-ethercat #450: lcec_deasda index homing

zmrdko, 2025-01-05:
Hello,
I am trying to accomplish index homing on Delta B3 EtherCAT drives. Index signal is not available within SDOs, so I think the only way is to use servo internal homing procedure. Would it be difficult to enable this for deasda drives? I know there is a component for cia402 external homing by rodw. I am gathering info, so any input is appreciated!

thanks

zmrdko, 2026-05-01:
I would love to see this working, but I have issues with following error. The internal homing procedures are moves not invoked by linuxcnc, thats why it is considered following error.

grandixximo, 2026-05-01:
The following-error trip during internal homing is a known issue, no merged fix yet. Two community workarounds people are using today:

1. rodw's `cia402_homecomp` (https://github.com/rodw-au/cia402_homecomp) replaces LinuxCNC's `homing.c` and masks FE by forcing `pos-fb` to track `pos-cmd` during the home window. Most-cited solution but needs LinuxCNC source on disk.
2. PCW's HAL-only mux2 loopback approach (https://forum.linuxcnc.org/ethercat/57300-how-to-disable-following-error-while-homing): no recompile, just route `pos-cmd <- pos-fb` while a "homing" pin is active.

If you tried rodw's component and still got FE, that likely means the loopback wasn't actually engaging during the right window, not that the approach is wrong. Worth confirming `is_custom_homing` is high during the move.

For the linuxcnc-ethercat side, the cleanest long-term fix would be: migrate `lcec_deasda` onto `lcec_class_cia402` (already has `homeMethod`/0x6098 plumbing), then bake the pos-cmd / pos-fb suppression into the driver itself so no external homecomp is needed. Are you up for testing that path on B3 hardware?

grandixximo, 2026-10-03:
Second ping (first was 2026-05-01, no reply). Since then `lcec_class_cia402` gained the `homeMethod` modParam (#438), so migrating the DeASDA driver onto the class lib would bring index homing with it - that is the path we would take. @zmrdko - can you commit to testing it on your hardware? If yes, we will do the migration.

zmrdko, 2026-10-03:
I would be happy to test it during in december/januar, as I am quite busy with other stuff right now.